project-alice-assistant/HermesLedControl #6: Matrix voice speaker

Desaydrone, 2018-10-23:
**How I did start SLC, with what arguments**
python3 main.py --hardware=matrixvoice --pattern=custom

**Led device name**
Matrix Voice

**The bug**
At launch i got this log 

```
Starting Snips Led Control
Initializing SnipsLedControl
Loading configurations
Loaded 5 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using custom as pattern with 18 leds
Initializing leds controller
Traceback (most recent call last):
  File "main.py", line 87, in <module>
    main()
  File "main.py", line 73, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl/models/SnipsLedControl.py", line 102, in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl/models/LedsController.py", line 48, in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl/models/LedsController.py", line 84, in initHardware
    self._interface = MatrixVoice(numLeds=self._hardware['numberOfLeds'])
  File "/home/pi/snipsLedControl/interfaces/matrixvoice.py", line 11, in __init__
    self._leds 		= Everloop(numLeds)
  File "/home/pi/snipsLedControl/libraries/everloop.py", line 12, in __init__
    raise FileNotFoundError("Matrix Voice Everloop doesn't seem to be installed")
NameError: global name 'FileNotFoundError' is not defined
```

**To Reproduce**
Every time a launch the app. Hope it help

Psychokiller1888, 2018-10-23:
"Matrix Voice Everloop doesn't seem to be installed"

Psychokiller1888, 2018-10-23:
Also, debugging a device I don't have is hard. We made Respeaker Mic array with Kraken, on discord, took a couple of hours trying

Psychokiller1888, 2018-10-23:
Ok, I'll try another way

Psychokiller1888, 2018-10-23:
I push a try at fixing. Please try it and let me know the result. If you can join discord it would be much easier also

Desaydrone, 2018-10-29:
Hi, i'm on hollidays until november 4th i can only watch your fix at this date !! (i can't wait :p)

Desaydrone, 2018-11-05:
It works !! thanks for you job

Psychokiller1888, 2018-11-05:
You're welcome, thank you for your help and tests!

closed 2018-11-05 by Psychokiller1888

joeduck, 2018-11-19:
> 
> 
> It works !! thanks for you job

Hi. Can you tell me how you get it working? I start the service.  Got no error but se nothing.
`2018-11-19 21:44:43,270 [MainThread] - [INFO] - Connected to data publisher with port 20024
2018-11-19 22:02:48,837 [MainThread] - [INFO] - Starting Snips Led Control
2018-11-19 22:02:48,850 [MainThread] - [INFO] - Initializing SnipsLedControl
2018-11-19 22:02:48,851 [MainThread] - [INFO] - Loading configurations
2018-11-19 22:02:48,858 [MainThread] - [INFO] - Loaded 6 hardware references
2018-11-19 22:02:48,859 [MainThread] - [INFO] - - Mqtt server set to localhost
2018-11-19 22:02:48,860 [MainThread] - [INFO] - - Mqtt port set to 1883
2018-11-19 22:02:48,861 [MainThread] - [INFO] - - Client id set to default
2018-11-19 22:02:48,861 [MainThread] - [INFO] - - Hardware set to Matrix Voice
2018-11-19 22:02:48,862 [MainThread] - [INFO] - - Using google as pattern with 18 leds
2018-11-19 22:02:48,863 [MainThread] - [INFO] - Initializing leds controller
2018-11-19 22:02:50,192 [MainThread] - [INFO] - Connected to data publisher with port 20024`

Can you help?

Psychokiller1888, 2018-11-20:
I will add a debugger asap on the master, so we can see if the messages are actually captured

What start command do you use?

Desaydrone, 2018-11-20:
@joeduck First think did you do this for your matrix to be well recognize : https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/core-installation/ i use this to install matrix voice on my rasp. Then install snisledcontrol and it work  :)

Psychokiller1888, 2018-11-20:
Well, that part is now automated actually @Desaydrone , by the installer. And SLC is actually running as we see it connected to the data publisher port. So I suspect there's no mqtt messages

joeduck, 2018-11-20:
Hi,

thanks for your answers. Installed the matrix core with the setup routine from psycho. Then i startet snipsledcontrol with " python main.py --hardware=matrixvoice --pattern=google". 

Cause this not work, i installed python3 pip and tryed this. But no success. Actual i use matrix_hotword from romkabouter in the snips forum. This works.

Dont know what is wrong.
Thanks for your help.

joeduck, 2018-11-20:
Hello again,

installed malos how described on the matrix Page. For the matrix_hotword Servcie i dont have to do that. This seems to work with the kernel modules. However.

After this i start snipsLedControl again and some LEDs goes on. But nothing else. If i do a command to snips they still the same. Here the output from the service status:

`Nov 20 21:00:46 raspberrypi python[688]:     self._controller.setLedRGB(i, led)
Nov 20 21:00:46 raspberrypi python[688]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 343, in setLedRGB
Nov 20 21:00:46 raspberrypi python[688]:     self.setLed(ledNum, color[0], color[1], color[2], brightness)
Nov 20 21:00:46 raspberrypi python[688]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 333, in setLed
Nov 20 21:00:46 raspberrypi python[688]:     self._interface.setPixel(ledNum, red, green, blue, brightness)
Nov 20 21:00:46 raspberrypi python[688]:   File "/home/pi/snipsLedControl_v1.4/interfaces/matrixvoice.py", line 20, in setPixel
Nov 20 21:00:46 raspberrypi python[688]:     self._leds.setPixel(ledNum=ledNum, red=red, green=green, blue=blue, white=brightness)
Nov 20 21:00:46 raspberrypi python[688]:   File "/home/pi/snipsLedControl_v1.4/libraries/everloop.py", line 76, in setPixel
Nov 20 21:00:46 raspberrypi python[688]:     self._colors[ledNum] = ledValue
Nov 20 21:00:46 raspberrypi python[688]: IndexError: list assignment index out of range
`

Hope this helps.
Thanks for your assistance

Desaydrone, 2018-11-20:
Have you try with the Custom pattern instead of google one ?

joeduck, 2018-11-20:
Wow! This works :-) But crazy colours, now WAF by my wife ;-) .
O.k. Looks the matter are the led patterns.

Psychokiller1888, 2018-11-20:
It's definitly not python 3 for sure. I'll check the issue with the pattern