project-alice-assistant/HermesLedControl #13: Off : Led Blink

lricher1, 2018-11-11:
I use a pi3 and a respeaker4  (12 leds)
When the leds are off,  the leds at 21h,22h,23h  blink (Color could red, green yellow)
I change the pattern but i have the same problem with google, alexa or custom.
Thanks for your help.

Psychokiller1888, 2018-11-11:
You have to start SLC with python main.py --hardware=respeaker4 at least

The default custom has an idle animation, a blink. If you don't want it, you can remove it by adding the argument --idlePattern=off

lricher1, 2018-11-11:
In the "off mode", rhe error is only 3 leds blind 
When tooggleOn or Off, it's perfert, i have the google pattern

Psychokiller1888, 2018-11-11:
I'm not sure I understand and I cannot reproduce with your informations

lricher1, 2018-11-11:
In the off mode, 3 leds blink
In toggleOn mode, the google led is OK. but the same led as in mode off blink

Mabye the pattern "Off" contains a mistake.

Psychokiller1888, 2018-11-12:
There's no toggleOn mode. You say it's ok but it blinks. What is your native language? FR? DE? I speak both too

- I freshly installed on a respeaker 4
- Ran all those without a problem
```mosquitto_pub -p 1883 -t 'hermes/hotword/toggleOn' -m '{"siteId" : "default"}'
mosquitto_pub -p 1883 -t 'hermes/leds/toggleOff' -m '{"siteId" : "default"}'
mosquitto_pub -p 1883 -t 'hermes/hotword/default/detected' -m '{"siteId" : "default"}'
mosquitto_pub -p 1883 -t 'hermes/leds/toggleOn' -m '{"siteId" : "default"}'
mosquitto_pub -p 1883 -t 'hermes/hotword/default/detected' -m '{"siteId" : "default"}'
mosquitto_pub -p 1883 -t 'hermes/hotword/toggleOn' -m '{"siteId" : "default"}'
```

I even tested using your starting command, without = signs and with sudo, it worked straight away, even without stopping SLC service

lricher1, 2018-11-12:
Bonjour,
Quand j’exécute vos 5 commandes, j'ai bien les leds qui s'allument (comme prévu) mais j'ai en plus des leds qui s'allument de temps à autre. A l'instant même, c'est la led situé à 11h qui s'allume lentement en vert. Quelques instant après ce sont les leds située à 9h et 11h qui s'allume en magenta.

Apres la commande" sudo systemctl stop snipsledcontrol", plus de souci, toutes les leds restent éteintes en permanence.

Psychokiller1888, 2018-11-12:
Si tu stop le service, ouai, y'a plus rien, c'est normal

Les leds qui s'allument de temps à autre, je ne vois pas. On a dev sur respeaker 4 et on a jamais vu ça.

Tu as installé ton respeaker comment?

lricher1, 2018-11-12:
voici la doc d'install que j'ai faite : 
install raspbian stech lite
sudo apt-get update 
sudo apt-get upgrade 
wget https://nodejs.org/dist/v8.9.0/node-v8.9.0-linux-armv6l.tar.gz
tar -xzf node-v8.9.0-linux-armv6l.tar.gz
cd node-v8.9.0-linux-armv6l/
sudo cp -R * /usr/local/
sudo npm install -g snips-sam
sam connect 127.0.0.1 pi raspberry
sam init
sam setup audio 
alsamixer =>PGA à 100% et digi à 68%
sam test speaker
sam test microphone
sam login 	
sam install assistant 
sam install actions -g https://github.com/snipsco/snips-skill-respeaker.git
sudo service snips-audio-server restart
vi /etc/snips.toml 
=> mqtt server localhost to 192.168.1.13 
sound_feedback_disabled_default = true
------------------------------------
--led 
--doc : https://github.com/Psychokiller1888/snipsLedControl
cd /var/lib/snips/skills/snips-skill-respeaker
 vi config.init.default change serveur mqtt 192.168.1.13
 ./setup.sh

Activate SPI: sudo raspi-config,Go to "Interfacing Options",Go to "SPI",Enable SPI,Exit the tool
cd /home/pi
git clone https://github.com/respeaker/4mics_hat.git
cd /home/pi/4mics_hat
pip install numpy
pip install spidev gpiozero
python pixels_demo.py
wget https://gist.githubusercontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/6f31a789c95f0571b1a4e03838c63fbadd0e1d0f/slc_download.sh
sudo chmod +x slc_download.sh
sudo ./slc_download.sh

Psychokiller1888, 2018-11-12:
Non non non, tu peux pas tout mélanger non plus :) Si tu veux utiliser Snips Skill Respeaker tu utilises Snips Skill respeaker, qui ne fonctionne aps actuellement, ou tu utilises SLC. Le soucis vient de là, il y a 3 trois installé qui gèrent les leds...

Psychokiller1888, 2018-11-12:
Vire le skill respeaker et utilise l'installation officielle

http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

```sudo apt-get update
sudo apt-get upgrade
git clone https://github.com/respeaker/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh
reboot```

lricher1, 2018-11-12:
ok je teste ce soir

lricher1, 2018-11-12:
I delete /var/lib/snips/skills/snips-skill-respeaker and now it's perfert 
thank you

closed 2018-11-12 by lricher1

lricher1, 2018-11-12:
Bonjour,

Vous me semblez super caler sur snips. Pouvez vous me dire pourquoi j'ai 
l'erreur suivante :

"The session was ended because one of the component didn't respond in a 
timely manner"

Merci par avance,

Lionel

------ log sam watch

[22:09:21] [Asr] was asked to stop listening on site default
[22:09:21] [Hotword] was asked to toggle itself 'on' on site default
[22:10:14] [Hotword] detected on site default, for model hey_snips
[22:10:14] [Asr] was asked to stop listening on site default
[22:10:14] [Hotword] was asked to toggle itself 'off' on site default
[22:10:14] [Dialogue] session with id 
'959faf5d-7351-4129-90ed-011a7577acd6' was started on site default
[22:10:14] [Asr] was asked to listen on site default
[22:10:18] [Asr] captured text "allume la lumière du bureau" in 3.0s
[22:10:18] [Asr] was asked to stop listening on site default
[22:10:18] [Nlu] was asked to parse input "allume la lumière du bureau"
[22:10:18] [Nlu] detected intent valf:lightsSetJeedom with probability 
0.922 for input "allume la lumière du bureau"
               Slots ->
                  house_room -> bureau (confidence: 1.000)
[22:10:18] [Dialogue] New intent detected valf:lightsSetJeedom with 
probability 0.922
               Slots ->
                  house_room -> bureau (confidence: 1.000)
[22:10:23] [Dialogue] session with id 
'959faf5d-7351-4129-90ed-011a7577acd6' was ended on site default. The 
session was ended because one of the component didn't respond in a 
timely manner
[22:10:23] [Asr] was asked to stop listening on site default
[22:10:23] [Hotword] was asked to toggle itself 'on' on site default