project-alice-assistant/HermesLedControl #14: (code=exited, status=1/FAILURE)

martinwallasch, 2018-11-12:
Hey Psychokiller1888,
my system is a raspberry pi3 with snips and matrixvoice.
The first issue was: "python[1283]: "ImportError: No module named zmq"
Then I installed the module.
And now I have the following problem:
`pi@raspberrypi:~ $ sudo service snipsledcontrol status
● snipsledcontrol.service - Snips Led Control
   Loaded: loaded (/etc/systemd/system/snipsledcontrol.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2018-11-12 20:17:09 CET; 1s ago
  Process: 1180 ExecStart=/usr/bin/python main.py --hardware=matrixvoice --pattern=google (code=exited, status
 Main PID: 1180 (code=exited, status=1/FAILURE)

Nov 12 20:17:09 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.`

what am I doing wrong? 
best regards

Psychokiller1888, 2018-11-12:
If you didn't have zmq that means you didn't install your device. Device auto installation is available only on the master branch and should release in the next few days.

For now, you can launch the installer manually, it is in installers/matrixvoice.sh

If not working, provide better logs, this is the service log only. `tail -500 /var/log/syslog`

Psychokiller1888, 2018-11-13:
https://github.com/Psychokiller1888/snipsLedControl/blob/master/installers/matrixvoice.sh

martinwallasch, 2018-11-13:
Thank you for the fast feedback. I will test it tonight.

Psychokiller1888, 2018-11-13:
I just release 1.4, so you might want to just reinstall following the same install procedure

martinwallasch, 2018-11-13:
Hello,
unfortunately both steps do not work for me.

```pi@raspberrypi:~ $ sudo service snipsledcontrol status
● snipsledcontrol.service - Snips Led Control
   Loaded: loaded (/etc/systemd/system/snipsledcontrol.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-11-13 19:18:01 CET; 3s ago
  Process: 2175 ExecStart=/usr/bin/python main.py --hardware=matrixvoice --pattern=google (code=exited, status=1/FAILURE)
 Main PID: 2175 (code=exited, status=1/FAILURE)
```
The log looks like this:
```Nov 13 19:22:36 raspberrypi python[2456]: Initializing leds controller
Nov 13 19:22:36 raspberrypi python[2456]: Traceback (most recent call last):
Nov 13 19:22:36 raspberrypi python[2456]:   File "main.py", line 103, in <module>
Nov 13 19:22:36 raspberrypi python[2456]:     main()
Nov 13 19:22:36 raspberrypi python[2456]:   File "main.py", line 89, in main
Nov 13 19:22:36 raspberrypi python[2456]:     slc = SnipsLedControl(args)
Nov 13 19:22:36 raspberrypi python[2456]:   File "/home/pi/snipsLedControl_v1.4/models/SnipsLedControl.py", line 114, in __init__
Nov 13 19:22:36 raspberrypi python[2456]:     self._ledsController = LedsController(self)
Nov 13 19:22:36 raspberrypi python[2456]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 56, in __init__
Nov 13 19:22:36 raspberrypi python[2456]:     if not self.initHardware():
Nov 13 19:22:36 raspberrypi python[2456]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 108, in initHardware
Nov 13 19:22:36 raspberrypi python[2456]:     from interfaces.matrixvoice import MatrixVoice
Nov 13 19:22:36 raspberrypi python[2456]:   File "/home/pi/snipsLedControl_v1.4/interfaces/matrixvoice.py", line 5, in <module>
Nov 13 19:22:36 raspberrypi python[2456]:     from libraries.everloop import Everloop
Nov 13 19:22:36 raspberrypi python[2456]:   File "/home/pi/snipsLedControl_v1.4/libraries/everloop.py", line 6, in <module>
Nov 13 19:22:36 raspberrypi python[2456]:     import zmq
Nov 13 19:22:36 raspberrypi python[2456]: ImportError: No module named zmq
Nov 13 19:22:36 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 19:22:36 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 19:22:36 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
Nov 13 19:22:41 raspberrypi systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
Nov 13 19:22:41 raspberrypi systemd[1]: Stopped Snips Led Control.
Nov 13 19:22:41 raspberrypi systemd[1]: Started Snips Led Control.
Nov 13 19:22:41 raspberrypi python[2461]: Starting Snips Led Control
Nov 13 19:22:41 raspberrypi python[2461]: Initializing SnipsLedControl
Nov 13 19:22:41 raspberrypi python[2461]: Loading configurations
Nov 13 19:22:41 raspberrypi python[2461]: Loaded 6 hardware references
Nov 13 19:22:41 raspberrypi python[2461]: - Mqtt server set to localhost
Nov 13 19:22:41 raspberrypi python[2461]: - Mqtt port set to 1883
Nov 13 19:22:41 raspberrypi python[2461]: - Client id set to default
Nov 13 19:22:41 raspberrypi python[2461]: - Hardware set to Matrix Voice
Nov 13 19:22:41 raspberrypi python[2461]: - Using google as pattern with 18 leds
Nov 13 19:22:41 raspberrypi python[2461]: Initializing leds controller
Nov 13 19:22:41 raspberrypi python[2461]: Traceback (most recent call last):
Nov 13 19:22:41 raspberrypi python[2461]:   File "main.py", line 103, in <module>
Nov 13 19:22:41 raspberrypi python[2461]:     main()
Nov 13 19:22:41 raspberrypi python[2461]:   File "main.py", line 89, in main
Nov 13 19:22:41 raspberrypi python[2461]:     slc = SnipsLedControl(args)
Nov 13 19:22:41 raspberrypi python[2461]:   File "/home/pi/snipsLedControl_v1.4/models/SnipsLedControl.py", line 114, in __init__
Nov 13 19:22:41 raspberrypi python[2461]:     self._ledsController = LedsController(self)
Nov 13 19:22:41 raspberrypi python[2461]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 56, in __init__
Nov 13 19:22:41 raspberrypi python[2461]:     if not self.initHardware():
Nov 13 19:22:41 raspberrypi python[2461]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 108, in initHardware
Nov 13 19:22:41 raspberrypi python[2461]:     from interfaces.matrixvoice import MatrixVoice
Nov 13 19:22:41 raspberrypi python[2461]:   File "/home/pi/snipsLedControl_v1.4/interfaces/matrixvoice.py", line 5, in <module>
Nov 13 19:22:41 raspberrypi python[2461]:     from libraries.everloop import Everloop
Nov 13 19:22:41 raspberrypi python[2461]:   File "/home/pi/snipsLedControl_v1.4/libraries/everloop.py", line 6, in <module>
Nov 13 19:22:41 raspberrypi python[2461]:     import zmq
Nov 13 19:22:41 raspberrypi python[2461]: ImportError: No module named zmq
Nov 13 19:22:41 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 19:22:41 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 19:22:41 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
Nov 13 19:22:46 raspberrypi systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
Nov 13 19:22:46 raspberrypi systemd[1]: Stopped Snips Led Control.
Nov 13 19:22:46 raspberrypi systemd[1]: Started Snips Led Control.
Nov 13 19:22:47 raspberrypi python[2466]: Starting Snips Led Control
[... cut ...]
Nov 13 19:23:58 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 19:23:58 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 19:23:58 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
```
The following error occurred during installation:
"notebook 5.7.0 has requirement pyzmq>=17"

I have updated by the following command:
"pip install 'pyzmq>17"

then the installation went through and there were no problems.

Thank you in advance for your help

Psychokiller1888, 2018-11-13:
To me it looks like you didn't install by the matrixvoice.sh installer

Let's clean up, and start over, because, it just goes very smoothly to install normally

```
sudo systemctl stop snipsledcontrol
cd /home/pi
sudo rm -rf matrixvoice
sudo rm -rf snipsLedControl_v1.4
wget https://gist.githubusercontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/e24882e8997730dcf7a308e303b3b88001dbbfa1/slc_download.sh
sudo chmod +x slc_download.sh
sudo ./slc_download.sh
```

Follow the instructions, say you want to install your matrix voice

martinwallasch, 2018-11-13:
no change

```pi@raspberrypi:~ $ sudo service snipsledcontrol status
● snipsledcontrol.service - Snips Led Control
   Loaded: loaded (/etc/systemd/system/snipsledcontrol.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-11-13 20:47:51 CET; 4s ago
  Process: 8468 ExecStart=/usr/bin/python main.py --hardware=matrixvoice --pattern=google (code=exited, status=1/FAILURE)
 Main PID: 8468 (code=exited, status=1/FAILURE)

Nov 13 20:47:51 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 20:47:51 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
```
LOG:
```
pi@raspberrypi:~ $ tail -500 /var/log/syslog
Nov 13 20:50:30 raspberrypi python[8630]: Initializing leds controller
Nov 13 20:50:30 raspberrypi python[8630]: Traceback (most recent call last):
Nov 13 20:50:30 raspberrypi python[8630]:   File "main.py", line 103, in <module>
Nov 13 20:50:30 raspberrypi python[8630]:     main()
Nov 13 20:50:30 raspberrypi python[8630]:   File "main.py", line 89, in main
Nov 13 20:50:30 raspberrypi python[8630]:     slc = SnipsLedControl(args)
Nov 13 20:50:30 raspberrypi python[8630]:   File "/home/pi/snipsLedControl_v1.4/models/SnipsLedControl.py", line 114, in __init__
Nov 13 20:50:30 raspberrypi python[8630]:     self._ledsController = LedsController(self)
Nov 13 20:50:30 raspberrypi python[8630]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 56, in __init__
Nov 13 20:50:30 raspberrypi python[8630]:     if not self.initHardware():
Nov 13 20:50:30 raspberrypi python[8630]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 108, in initHardware
Nov 13 20:50:30 raspberrypi python[8630]:     from interfaces.matrixvoice import MatrixVoice
Nov 13 20:50:30 raspberrypi python[8630]:   File "/home/pi/snipsLedControl_v1.4/interfaces/matrixvoice.py", line 5, in <module>
Nov 13 20:50:30 raspberrypi python[8630]:     from libraries.everloop import Everloop
Nov 13 20:50:30 raspberrypi python[8630]:   File "/home/pi/snipsLedControl_v1.4/libraries/everloop.py", line 6, in <module>
Nov 13 20:50:31 raspberrypi python[8630]:     import zmq
Nov 13 20:50:31 raspberrypi python[8630]: ImportError: No module named zmq
Nov 13 20:50:31 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 20:50:31 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 20:50:31 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
Nov 13 20:50:36 raspberrypi systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
Nov 13 20:50:36 raspberrypi systemd[1]: Stopped Snips Led Control.
Nov 13 20:50:36 raspberrypi systemd[1]: Started Snips Led Control.
Nov 13 20:50:36 raspberrypi python[8635]: Starting Snips Led Control
Nov 13 20:50:36 raspberrypi python[8635]: Initializing SnipsLedControl
Nov 13 20:50:36 raspberrypi python[8635]: Loading configurations
Nov 13 20:50:36 raspberrypi python[8635]: Loaded 6 hardware references
Nov 13 20:50:36 raspberrypi python[8635]: - Mqtt server set to localhost
Nov 13 20:50:36 raspberrypi python[8635]: - Mqtt port set to 1883
Nov 13 20:50:36 raspberrypi python[8635]: - Client id set to default
Nov 13 20:50:36 raspberrypi python[8635]: - Hardware set to Matrix Voice
Nov 13 20:50:36 raspberrypi python[8635]: - Using google as pattern with 18 leds
Nov 13 20:50:36 raspberrypi python[8635]: Initializing leds controller
Nov 13 20:50:36 raspberrypi python[8635]: Traceback (most recent call last):
Nov 13 20:50:36 raspberrypi python[8635]:   File "main.py", line 103, in <module>
Nov 13 20:50:36 raspberrypi python[8635]:     main()
Nov 13 20:50:36 raspberrypi python[8635]:   File "main.py", line 89, in main
Nov 13 20:50:36 raspberrypi python[8635]:     slc = SnipsLedControl(args)
Nov 13 20:50:36 raspberrypi python[8635]:   File "/home/pi/snipsLedControl_v1.4/models/SnipsLedControl.py", line 114, in __init__
Nov 13 20:50:36 raspberrypi python[8635]:     self._ledsController = LedsController(self)
Nov 13 20:50:36 raspberrypi python[8635]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 56, in __init__
Nov 13 20:50:36 raspberrypi python[8635]:     if not self.initHardware():
Nov 13 20:50:36 raspberrypi python[8635]:   File "/home/pi/snipsLedControl_v1.4/models/LedsController.py", line 108, in initHardware
Nov 13 20:50:36 raspberrypi python[8635]:     from interfaces.matrixvoice import MatrixVoice
Nov 13 20:50:36 raspberrypi python[8635]:   File "/home/pi/snipsLedControl_v1.4/interfaces/matrixvoice.py", line 5, in <module>
Nov 13 20:50:36 raspberrypi python[8635]:     from libraries.everloop import Everloop
Nov 13 20:50:36 raspberrypi python[8635]:   File "/home/pi/snipsLedControl_v1.4/libraries/everloop.py", line 6, in <module>
Nov 13 20:50:36 raspberrypi python[8635]:     import zmq
Nov 13 20:50:36 raspberrypi python[8635]: ImportError: No module named zmq
Nov 13 20:50:36 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 20:50:36 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 20:50:36 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
Nov 13 20:50:41 raspberrypi systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
[... cut ...]
Nov 13 20:51:53 raspberrypi python[8708]:     import zmq
Nov 13 20:51:53 raspberrypi python[8708]: ImportError: No module named zmq
Nov 13 20:51:53 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Nov 13 20:51:53 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Nov 13 20:51:53 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
```

Is it helpful if the system is reinstalling

best regards

Psychokiller1888, 2018-11-14:
Dude, there's no need to post 500 lines of logs each time, the error is clearly to be seen :)

Is the installation process going through well, without error?

You can try this

```
sudo mkdir -p /home/pi/matrixvoice
cd /home/pi/matrixvoice
wget "https://github.com/matrix-io/matrix-creator-malos/blob/master/src/python_test/Pipfile" -O Pipfile
wget "https://github.com/matrix-io/matrix-creator-malos/blob/master/src/python_test/Pipfile.lock" -O Pipfile.lock
wget "https://raw.githubusercontent.com/matrix-io/matrix-creator-malos/master/src/python_test/requirements.txt" -O requirements.txt
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip
sudo pip install -r requirements.txt
sudo rm -rf /home/pi/matrixvoice
```

martinwallasch, 2018-11-14:
hey,
the installation process going through well!
I've gone through the steps from your last post, but the problem remains.

```pi@raspberrypi:~ $ sudo service snipsledcontrol status
● snipsledcontrol.service - Snips Led Control
   Loaded: loaded (/etc/systemd/system/snipsledcontrol.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-11-14 18:44:48 CET; 1s ago
  Process: 1020 ExecStart=/usr/bin/python main.py --hardware=matrixvoice --pattern=google (code=exited, status=1/FAILURE)
 Main PID: 1020 (code=exited, status=1/FAILURE)

Nov 14 18:44:48 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
```
best regards

Psychokiller1888, 2018-11-14:
pip install zmq?

martinwallasch, 2018-11-14:
```Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: zmq in /usr/local/lib/python3.5/dist-packages (0.0.0)
Requirement already satisfied: pyzmq in /usr/local/lib/python3.5/dist-packages (from zmq) (16.0.2)
```

Psychokiller1888, 2018-11-14:
See, this is why the template asks you how you start SLC....

Use python 2

martinwallasch, 2018-11-14:
ok, I will uninstall python3.5 and install python 2

martinwallasch, 2018-11-14:
# update
here there are only errors, I reinstall the system

Psychokiller1888, 2018-11-14:
SLC is now running fine.
The error you see is the snips-skill-server not working

martinwallasch, 2018-11-14:
Thanks for the quick answers ... I will continue tomorrow night, because the alarm clock at 3 ring.
Yes the status of the Snips Skill Server is:
``` sudo systemctl status "snips-*"
● snips-skill-server.service - Snips Skill Server
   Loaded: loaded (/lib/systemd/system/snips-skill-server.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2018-11-14 22:46:30 CET; 942ms ago
  Process: 1162 ExecStart=/usr/bin/snips-skill-server (code=exited, status=1/FAILURE)
 Main PID: 1162 (code=exited, status=1/FAILURE)

Nov 14 22:46:30 raspberrypi systemd[1]: snips-skill-server.service: Unit entered failed state.
Nov 14 22:46:30 raspberrypi systemd[1]: snips-skill-server.service: Failed with result 'exit-code'.```

How can I fix this?

the rest looks good:
```● snips-asr.service - Snips ASR
   Loaded: loaded (/lib/systemd/system/snips-asr.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 593 (snips-asr)
   CGroup: /system.slice/snips-asr.service
           └─593 /usr/bin/snips-asr

Nov 14 22:43:42 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0       entered AsrRunner::run
Nov 14 22:43:42 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0+0.000 capture started
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0+2.867 endpoint detected (rule:4) frame:168 samples:43008 sig
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : Source thread stop on push: "Disconnected(..)"
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0+2.882 capture ended
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0+2.897 decoder finalized
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_lib::asr        : T0+2.903 lookup and post-processing done
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_hermes          : Cleanup
Nov 14 22:43:45 raspberrypi snips-asr[593]: INFO:snips_asr_hermes          : Preparing decoder
Nov 14 22:43:46 raspberrypi snips-asr[593]: INFO:snips_asr_hermes          : Idle

● snips-nlu.service - Snips NLU
   Loaded: loaded (/lib/systemd/system/snips-nlu.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 599 (snips-nlu)
   CGroup: /system.slice/snips-nlu.service
           └─599 /usr/bin/snips-nlu

Nov 14 22:24:46 raspberrypi systemd[1]: Started Snips NLU.
Nov 14 22:24:48 raspberrypi snips-nlu[599]: INFO:snips_nlu_hermes: loading nlu engine "/usr/share/snips/assistant/nlu_engine"
Nov 14 22:24:50 raspberrypi snips-nlu[599]: INFO:snips_nlu_hermes: model loaded in 2481 ms
Nov 14 22:38:22 raspberrypi snips-nlu[599]: INFO:snips_nlu_hermes: reload nlu engine event received
Nov 14 22:38:23 raspberrypi snips-nlu[599]: INFO:snips_nlu_hermes: loading nlu engine "/var/lib/snips/injections/20181114T213333291042014/in
Nov 14 22:38:24 raspberrypi snips-nlu[599]: INFO:snips_nlu_hermes: model loaded in 1691 ms

● snips-dialogue.service - Snips Dialogue
   Loaded: loaded (/lib/systemd/system/snips-dialogue.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 605 (snips-dialogue)
   CGroup: /system.slice/snips-dialogue.service
           └─605 /usr/bin/snips-dialogue

Nov 14 22:41:33 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Session [730dee65-a8c8-497a-b522-b135b60
Nov 14 22:41:46 raspberrypi snips-dialogue[605]: WARN:snips_dialogue_lib::coordinator::coordinator: No Session found for this response: Audi
Nov 14 22:41:46 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Session [730dee65-a8c8-497a-b522-b135b60
Nov 14 22:41:46 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Site [default]: reset site -> turn off A
Nov 14 22:43:42 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Site [default]: reset site -> turn off A
Nov 14 22:43:42 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Session [fea43193-3f04-4df5-96e2-5183537
Nov 14 22:43:47 raspberrypi snips-dialogue[605]: WARN:snips_dialogue_lib::coordinator::coordinator: No Session found for this response: NluI
Nov 14 22:43:48 raspberrypi snips-dialogue[605]: WARN:snips_dialogue_lib::coordinator::coordinator: No Session found for this response: Audi
Nov 14 22:43:48 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Session [fea43193-3f04-4df5-96e2-5183537
Nov 14 22:43:48 raspberrypi snips-dialogue[605]: INFO:snips_dialogue_lib::coordinator::coordinator: Site [default]: reset site -> turn off A

● snips-injection.service - Snips Injection
   Loaded: loaded (/lib/systemd/system/snips-injection.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 604 (snips-injection)
   CGroup: /system.slice/snips-injection.service
[... 31 lines cut ...]
● snips-audio-server.service - Snips Audio Server
   Loaded: loaded (/lib/systemd/system/snips-audio-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 600 (snips-audio-ser)
   CGroup: /system.slice/snips-audio-server.service
[... cut ...]
Nov 14 22:43:47 raspberrypi snips-audio-server[600]: INFO:snips_audio_portaudio      : Playing "1d5f3671-ac72-47e8-86e2-e25c5c4ce905" using
Nov 14 22:43:48 raspberrypi snips-audio-server[600]: INFO:snips_audio_portaudio      : Playing of "1d5f3671-ac72-47e8-86e2-e25c5c4ce905" fin

● snips-tts.service - Snips TTS
   Loaded: loaded (/lib/systemd/system/snips-tts.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 22:24:46 CET; 25min ago
 Main PID: 606 (snips-tts)
   CGroup: /system.slice/snips-tts.service
           └─606 /usr/bin/snips-tts
```

Psychokiller1888, 2018-11-14:
I don't know, I don't use snips skill server. Seek help on snips forum

closed 2018-11-14 by Psychokiller1888