project-alice-assistant/HermesLedControl #15: Respeaker Core v2

jenskastensson, 2018-11-26:
**Respeaker Core v2**
The full name of the device you would like to see supported by SLC

**Website**
http://wiki.seeedstudio.com/ReSpeaker_Core_v2.0

Looks great , but I have aunsupported device it looks like, do you plan to support the Reaspeaker Core v2?

Psychokiller1888, 2018-11-26:
I don't have any device myself, I can have a look though

Psychokiller1888, 2018-11-26:
Found documentation on the bus they use. I'll give shot at implementing it!

jenskastensson, 2018-11-26:
Thanks in advance!

Psychokiller1888, 2018-12-02:
Didn't forget you, but I never got any reply back from Seeed (Respeaker makers) so I'm waiting. And I won't buy any respeaker core v2

jenskastensson, 2018-12-02:
Let me try to escalate the support request, can you send your question to
Jens at ProKNX dot com , please?

Psychokiller1888, 2018-12-02:
There's no real question, they should help me implementing all their hardware. They said, through email, privately, that they would and forked the repo 3 weeks ago and never heard back from them, as usual, and of course the repo fork did not move an inch.

jenskastensson, 2018-12-02:
Would it help you if I lend you one of my cards?

Psychokiller1888, 2018-12-02:
Sure. But first of all, do you code? I mean, I can do this distantly over Discord if you use git and can open a few files and search/replace/comment/print

jenskastensson, 2018-12-03:
Absolutely, that sounds like the solution. Tell me what to do!

Psychokiller1888, 2018-12-04:
Actually, could you try to start using respeaker 4 interface?

python main.py --hardware=respeaker4

jenskastensson, 2018-12-12:
Hi and thanks for the reply!
Prior to installing SLD I already installed the SEED  pixel_ring_server. It
is not running right now but maybe it can cause some issues?

When I run SLD I get the output below, not sure what goes wrong:

respeaker@v2:/home/pi/snipsLedControl_v1.5$ sudo  /usr/bin/python main.py
--hardware=respeaker4
Starting Snips Led Control
Initializing SnipsLedControl
Loading configurations
Loaded 8 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker 4
- Using google as pattern with 12 leds
Initializing leds controller
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452:
PinFactoryFallback: Falling back from rpigpio: This module can only be run
on a Raspberry Pi!
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452:
PinFactoryFallback: Falling back from rpio: No module named RPIO
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452:
PinFactoryFallback: Falling back from pigpio: No module n amed pigpio
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452:
PinFactoryFallback: Falling back from native: unable to determine RPi
revision
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "main.py", line 113, in <module>
    main()
  File "main.py", line 99, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.5/models/SnipsLedControl.py", line 121,
in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl_v1.5/models/LedsController.py", line 60,
in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl_v1.5/models/LedsController.py", line 100,
in initHardware
    from interfaces.apa102 import APA102
  File "/home/pi/snipsLedControl_v1.5/interfaces/apa102.py", line 4, in
<module>
    from gpiozero                       import LED
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/__init__.py", line
22, in <module>
    from .devices import (
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py", line
486, in <module>
    Device.pin_factory = _default_pin_factory()
  File "/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py", line
453, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
respeaker@v2:/home/pi/snipsLedControl_v1.5$

Psychokiller1888, 2018-12-12:
The error says rpigpio runs only on a raspberry. Are you trying to run SLC on the respeaker core directly?

jenskastensson, 2018-12-12:
Yes, I'm trying to run it on a Respeaker Core v2.

Psychokiller1888, 2018-12-12:
Well, I can't say how to get it to work, I don't have the respeaker core and I don't know what it runs or not. I know it can use the same apa102 interface, but obviously it can't run gpiozero. Seeed studio doesn't answer anymore so well, f*** their products :)

jenskastensson, 2018-12-13:
How about I try to get it running with your support? I'm motivated because
from what I can see from your documentation the SLD looks very useful! If
you think this is worth a shot, we can (if you prefer) communicate directly
e.g telegram @jenskastensson

Psychokiller1888, 2018-12-18:
Sure we can communicate directly, on Discord, through private messages

Psychokiller1888, 2019-01-09:
V1.6 implemented core v2 already