project-alice-assistant/HermesLedControl #16: how to use --gpioPin without any hardware define

dony71, 2018-12-01:
I don't have any hardware define in hardware support(Respeaker 2, Respeaker 4, Respeaker Mic Array V2, NeoPixels ring, Matrix Voice), instead I'm using USB microphone + 2 LEDs connected to GPIO 
How to use --gpioPin only? Default --hardware=respeaker2, I cannot assign --hardware=none
Any example?

Psychokiller1888, 2018-12-01:
You can't, as there's no support for direct gpio. But you can make the interface

dony71, 2018-12-01:
Any plan for you to make this interface?
I'm not good in programming

Psychokiller1888, 2018-12-02:
I can do it, yes

Psychokiller1888, 2018-12-03:
I have started implementing something, on dev 1.5 branch. Should be usable by this evening

Psychokiller1888, 2018-12-03:
Version 1.5 released with a working setup for gpio leds

Use arguments `--hardware=puregpio --pureGpioPinout=[YOUR, LEDS, PIN, NUMBERS]`

By default it's wired as cathode to ground and anode to resistor and gpio. You can invert it if you want but use the `--activeHigh=False` argument

closed 2018-12-03 by Psychokiller1888