project-alice-assistant/HermesLedControl #17: Matrix Voice IndexError: list assignment index out of range

jbperrin88, 2018-12-13:
Hi there !

First of all , thanks you for all the stuff you give to snips community !! That amazing ! 

Like a tell you on discord yesterday , i got some strange behavior with snips led control and matrix voice 

First log with the initial error 
https://pastebin.com/HuaV0rKD

I've made the modification you give to me
` def setPixel(self, ledNum, red, green, blue, white):
        ledValue = io_pb2.LedValue()
        ledValue.red   = red
        ledValue.green = green
        ledValue.blue  = blue
        ledValue.white = white
        print(ledNum)
        print(self._numLeds)
        if ledNum < len(self._colors):
            self._colors[ledNum] = ledValue
        else:
            self._logger.warning("Tried to access a led number that doesn't exist")`

Result with google pattern :
https://pastebin.com/WecihHyX
Look like the num_leds is multiply by 2 somewhere .... 

Result with alexa pattern : 
https://pastebin.com/ddNLfF4a

I remove some line from both alexa and google export (pastebin failed without removing it ...)

If you need some help, please ask !

Psychokiller1888, 2018-12-20:
By looking at the timestamps, I take it as the wakeup animation right after start works, but it fails when you try to call your assistant?

jbperrin88, 2018-12-20:
Nope ... The startup annulation looks good . It is the wakeup pattern just after hotwire is detected

jbperrin88, 2018-12-20:
Hotword

Psychokiller1888, 2018-12-20:
Yeah, that's what I said...

jbperrin88, 2018-12-20:
Hum .... Sorrrrrry I've read to fast and I'm French haha fail ! 

Thanks you .

I'll try soon. ;B

commit 9de6b895 (2018-12-26): Trying to fix #17
closed 2018-12-26 by Psychokiller1888