project-alice-assistant/HermesLedControl #21: ReSpeaker v2 not working, Initializing Led controler failed

fluffykraken, 2018-12-27:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
sudo python main.py --pattern=kiboost  --hardware=respeakerMicArrayV2

**Led device name**
ReSpeaker MicArray V2

**The bug**

 sudo python main.py --pattern=kiboost  --hardware=respeakerMicArrayV2
Starting Snips Led Control v. 1.6.1
Initializing SnipsLedControl
Loading configurations
Loaded 9 hardware references
- Mqtt server set to 192.168.1.43
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker Mic array v2
- Using kiboost as pattern with 12 leds
Initializing leds controller
Traceback (most recent call last):
  File "main.py", line 116, in <module>
    main()
  File "main.py", line 102, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.6.1/models/SnipsLedControl.py", line 121, in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl_v1.6.1/models/LedsController.py", line 50, in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl_v1.6.1/models/LedsController.py", line 106, in initHardware
    self._interface = APA102(numLed=self._hardware['numberOfLeds'])
  File "/home/pi/snipsLedControl_v1.6.1/interfaces/apa102.py", line 16, in __init__
    self._leds  = AAPA102(numLed, global_brightness=global_brightness, order=order, bus=bus, device=device, max_speed_hz=max_speed_hz)
  File "/home/pi/snipsLedControl_v1.6.1/libraries/apa102.py", line 90, in __init__
    self.spi.open(bus, device)  # Open SPI port 0, slave device (CS) 1
IOError: [Errno 2] No such file or directory

**To Reproduce**
Fresh install SLC
Hardware: Select Option 3
Pattern: Google
Install reSpeaker?: No

fluffykraken, 2018-12-27:
This is the Log after a fresch install.....
2018-12-27 18:09:33,412 [MainThread] - [INFO] - Starting Snips Led Control v. 1.6.1
2018-12-27 18:09:33,424 [MainThread] - [INFO] - Initializing SnipsLedControl
2018-12-27 18:09:33,426 [MainThread] - [INFO] - Loading configurations
2018-12-27 18:09:33,431 [MainThread] - [INFO] - Loaded 9 hardware references
2018-12-27 18:09:33,432 [MainThread] - [INFO] - - Mqtt server set to 192.168.1.43
2018-12-27 18:09:33,433 [MainThread] - [INFO] - - Mqtt port set to 1883
2018-12-27 18:09:33,434 [MainThread] - [INFO] - - Client id set to default
2018-12-27 18:09:33,434 [MainThread] - [INFO] - - Hardware set to ReSpeaker Mic array v2
2018-12-27 18:09:33,435 [MainThread] - [INFO] - - Using google as pattern with 12 leds
2018-12-27 18:09:33,435 [MainThread] - [INFO] - Initializing leds controller

Psychokiller1888, 2018-12-27:
SPI is not enabled

fluffykraken, 2018-12-27:
You are rigth, I was not aware that I would need to enable them, as I am not using that interface. My ReSpeaker is connected via USB.
The exception is gone now, but the LEDs wont turn on. 
No error in logs.

fluffykraken, 2018-12-28:
I've tried to enable a custom hotword as described here : https://github.com/Psychokiller1888/snipsLedControl/issues/18
but unfortunatly no luck.......
Please let me know what kind of information you would need to make this work.
And thanks for all your effort and work you put into this! It is great to see a continous progress on this component....

Psychokiller1888, 2018-12-28:
Rerun the install and chose to install your respeaker, or check this: https://github.com/Psychokiller1888/snipsLedControl/blob/master/installers/respeakerMicArrayV2.sh and run the commands manually. I think it's not installed

fluffykraken, 2018-12-28:
I did this allready with no luck. The SeedWiki Demo runs fine (https://github.com/respeaker/pixel_ring.git), and also my SLC Version form October works with my customLedPattern. The firmware on my respeaker is up to date.

fluffykraken, 2018-12-28:
I've just tested all available versions to root out the problem and the last working version is 1.5.
Allthough getting the "Trying to access a led index out of reach" error it is working as expected. Maybe that helps to locate the problem.

Psychokiller1888, 2019-01-21:
Sorry that took so long, got out of my head and I was busy at work. Anyway, the Trying to access a led out of reach error comes from the pattern. What pattern are you using? If custom, make sure your leds are 0 indexed

fluffykraken, 2019-01-24:
I was using the provided google pattern with no modifaication at all. 
Will try to install v1.7 and report back

fluffykraken, 2019-01-27:
Just didi a fresh install on my pi3 B+. While installing SLC I selcted to install my respeaker micarray v2 the following errors were shown:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package pyusb
E: Unable to locate package click
Cloning into 'usb_4_mic_array'...
remote: Enumerating objects: 59, done.
remote: Total 59 (delta 0), reused 0 (delta 0), pack-reused 59
Unpacking objects: 100% (59/59), done.
Traceback (most recent call last):
  File "dfu.py", line 15, in <module>
    import usb.core
ImportError: No module named usb.core

```
So I manually installed those using

`sudo pip install pyusb click`

Still no luck...

I installed the seed demo pixel_ring..

```
git clone https://github.com/respeaker/pixel_ring.git
cd pixel_ring
sudo python setup.py install
sudo python examples/usb_mic_array.py
```

This works falwlessly, and takes over control of the LEDs

reinstalled SLC and it is not woking. SLC is no taking control over the LEDs. The respeaker LEDs are still in theis default state, where they light up blue and indicate the direction of sound.

fluffykraken, 2019-01-27:
Update: changed the interface in hardware.json again from apa102 to respeakerMicArrayV2 and SLC is able to take controle over LEDs. 
Everything works, also no LED indes out of reach.

```
"respeakerMicArrayV2": {
		"name": "ReSpeaker Mic array v2",
		"numberOfLeds": 12,
		"interface": "respeakerMicArrayV2",
		"vid": "0x2886",
		"pid": "0x0018",
		"endFrame": "0"

```

Psychokiller1888, 2019-01-27:
Ok, thx for reporting. The wrong interface is fixed on the 1.7.1 branch

closed 2019-01-27 by Psychokiller1888