project-alice-assistant/HermesLedControl #22: Trouble getting Matrix Voice to work.

davesilva, 2019-01-05:
**How I did start SLC, with what arguments**
`python main.py --hardware=matrixvoice --pattern=google --debug True`

**Led device name**
Matrix Voice

**The bug**
When I start SLC, it doesn't crash or give an error, but it appears to get stuck. Here's the output:

```
Starting Snips Led Control v. 1.6.1
Initializing SnipsLedControl
Loading configurations
Loaded 9 hardware references
- Mqtt server set to 192.168.1.8
- Mqtt port set to 1883
- Client id set to living-room
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
```
...and after that I don't see any more log messages, and I can't get it to respond to any MQTT messages.

**To Reproduce**
Install SLC using the installer with Matrix Voice selected, then start SLC and publish an MQTT message like:

```
mosquitto_pub -h 192.168.1.8 -t "hermes/leds/toggleOn" -m '{"siteId":"living-room"}'
```

Psychokiller1888, 2019-01-06:
You're missing a "=" for --debug=True

davesilva, 2019-01-06:
Oh sorry, you're right. I actually tried it both ways and just posted the wrong one. I see the same output with the "=".

```
pi@snips-living-room:~/snipsLedControl_v1.6.1 $ sudo /usr/bin/python main.py --hardware=matrixvoice --pattern=google --debug=True
Starting Snips Led Control v. 1.6.1
Initializing SnipsLedControl
Loading configurations
Loaded 9 hardware references
- Mqtt server set to 192.168.1.8
- Mqtt port set to 1883
- Client id set to living-room
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
```

Psychokiller1888, 2019-01-06:
So if you don't see that message `On leds toggle on triggered` with the debug option turned on, your mqtt messages are not sent to the correct place. Where does the 192.168.1.8 come from? You've set it in the snips.toml file, but is that address correct? Shouldn't it be 127.0.0.1?

davesilva, 2019-01-06:
Yes, the address is correct. I'm running an external MQTT broker (mosquitto) at 192.168.1.8. I run several things in the house off of it so I know that it's working. And Snips is definitely using it because I can publish 
```
mosquitto_pub -h 192.168.1.8 -t 'hermes/tts/say' -m '{"text": "Hello", "lang": "en-GB", "siteId": "living-room"}'
``` 
and hear it say "Hello".

davesilva, 2019-01-07:
Here's some more information that might be helpful. I tried starting SLC with just `/usr/bin/python main.py --debug=True` and the script started up successfully:
```
Starting Snips Led Control v. 1.6.1
Initializing SnipsLedControl
Loading configurations
Loaded 9 hardware references
- Mqtt server set to 192.168.1.8
- Mqtt port set to 1883
- Client id set to living-room
- Hardware set to ReSpeaker 2
- Using google as pattern with 3 leds
Initializing leds controller
Snips Led Control started
```
...and then when I ran
```
mosquitto_pub -h 192.168.1.8 -t "hermes/leds/toggleOn" -m '{"siteId":"living-room"}'
```
I saw
```
On leds toggle on triggered
```

So the MQTT connection is definitely working. I'm noticing though that when I start it with `--hardware=matrixvoice`, it never says "Snips Led Control started", and I can't kill it with `^C`. It just seems to hang forever after saying "Connected to data publisher with port 20024".

Psychokiller1888, 2019-01-07:
Did you install your matrix voice when installing SLC?

davesilva, 2019-01-07:
Yes I did. I initially made the mistake of not installing it and that resulted in an exception due to missing libraries. Once I went back and installed it the exceptions went away but I got stuck at this hanging issue.

Chrissiwai, 2019-01-12:
I have the exact same problem with my Matrix Voice. The debug is exactly the same. 

Jan 12 19:18:24 raspberrypi python[854]: Initializing SnipsLedControl
Jan 12 19:18:24 raspberrypi python[854]: Loading configurations
Jan 12 19:18:24 raspberrypi python[854]: Loaded 9 hardware references
Jan 12 19:18:24 raspberrypi python[854]: - Mqtt server set to localhost
Jan 12 19:18:24 raspberrypi python[854]: - Mqtt port set to 1883
Jan 12 19:18:24 raspberrypi python[854]: - Client id set to default
Jan 12 19:18:24 raspberrypi python[854]: - Hardware set to Matrix Voice
Jan 12 19:18:24 raspberrypi python[854]: - Using alexa as pattern with 18 leds
Jan 12 19:18:24 raspberrypi python[854]: Initializing leds controller
Jan 12 19:18:26 raspberrypi python[854]: Connected to data publisher with port 20024

And if i use hardware=respeaker 2 it says:

Jan 12 19:09:02 raspberrypi python[773]: Initializing SnipsLedControl
Jan 12 19:09:02 raspberrypi python[773]: Loading configurations
Jan 12 19:09:02 raspberrypi python[773]: Loaded 9 hardware references
Jan 12 19:09:02 raspberrypi python[773]: - Mqtt server set to localhost
Jan 12 19:09:02 raspberrypi python[773]: - Mqtt port set to 1883
Jan 12 19:09:02 raspberrypi python[773]: - Client id set to default
Jan 12 19:09:02 raspberrypi python[773]: - Hardware set to ReSpeaker 2
Jan 12 19:09:02 raspberrypi python[773]: - Using alexa as pattern with 3 leds
Jan 12 19:09:02 raspberrypi python[773]: Initializing leds controller
Jan 12 19:09:03 raspberrypi python[773]: Snips Led Control started

Maybe u can help?

Psychokiller1888, 2019-01-13:
Ok, time to look into this

- Are you guys user 'pi'?

Chrissiwai, 2019-01-13:
now i get this SLC Status:

snipsledcontrol.service - Snips Led Control
   Loaded: loaded (/etc/systemd/system/snipsledcontrol.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-01-13 14:58:01 CET; 2s ago
  Process: 959 ExecStart=/usr/bin/python main.py --hardware=matrixvoice --pattern=google (code=exited, status=1/FAILURE)
 Main PID: 959 (code=exited, status=1/FAILURE)

Jan 13 14:58:01 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.

Psychokiller1888, 2019-01-13:
Check your logs?

Chrissiwai, 2019-01-13:
How can i do this?

Psychokiller1888, 2019-01-13:
`tail -n 100 /var/log/syslog`

Chrissiwai, 2019-01-13:
Jan 13 15:01:57 raspberrypi python[1193]: ImportError: No module named zmq
Jan 13 15:01:57 raspberrypi systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 15:01:57 raspberrypi systemd[1]: snipsledcontrol.service: Unit entered failed state.
Jan 13 15:01:57 raspberrypi systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
Jan 13 15:02:02 raspberrypi systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
Jan 13 15:02:02 raspberrypi systemd[1]: Stopped Snips Led Control.
Jan 13 15:02:02 raspberrypi systemd[1]: Started Snips Led Control.
Jan 13 15:02:03 raspberrypi python[1199]: Starting Snips Led Control v. 1.6.1
Jan 13 15:02:03 raspberrypi python[1199]: Initializing SnipsLedControl
Jan 13 15:02:03 raspberrypi python[1199]: Loading configurations
Jan 13 15:02:03 raspberrypi python[1199]: Loaded 9 hardware references
Jan 13 15:02:03 raspberrypi python[1199]: - Mqtt server set to localhost
Jan 13 15:02:03 raspberrypi python[1199]: - Mqtt port set to 1883
Jan 13 15:02:03 raspberrypi python[1199]: - Client id set to default
Jan 13 15:02:03 raspberrypi python[1199]: - Hardware set to Matrix Voice
Jan 13 15:02:03 raspberrypi python[1199]: - Using google as pattern with 18 leds
Jan 13 15:02:03 raspberrypi python[1199]: Initializing leds controller
Jan 13 15:02:03 raspberrypi python[1199]: Traceback (most recent call last):
Jan 13 15:02:03 raspberrypi python[1199]:   File "main.py", line 116, in <module>
Jan 13 15:02:03 raspberrypi python[1199]:     main()
Jan 13 15:02:03 raspberrypi python[1199]:   File "main.py", line 102, in main
Jan 13 15:02:03 raspberrypi python[1199]:     slc = SnipsLedControl(args)
Jan 13 15:02:03 raspberrypi python[1199]:   File "/home/pi/snipsLedControl_v1.6.1/models/SnipsLedControl.py", line 121, in __init__
Jan 13 15:02:03 raspberrypi python[1199]:     self._ledsController = LedsController(self)
Jan 13 15:02:03 raspberrypi python[1199]:   File "/home/pi/snipsLedControl_v1.6.1/models/LedsController.py", line 50, in __init__
Jan 13 15:02:03 raspberrypi python[1199]:     if not self.initHardware():
Jan 13 15:02:03 raspberrypi python[1199]:   File "/home/pi/snipsLedControl_v1.6.1/models/LedsController.py", line 117, in initHardware
Jan 13 15:02:03 raspberrypi python[1199]:     from interfaces.matrixvoice import MatrixVoice
Jan 13 15:02:03 raspberrypi python[1199]:   File "/home/pi/snipsLedControl_v1.6.1/interfaces/matrixvoice.py", line 5, in <module>
Jan 13 15:02:03 raspberrypi python[1199]:     from libraries.everloop import Everloop
Jan 13 15:02:03 raspberrypi python[1199]:   File "/home/pi/snipsLedControl_v1.6.1/libraries/everloop.py", line 6, in <module>
Jan 13 15:02:03 raspberrypi python[1199]:     import zmq

Psychokiller1888, 2019-01-13:
Buddy, the error is clearly written, no module named ZMQ, you don't need to post a wall of logs just for that no?

Psychokiller1888, 2019-01-14:
I don't get it... Gotta get a Matrix Voice for testing. I'm pretty sure it's a package missing. If it's not to be killed it means it's stuck in a loop. Before SLC is started, the only loop is the device has to answer the boot request.

Psychokiller1888, 2019-01-14:
Not to mention I'm sure someone had this problem before and we solved it, but I didn't keep it all in my mind unfortunately

Psychokiller1888, 2019-01-14:
Ok, can you do the following then:

```
curl https://apt.matrix.one/doc/apt-key.gpg | sudo apt-key add -
echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install matrixio-malos
echo "deb http://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/Debian_9.0/ ./" | sudo tee /etc/apt/sources.list.d/zeromq.list
wget https://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/Debian_9.0/Release.key -O- | sudo apt-key add
```

joeduck, 2019-01-14:
Hi,
like i described in Snips Forum it works for me partly. If i can help, write to me.
Sorry for my bad english.

davesilva, 2019-01-15:
@Chrissiwai If you're still having issues, try doing a `pip install zmq` followed by the commands @Psychokiller1888 mentioned above, and then reboot and try again.

Psychokiller1888, 2019-01-15:
Who's up for a clean reinstall? I don't know what fixed the problem here

joeduck, 2019-01-17:
I can do this at weekend. Clean reinstall means new Raspbian, the snips and at last the install script for slc ?