project-alice-assistant/HermesLedControl #23: Respeaker Core v2 not working with version 1.6

hpposch, 2019-01-10:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
sudo python main.py --mqttServer=127.0.0.1 --hardware=respeakerCoreV2 --defaultBrightness=70 --debug h

**Led device name**
respeakerCoreV2

**The bug**
I just do not see the leds flashing i tried to start the script with multiple options like mqttPort but nothing changed.

**To Reproduce**
just execute: sudo python main.py --mqttServer=127.0.0.1 --hardware=respeakerCoreV2 --defaultBrightness=70 --debug h

**DEBUG LOG**
sudo python main.py --mqttServer=127.0.0.1 --hardware=respeakerCoreV2 --defaultBrightness=70 --debug h

Starting Snips Led Control v. 1.6.1
Initializing SnipsLedControl
Loading configurations
Loaded 9 hardware references
- Mqtt server set to 127.0.0.1
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Respeaker Core v2
- Using google as pattern with 12 leds
Initializing leds controller
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpigpio: This module can only be run on a Raspberry Pi!
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from rpio: No module named RPIO
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from pigpio: No module named pigpio
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python2.7/dist-packages/gpiozero/devices.py:452: PinFactoryFallback: Falling back from native: unable to determine RPi revision
  'Falling back from %s: %s' % (name, str(e))))
Device not using gpio zero, ignore power
Snips Led Control started
On hotword triggered
On listen triggered
On think triggered
On say triggered
On hotword toggle on triggered

Psychokiller1888, 2019-01-10:
To me it looks like you haven't run the hardware installation, as gpiozero is uninstalled for respeaker core v2 during SLC install

`pip uninstall -y gpiozero`

hpposch, 2019-01-10:
ok so when i ask to install respeaker core v2 i must select yes?

hpposch, 2019-01-10:
or can you please tell me the steps to enable the leds :)

Psychokiller1888, 2019-01-10:
just run the command I gave you to get rid of gpiozero

hpposch, 2019-01-10:
it says:
pip uninstall -y gpiozero
Cannot uninstall requirement gpiozero, not installed

hpposch, 2019-01-11:
It is working now on my device

closed 2019-01-11 by hpposch

Psychokiller1888, 2019-01-11:
And what did you do?

hpposch, 2019-01-11:
I just followed your instruction.
And today, after a reboot suddenly everything worked.