project-alice-assistant/HermesLedControl #24: ReSpeaker 4-mic Array - can't get led patterns working

denkmic, 2019-01-18:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments:**
ExecStart=/usr/bin/python main.py --hardware=respeaker4 --pattern=alexa

**Led device name:**
ReSpeaker 4

**The bug**
It seems like SLC is working, but it does not react to what Snips is doing. I have a clean Home Assistant installation running (on Hassbian), Snips installed and running, Mosquitto also is running fine.

All is working just fine, but it looks like SLC can't talk with Snips. Here is what I am talking about:

https://vimeo.com/312179822

Might it be due to my Mosquitto settings? I have a password secured mqtt.

**To Reproduce**
Hmmm... not sure, I can only tell you what I did:
1. Clean installation of hassbian
2. Installed Snips using instructions: https://www.home-assistant.io/components/snips/
3. Installed Mosquitto
4. Set up nicely configuration.yaml, snips.toml and mosquitto
5. Installed SLC as described ion your WIKI

Result as you can see on my video. No errors... What am I missing?

Psychokiller1888, 2019-01-19:
Please stop the service and run it manually with debug info:

```sudo systemctl stop snipsledcontrol
cd snipsLedControl_v1.6.1
sudo chown -R pi logs
python main.py --hardware=respeaker4 --pattern=alexa --debug=True
```

And try to trigger your assistant. You'll see messages triggering, or not

Psychokiller1888, 2019-01-19:
Yeah, it's the problem of course. I will add secure connection in a little while, have to finish a couple of things at work, but will do this afternoon

Psychokiller1888, 2019-01-19:
I'll publish 1.7 in a while

closed 2019-01-19 by Psychokiller1888

denkmic, 2019-01-19:
It works like a charm! Thank you buddy!