project-alice-assistant/HermesLedControl #25: Matrix Voice only partially working

joeduck, 2019-01-20:
SLC works with my Matrix Voice only partialy. After first Hotword its always detecting “on hotword”. if i did a new “hey snips” without a question the leds goes of after time.

Also the service does not end. Musst kill the process from another console.
But malos works, because another LED Script from romkabouter does work.

Here the debug:

```
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to 127.0.0.1
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using alexa as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On sucess received, but it wasn't not for me
DEBUG:SnipsLedControl:On sucess received, but it wasn't not for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Led number missmatch (18/18), aborting
WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On sucess received, but it wasn't not for me
DEBUG:SnipsLedControl:On sucess received, but it wasn't not for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
```

Psychokiller1888, 2019-01-21:
Ok, let's try this first:

Open libraries/everloop.py. Change line 76 to fix the missmatch to:

`if ledNum <= self._numLeds:` (notice the 'inferior or equal)

joeduck, 2019-01-21:
OK. Now the are no lights. Also no shutdown possible
Here ist the output:

```
pi@raspberrypi:~ $ cd snipsLedControl_v1.7
pi@raspberrypi:~/snipsLedControl_v1.7 $ python main.py --mqttServer=127.0.0.1 --hardware=matrixvoice --pattern=alexa --debug=True
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to 127.0.0.1
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using alexa as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 340, in _runAnimation
    func()
  File "/home/pi/snipsLedControl_v1.7/ledPatterns/AlexaLedPattern.py", line 28, in wakeup
    self._controller.setLedRGB(self._numLeds - i, self._colors['white'], brightness)
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 360, in setLedRGB
    self.setLed(ledNum, color[0], color[1], color[2], brightness)
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 350, in setLed
    self._interface.setPixel(ledNum, red, green, blue, brightness)
  File "/home/pi/snipsLedControl_v1.7/interfaces/matrixvoice.py", line 20, in setPixel
    self._leds.setPixel(ledNum=ledNum, red=red, green=green, blue=blue, white=brightness)
  File "/home/pi/snipsLedControl_v1.7/libraries/everloop.py", line 77, in setPixel
    self._colors[ledNum] = ledValue
IndexError: list assignment index out of range

On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
^CShutting down Snips Led Control
INFO:SnipsLedControl:Shutting down Snips Led Control
Traceback (most recent call last):
  File "main.py", line 120, in <module>
    main()
  File "main.py", line 116, in main
    slc.onStop()
  File "/home/pi/snipsLedControl_v1.7/models/SnipsLedControl.py", line 151, in onStop
    self._ledsController.onStop()
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 413, in onStop
    self._pattern.onStop()
  File "/home/pi/snipsLedControl_v1.7/models/LedPattern.py", line 45, in onStop
    def onStop(self, *args)             : self.off()
  File "/home/pi/snipsLedControl_v1.7/ledPatterns/AlexaLedPattern.py", line 95, in off
    self._controller.setLedRGB(self._numLeds - i, self._colors['blank'], 0)
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 360, in setLedRGB
    self.setLed(ledNum, color[0], color[1], color[2], brightness)
  File "/home/pi/snipsLedControl_v1.7/models/LedsController.py", line 350, in setLed
    self._interface.setPixel(ledNum, red, green, blue, brightness)
  File "/home/pi/snipsLedControl_v1.7/interfaces/matrixvoice.py", line 20, in setPixel
    self._leds.setPixel(ledNum=ledNum, red=red, green=green, blue=blue, white=brightness)
  File "/home/pi/snipsLedControl_v1.7/libraries/everloop.py", line 77, in setPixel
    self._colors[ledNum] = ledValue
IndexError: list assignment index out of range

```

joeduck, 2019-01-21:
Hi Psycho,

did a new install an saw this error, perhaps this is the reason?:
```
chown: invalid user: ‘user’
chmod: cannot access './installers/matrixvoice.sh': No such file or directory

```

```
pi@raspberrypi:~ $ wget https://gist.githubusercontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/e24882e8997730dcf7a308e303b3b88001dbbfa1/slc_download.sh
--2019-01-21 18:51:11--  https://gist.githubusercontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/e24882e8997730dcf7a308e303b3b88001dbbfa1/slc_download.sh
Resolving gist.githubusercontent.com (gist.githubusercontent.com)... 151.101.16.133
Connecting to gist.githubusercontent.com (gist.githubusercontent.com)|151.101.16.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 499 [text/plain]
Saving to: ‘slc_download.sh’

slc_download.sh                  100%[==========================================================>]     499  --.-KB/s    in 0s

2019-01-21 18:51:12 (4.10 MB/s) - ‘slc_download.sh’ saved [499/499]

pi@raspberrypi:~ $ sudo chmod +x slc_download.sh
pi@raspberrypi:~ $ sudo ./slc_download.sh
rm: cannot remove 'v1.7': No such file or directory
--2019-01-21 18:51:34--  https://api.github.com/repos/Psychokiller1888/snipsLedControl/tarball/v1.7
Resolving api.github.com (api.github.com)... 192.30.253.116, 192.30.253.117
Connecting to api.github.com (api.github.com)|192.30.253.116|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.7 [following]
--2019-01-21 18:51:35--  https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.7
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120, 192.30.253.121
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.7’

v1.7                                 [ <=>                                                       ]  36.51K  --.-KB/s    in 0.1s

2019-01-21 18:51:36 (261 KB/s) - ‘v1.7’ saved [37390]

What device do you wish to control with SLC?
 1) respeaker2
 2) respeaker4
 3) respeakerMicArrayV2
 4) neoPixelsSK6812RGBW
 5) neoPixelsWS2812RGB
 6) matrixvoice
 7) matrixcreator
 8) respeakerCoreV2
 9) respeaker6MicArray
10) googleAIY
11) I'm using simple leds on GPIOs
12) don't overwrite existing parameters
13) cancel
#? 6
What pattern do you want to use?
1) google
2) alexa
3) custom
4) cancel
#? 2
Warning: snipsledcontrol.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Ign:1 http://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/Debian_9.0 ./ InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 http://archive.raspberrypi.org/debian stretch InRelease
Hit:4 http://download.opensuse.org/repositories/network:/messaging:/zeromq:/release-stable/Debian_9.0 ./ Release
Hit:5 https://deb.nodesource.com/node_11.x stretch InRelease
Hit:6 https://apt.matrix.one/raspbian stretch InRelease
Hit:7 https://raspbian.snips.ai/stretch stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
mosquitto is already the newest version (1.4.10-3+deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
mosquitto-clients is already the newest version (1.4.10-3+deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
portaudio19-dev is already the newest version (19.6.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-numpy is already the newest version (1:1.12.1-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages
Requirement already satisfied: spidev in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: gpiozero in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: paho-mqtt in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: pytoml in /usr/local/lib/python2.7/dist-packages
chown: invalid user: ‘user’
chmod: cannot access './installers/matrixvoice.sh': No such file or directory
Do you need to install / configure your matrixvoice?
1) yes
2) no
3) cancel
#? sudo ./slc_download.sh^C
pi@raspberrypi:~ $
```

joeduck, 2019-01-21:
O.k: changed the installername in the install.sh and run it again. Selected "install matrix voice" to (re)install the requirements. 

Did the complete procedure as above. But same result. Sadly it does not work.

Psychokiller1888, 2019-01-21:
I don't get it. Do you also have the array error using google pattern?

joeduck, 2019-01-21:
Yes. Tryed this yesterday.

Psychokiller1888, 2019-01-21:
Ok, forget about my <= fix, revert that. This fixes AlexaPattern

Open patterns/AlexaLedPattern.py and change the wakeup function to this (line 23):

```
	def wakeup(self, direction=0):
		brightness = max(5, self._controller.defaultBrightness - 50)
		for i in range(int(round(self._numLeds / 2)) + 1):
			brightness += 5
			self._controller.setLedRGB(i, self._colors['white'], brightness)

			if i > 0:
				self._controller.setLedRGB(self._numLeds - i, self._colors['white'], brightness)

			if i > 1:
				self._controller.setLedRGB(i - 2, self._colors['blue'], brightness)
				if i > 2:
					self._controller.setLedRGB(self._numLeds - i + 2, self._colors['blue'], brightness)

			self._controller.show()
			time.sleep(0.02)
		time.sleep(0.5)
```

[image]

joeduck, 2019-01-21:
o.k. changed this. But it stays after first wake in speach mode. I always have to kill (HUP) the Process. Did a try with google again.  This is the Log:

```
pi@raspberrypi:~/snipsLedControl_v1.7 $ sudo python main.py --mqttServer=127.0.0.1 --hardware=matrixvoice --pattern=google --debug=True
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to 127.0.0.1
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On sucess received, but it wasn't not for me
DEBUG:SnipsLedControl:On sucess received, but it wasn't not for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
```

Psychokiller1888, 2019-01-21:
Yeah, but no error anymore though

joeduck, 2019-01-21:
Yes, right. I start all services new and do an new try with google pattern.

joeduck, 2019-01-21:
sadly the same:
```
pi@raspberrypi:~/snipsLedControl_v1.7 $ sudo python main.py --mqttServer=127.0.0.1 --hardware=matrixvoice --pattern=google --debug=True
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to 127.0.0.1
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On sucess received, but it wasn't not for me
DEBUG:SnipsLedControl:On sucess received, but it wasn't not for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
```

joeduck, 2019-01-21:
Python ist the right version?
`Python 2.7.13`

Psychokiller1888, 2019-01-22:
Python is ok yes. There's no error and the device is reported as connected. Can you check /va/rlog/syslog for error?

joeduck, 2019-01-23:
Check this this evening.

joeduck, 2019-01-23:
Hi, dont know what to search. looks all good. But SLC recognize the Hotword and after the answer LEDs stays ind "talk mode". When i say the hotword again, it goes to listen mode, when i say nothing the LEDs goes down. 

I have activated the Pulse Audio AEC, can this be the error?
Sould i upload the logfile?

joeduck, 2019-01-24:
Hi. Installed today a satellite with respeaker 2 Mic Hat. Witch this device i have the same issue. Google Pattern. After first wake word and answer SLC keeps blink.

I do my home automation, also the answers, with node red. can i be the sessionid when i send an answer to the device?

Psychokiller1888, 2019-01-25:
I think you have something else, because it works just fine on respeakers. How did you install on the satellite?

joeduck, 2019-01-25:
Sorry, perhaps you misunderstood. The respeaker has the same issue. After time i think this is my fault with the node-red answer per mqtt. I use hermes/tts/say to do the voice output. Perhaps SLC listen for the end of the dialog manager and not for hermes/tts/say/finished?

I think my dialog is not korrekt!?

joeduck, 2019-01-25:
Did the satellite install with your sat tool. great work! Thanks :-)

joeduck, 2019-01-25:
Try now Parameter --offListener

joeduck, 2019-01-25:
Testet again, only with skills with actions. Set to debug and look what is happen. Set "offListener" to "hermes/hotword/toggleOn". Sometimes LED goes off. Sometimes not. But if say only the Hotword, wait and say nothing.... LED goes always off.

Perhaps a issue with the Listener?

```
On hotword toggle on triggered ==>question
On hotword triggered
On listen triggered
On think triggered
On sucess received, but it wasn't not for me
On say triggered ==> action script say answer
On hotword toggle on triggered ==> LED goes not off
On hotword triggered ==>only Hotword without a question
On listen triggered
On think triggered
On hotword toggle on triggered ==> LED goes off
```

Psychokiller1888, 2019-01-25:
`On hotword toggle on triggered` this always triggers the leds off actually. If you see that message, it's linked to idle. Idle in Google and Alexa are leds off. Try to use "--endFrame=0" as an argument please

joeduck, 2019-01-25:
Looks like this an unkown argument: 

`Snips Led Control: error: unrecognized arguments: --endFrame=0`

Psychokiller1888, 2019-01-25:
Ho, it's on 1.7.1, forgot it. Open hardware.json and change endFrame to 0 for respeaker 2

joeduck, 2019-01-25:
Ok. Sadly the same.

Psychokiller1888, 2019-01-25:
gimme a second

Psychokiller1888, 2019-01-25:
What did you install on your satellite?

joeduck, 2019-01-25:
Had set yet in my flow after the "say" mqtt message a "hermes/hotword/toggleOn". then the LEDs goes of. Must optimize the flows/dialog.

Psychokiller1888, 2019-01-25:
Yeah, respeaker 2 is not respeaker core v2 :)

Psychokiller1888, 2019-01-25:
Thank you ;)

Now, what about the matrix voice?

joeduck, 2019-01-25:
Hi, must look tomorow. my time for snips today runs out :-(

joeduck, 2019-01-27:
Hi. Matrix voice is running with the actual SLC an the Alexa Pattern. Sometimes one LED stays on. When I again say "hey snips" and nothing else, all LEDs goes off.

Had all my node-red Flows make with the mqtt Dialog Messages instead of the "say" Message. So the Dialog ends correkt and SLC regognize the End of the Session and switch the LEDs of.

But the shutdown issue is still here. Musst wait about 2 Minutes:

```
Jan 27 13:03:43 jeannie systemd[1]: Started User Manager for UID 1000.
Jan 27 13:04:12 jeannie systemd[1]: Started Session c5 of user pi.
Jan 27 13:04:34 jeannie systemd[1]: Stopping Snips Led Control...
Jan 27 13:04:34 jeannie python[5333]: Shutting down Snips Led Control
Jan 27 13:04:34 jeannie python[5333]: INFO:SnipsLedControl:Shutting down Snips Led Control
Jan 27 13:04:34 jeannie python[5333]: Led number missmatch (18/18), aborting
Jan 27 13:04:34 jeannie python[5333]: WARNING:SnipsLedControl:Led number missmatch (18/18), aborting
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: State 'stop-sigterm' timed out. Killing.
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: Killing process 5333 (python) with signal SIGKILL.
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: Killing process 5344 (python) with signal SIGKILL.
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: Main process exited, code=killed, status=9/KILL
Jan 27 13:06:04 jeannie systemd[1]: Stopped Snips Led Control.
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: Unit entered failed state.
Jan 27 13:06:04 jeannie systemd[1]: snipsledcontrol.service: Failed with result 'timeout'.

```

Psychokiller1888, 2019-01-27:
The missmatch should be solved in 1.7.1 branch. I will release it soon, but you can try it if you feel like it

joeduck, 2019-01-28:
Ok. I will try it  :-)

joeduck, 2019-01-28:
Downloaded v1.7.1 as zip.
But its the same. 1 LED will allways stay on and the error stays also. I wonder why SLC reports 1.7?
```
2019-01-28 20:10:09,385 [MainThread] - [INFO] - Starting Snips Led Control v. 1.7
2019-01-28 20:10:09,399 [MainThread] - [INFO] - Initializing SnipsLedControl
2019-01-28 20:10:09,400 [MainThread] - [INFO] - Loading configurations
2019-01-28 20:10:09,407 [MainThread] - [INFO] - Loaded 11 hardware references
2019-01-28 20:10:09,408 [MainThread] - [INFO] - - Mqtt server set to 127.0.0.1
2019-01-28 20:10:09,409 [MainThread] - [INFO] - - Mqtt port set to 1883
2019-01-28 20:10:09,409 [MainThread] - [INFO] - - Client id set to default
2019-01-28 20:10:09,410 [MainThread] - [INFO] - - Hardware set to Matrix Voice
2019-01-28 20:10:09,411 [MainThread] - [INFO] - - Using alexa as pattern with 18 leds
2019-01-28 20:10:09,412 [MainThread] - [INFO] - Initializing leds controller
2019-01-28 20:10:10,807 [MainThread] - [INFO] - Connected to data publisher with port 20024
2019-01-28 20:10:10,822 [MainThread] - [INFO] - Counted 18 leds on device
2019-01-28 20:10:10,823 [MainThread] - [INFO] - LED count obtained. Disconnecting from data publisher 20024
2019-01-28 20:10:10,842 [MainThread] - [INFO] - Snips Led Control started
2019-01-28 20:10:12,118 [Thread-1] - [WARNING] - Led number missmatch (18/18), aborting
2019-01-28 20:10:55,606 [Thread-2] - [DEBUG] - On hotword triggered
2019-01-28 20:10:56,114 [Thread-2] - [DEBUG] - On listen triggered
2019-01-28 20:11:00,381 [Thread-2] - [DEBUG] - On think triggered
2019-01-28 20:11:01,520 [Thread-2] - [DEBUG] - On success received, but it wasn't for me
2019-01-28 20:11:07,340 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2019-01-28 20:11:07,410 [Thread-1] - [WARNING] - Led number missmatch (18/18), aborting
2019-01-28 20:11:13,487 [MainThread] - [INFO] - Shutting down Snips Led Control
2019-01-28 20:11:13,489 [MainThread] - [WARNING] - Led number missmatch (18/18), aborting
```

Psychokiller1888, 2019-01-29:
It report 1.7 because it's not yet tagged :) 

Checking the missmatch

Psychokiller1888, 2019-01-29:
Check the commit I just made, should fix Alexa's problem

joeduck, 2019-01-29:
Thanks.  Can do this earliest tomorrow evening. Sorry.

joeduck, 2019-01-30:
Now the LEDs working correct. Thanks! Any Idea about the shutdown Issue? I must alwas kill the Process. It took about 2 Minutes until the Process ends if not killed.

Psychokiller1888, 2019-01-31:
Will check that, surely a problem in the communication protocol with the matrix unit

joeduck, 2019-02-04:
Hi. Still have the Issue with one LED inverted (stays on) with the Alexa Pattern. Can i Help to find the error.

marcelser, 2019-06-06:
Hi,

Sorry to chime in but I also have a another Problem. Don't know if it's only matrix voice related. However every app that outputs long texts like wiki pages or public transport time tables leads to animation stopping after some seconds and leds not turning off. It's like something times out. For apps with shorter output like time of day or calculator it works fine. I'm currently running on 1.7.1 from installer. Tell me in case I need to update to get some more fixes. Thank you.

Psychokiller1888, 2019-06-28:
I just pushed an update for Matrix devices. This is untested, as I don't have any Matric devices. If you feel brave enough to test and report here it is: https://github.com/Psychokiller1888/snipsLedControl/releases/tag/v1.8-beta5

closed 2019-07-15 by Psychokiller1888

joeduck, 2019-07-15:
Now it works. Many thanks 👍