project-alice-assistant/HermesLedControl #26: Respeaker Mic Array V2 not working

ebkde, 2019-01-21:
Hi,

I installed snipsLedControl on my RPi 3 with a Respeaker Mic Array v2 using the slc_download.sh script. It installs fine and also starts without errors. But all I see is the default behaviour of the mic array. SPI is enable (I see spidev0.0 and spidev0.1 in /dev).

**How I did start SLC, with what arguments**
/usr/bin/python main.py --hardware=respeakerMicArrayV2 --pattern=alexa

I also tried other patterns but there is no difference in the led behaviour.

**Led device name**
ReSpeaker Mic Array V2

**The bug**
The LEDs are not controlled

**To Reproduce**
fresh install of raspbian stretch lite + snips + snipsLedControl

Seams to be the same Issue as reported in [#18](https://github.com/Psychokiller1888/snipsLedControl/issues/18). But switching to a custom hotword did not resolve the issue.

Psychokiller1888, 2019-01-21:
Did you instal lthe hardware using SLC?

ebkde, 2019-01-21:
Hi

> Did you instal lthe hardware using SLC?
yes, I choose to use 6 channels (is that correct/the best option?)

I tried debug mode 2 times

with alexa pattern
`sudo python main.py --hardware=respeakerMicArrayV2 --pattern=alexa --debug=True
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker Mic array v2
- Using alexa as pattern with 12 leds
Initializing leds controller
Snips Led Control started
On hotword triggered
On listen triggered
On think triggered
On sucess received, but it wasn't not for me
On say triggered
On sucess received, but it wasn't not for me
On hotword toggle on triggered
`
and with kiboost pattern
`sudo python main.py --hardware=respeakerMicArrayV2 --pattern=kiboost --debug=True
Starting Snips Led Control v. 1.7
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker Mic array v2
- Using kiboost as pattern with 12 leds
Initializing leds controller
Snips Led Control started
On hotword triggered
On listen triggered
On think triggered
On sucess received, but it wasn't not for me
On say triggered
On sucess received, but it wasn't not for me
On hotword toggle on triggered`

as you see there was no error, but the led behaviour was completely the same (a ring of blue leds with a turqoise led indicating DOA whenever the array recognizes any sound, but nothing else).

Psychokiller1888, 2019-01-21:
- When you start SLC do you have any led animation?
- There is no DOA implemented in SLC yet, though I'm working on it, so if you have a DOA, then you have something else running overriding the leds

ebkde, 2019-01-21:
> * When you start SLC do you have any led animation?
> * There is no DOA implemented in SLC yet, though I'm working on it, so if you have a DOA, then you have something else running overriding the leds

no, there is no animation when I start SLC. Do you have any hint on how to check what might override the leds?

I also noticed that after running the one of the pixel ring examples
`sudo python examples/usb_mic_array.py`
(which works fine)

any led animation except the red led in the center of the array is gone and won't come back until I reboot the device

ebkde, 2019-01-21:
nothing but SLC ;-)

I just followed the [RPi Quickstart Guide from Snips](https://docs.snips.ai/getting-started/quick-start-raspberry-pi) (I stopped after step 6). and then used the slc_download script. That's it

Psychokiller1888, 2019-01-21:
Also, open hardware.json and change, on line 26:

`"interface": "respeakerMicArrayV2",`

ebkde, 2019-01-22:
> Well, you tell me that a blue ring fires up with a violet led for DOA. SLC doesn't do DOA yet
> 
> The Alexa pattern is at boot is a blue circle that ends with a violet one, always at the same position for now. When you say your hotword, the pattern shows up?

I had a blue ring with a single turqoise led indicating DOA, which is the Respeaker Mic Array V2's default animation. When saying the hotword nothing changed.

> Also, open hardware.json and change, on line 26:
> 
> `"interface": "respeakerMicArrayV2",`

That did the trick!
Thank you very much.

closed 2019-01-22 by ebkde