project-alice-assistant/HermesLedControl #27: Feature Request: For Respeaker Mic Array v2 make volume control available

herrwusel, 2019-02-01:
It would be nice to control the volume over SLC as well.
http://wiki.seeedstudio.com/ReSpeaker_Mic_Array_v2.0/#tuning

herrwusel, 2019-02-01:
If you are using the audio output of the mic array, you are able to set a volume through pixel_ring. SLC might as well do that. If you don't set it manually, it is 100%.

Psychokiller1888, 2019-02-01:
Ok, you have 2 new topics:

- 'hermes/leds/vadLed' => payload['state'] int (0-1)
- 'hermes/volume/set' => payload['volume'] int (hardware min - hardware max)

closed 2019-02-01 by Psychokiller1888

Psychokiller1888, 2019-02-02:
Did this do the trick?

OnkelBuTscH, 2019-06-17:
I do not know exactly what I have to do.
Could you please show me an example how I can change the volume?

Psychokiller1888, 2019-06-17:
Just publish a mqtt topic on `hermes/volume/set` with a payload containing volume for the desired volume level

OnkelBuTscH, 2019-06-18:
thank you, it works