project-alice-assistant/HermesLedControl #29: CoreV2 & kiboost pattern stuck in "orange wave" after hey Snips only

jenskastensson, 2019-02-02:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py xxxxxxxxx
respeaker@v2:/home/pi/snipsLedControl_v1.7.1$ sudo python main.py --debug=True --hardware=respeakerCoreV2 --pattern=kiboost

**Led device name**
respeakerCoreV2

**The bug**
A. Pattern kiboost slow flashing blue led

when it works:
B. Say "Hey Snips" only 
-> the 'blue wave' runs
-> the 'blue wave' stop after ~4 sec

when it fails:
C.  Say "Hey Snips" only 
-> the 'blue wave' runs
-> the 'blue wave'  stops after ~4 sec and is replaced by an orange wave which runs forever
Say  "Hey Snips" again, it wave turns red one sec, then blue wave for 3 sec, then back to orange wave "forever" 

**Debug output**
Follow instructions on https://github.com/Psychokiller1888/snipsLedControl/wiki/Run-in-debug-mode

Starting Snips Led Control v. 1.7.1
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to bedroom
- Hardware set to Respeaker Core v2
- Using kiboost as pattern with 12 leds
Initializing leds controller
Snips Led Control started

On hotword triggered   // start of first sequence ok
On listen triggered
On think triggered
On hotword toggle on triggered
On say received, but it wasn't for me    // end of first sequense, ok 

On hotword triggered    // start of second sequence which will end with orange
On listen triggered
On think triggered
On hotword toggle on triggered
On say received, but it wasn't for me   // start of second sequence which will end with orange

**To Reproduce**
Steps to reproduce the behavior
Random, I haven't been able to find a pattern, it happens randomly and often

Psychokiller1888, 2019-02-02:
Afaik Kiboost has an idle pattern. Try starting with --idlePattern=nothing

jenskastensson, 2019-02-02:
Unfortunately that didn't  help, it still gets stuck in "orange wave".
The kiboost normal idle pattern is "blue glow", with nopattern all leds are
off (as expected).
But still randomly stuck in Orange Wave

closed 2019-07-15 by Psychokiller1888