project-alice-assistant/HermesLedControl #32: Error loading configfile

martament, 2019-02-19:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py --debug=True

**Led device name**
ReSpeaker 4

**The bug**
Snipsledcontrol is working fine on my device with Respeaker 2
On my freshly installed Respeaker4 i can use the microphones, but snipsledcontrol wont start.
Error seems he can't load config.
Rights on configfile are ok.

**Debug output**
Starting Snips Led Control v. 1.7.1
Initializing SnipsLedControl
Loading configurations
Traceback (most recent call last):
  File "main.py", line 121, in <module>
    main()
  File "main.py", line 107, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.7.1/models/SnipsLedControl.py", line 47, in __init__
    self._snipsConfigs 	= self.loadConfigs()
  File "/home/pi/snipsLedControl_v1.7.1/models/SnipsLedControl.py", line 167, in loadConfigs
    configs = pytoml.load(confFile)
AttributeError: 'module' object has no attribute 'load'

**To Reproduce**
This is every time i start Snips Led Control.

martament, 2019-02-27:
I've changeed the Respeakter to a device working formerly with a Reaspeaker 2.
Error is the same.

Psychokiller1888, 2019-02-27:
I don't know unfortunately, never seen this. Python 3 pytoml maybe?

martament, 2019-05-17:
Got it working.
Steps to reproduce:
- installed raspian
- followed the installation procedure for slc
- Error as described
- Fix:
1.) change in /etc/systemd/system/snipsLedControl.service python to python3
2.) install missing pip3 => sudo apt-get install python3-pip
3.) install missing packages => sudo pip3 install paho-mqtt python-etcd spidev

Now slc is working as expected.

Psychokiller1888, 2019-05-20:
I will update to python 3, thank you

closed 2019-05-20 by Psychokiller1888