project-alice-assistant/HermesLedControl #34: No indictions - no MQTT messages after start

jenskastensson, 2019-03-16:
First of all, thanks for this great piece of code!

**How I did start SLC, with what arguments**
respeaker@v2:/home/pi/snipsLedControl_v1.7.1$ sudo python main.py --debug=True --hardware=respeakerCoreV2 --pattern=alexa

**Led device name**
respeakerCoreV2 

**The bug**
After start of SLC, sometimes (~1 time of 10) SLC doesn't receive any MQTT messages at all from Snips. At the same time Snips itself works as expected, My Snips application receives all messages from Snips. Only SLC fail to receive messages.
I'm sure SLC doesn't receive any messages as I put a extra log line in SLC.
There are no error message during init from SLC, nor are there any difference in the log output when I compare a working and non-working SLC start. The logs are identical.

The work-around I have put into place is to send a "ACK message" from SLC as soon as it receives any message from Snips. My application listens to the ACK and Snips. Whenever my app receives a Snips message and not an ACK within 2 sec, it restarts SLC. That kind of works but is not ideal.

It seems something goes wrong during the startup of SLC and the initialization of the MQTT client.  Can you help out with some extended / additional logging during startup , maybe some ideas how-to change the timing during start? 
Additional observation:
I think that SLC is more likely to fail if I restart it with "systemctl restart" rather than doing a "stop + wait 2 sec + start",
Do it might be a timing issue

**Debug output**
See above

**To Reproduce**
See above

jenskastensson, 2019-06-17:
A one sec delay in onconnect seems to fix the issue

Psychokiller1888, 2019-08-01:
Ok, finally tried this on a respeaker core v2, I couldn't repro, is it still actual?

jenskastensson, 2019-08-01:
Thanks for checking it out.
It happened 1 time out of 10-20, probably depending on the CPU load.
After adding a delay in SnipsLedControl.py onConnect, the problem seems to
be gone.

def onConnect(self, client, userdata, flags, rc):
*time.sleep(1)*
self._mqttClient.subscribe([

closed 2019-08-13 by Psychokiller1888