project-alice-assistant/HermesLedControl #35: Use leds for notifications?

solars, 2019-04-15:
Hi there,

Quick question: Is it possible to use the leds for notifications?

E.g. I imagine something like control over individual LEDs (multiple) where red led means
waste will be collected tomorrow, blue LED means there is a window open after 10pm... and things like that :) Controlled via openhab...

Thank you!

Psychokiller1888, 2019-04-16:
Hi!
Well, it's possible, but not really the point of SLC maybe? As SLC is to control your leds to give your assistant some kind of visual feedback on its interaction state.

That being said, it wouldn't be too hard for me to implement a few "customStates" that would be reachable through mqtt, for any needs

solars, 2019-04-16:
Yeah I thought it's probably not too much effort as you are driving them anyway.

What I want to do is build a device that reminds me of things through color codes.
For example

- door bell
- laundry complete
- waste will be collected tomorrow
- pending notification, much like alexa, so I can just ask "hey snips, whats up?" and it will tell me
what's going on (as above)

I thought since the respeaker has LEDs anyway, it might be cool to re-use it for this purpose.

closed 2019-05-31 by Psychokiller1888

solars, 2019-06-06:
:/