project-alice-assistant/HermesLedControl #36: SnipsLedControl Reboot piZero

lricher1, 2019-05-26:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py with respeker4 and google pattern

**Led device name**
The name of the device such 
 "ReSpeaker 4" on a pi0 and the snips-satellite software

**The bug**
After i installed the snipsLedControl, the device reboot. Each 10s after the reboot, the pi Zero is rebooted
I plugued off, the respeaker4, i disabled the snipsLedControl service
Now, my pi Zero  started and snips is OK but not the led !

**Debug output**
Follow instructions on https://github.com/Psychokiller1888/snipsLedControl/wiki/Run-in-debug-mode
No information in the logs => The device reboot

**To Reproduce**
Steps to reproduce the behavior
When i executed "python main.py", the pi zero reboot

Psychokiller1888, 2019-05-26:
This sounds like an undervoltage to me

`tail -f /var/log/syslog`

And start SLC on another console. Watch the logs

lricher1, 2019-05-26:
my syslog :
May 26 15:27:38 raspberrypi snips-satellite[821]: INFO:snips_audio_server_cli      : Starting audio server on mqtt for site bureau
May 26 15:27:38 raspberrypi snips-satellite[821]: INFO:snips_broadcast_mqtt_server : Starting broadcasting audio on hermes as site "bureau"
May 26 15:27:42 raspberrypi kernel: [   92.450403] random: crng init done
May 26 15:27:42 raspberrypi kernel: [   92.450428] random: 7 urandom warning(s) missed due to ratelimiting
May 26 15:27:42 raspberrypi kernel: [   92.485408] bcm2835-i2s 20203000.i2s: I2S SYNC error!
May 26 15:28:09 raspberrypi systemd[1]: Started Session c6 of user pi.
May 26 15:28:09 raspberrypi kernel: [  120.140383] bcm2835-i2s 20203000.i2s: I2S SYNC error!
May 26 15:28:44 raspberrypi kernel: [  154.291553] bcm2835-i2s 20203000.i2s: I2S SYNC error!
May 26 15:28:53 raspberrypi kernel: [  164.265247] bcm2835-i2s 20203000.i2s: I2S SYNC error!
May 26 15:29:02 raspberrypi kernel: [  172.734939] bcm2835-i2s 20203000.i2s: I2S SYNC error!
May 26 15:29:23 raspberrypi kernel: [  194.244792] bcm2835-i2s 20203000.i2s: I2S SYNC error!

On anothe console, i executed  python main.py --hardware=respeaker4 --pattern=alexa --debug=True

=> no new log on syslog and the piZero restart
it's strange

lricher1, 2019-05-26:
the power is 2400mA

onlinepilot, 2019-05-30:
> **Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**
>
> **How I did start SLC, with what arguments**
> python main.py with respeker4 and google pattern
>
> **Led [...]

Hi,
I have the same error with the same hardware.

Psychokiller1888, 2019-05-30:
Can you guys try to stop the service and start with the brightness argument?

```
sudo systemctl stop snipsLedControl
python main.py --hardware=respeaker4 --pattern=google --defaultBrightness=1
```
I can't help it, but it looks like power outage

Psychokiller1888, 2019-05-30:
By the way, I hope you are not running the full snips on a pi zero? Because that won't work for sure....

lricher1, 2019-05-30:
With python main.py --hardware=respeaker4 --pattern=google --defaultBrightness=1, i have the same problem.
Of course, i use snips-satellite

onlinepilot, 2019-05-30:
Hi, the same here.

The following steps I do for Installation: 

`sudo apt-get update`
`sudo apt-get upgrade`

`sudo apt-get install -y dirmngr`
`sudo bash -c 'echo "deb https://raspbian.snips.ai/$(lsb_release -cs) stable main" > /etc/apt/sources.list.d/snips.list'`
`sudo apt-key adv --keyserver gpg.mozilla.org --recv-keys D4F50CDCA10A2849`
`sudo apt-get update`
`sudo apt install snips-hotword-model-heysnipsv4`
`sudo apt install snips-satellite`
`sudo reboot`

`sudo raspi-config`
configuring password, timezone, localisation
enable SPI interface
expanding filesystem
->reboot

`sudo apt install git`
`git clone https://github.com/respeaker/seeed-voicecard`
`cd seeed-voicecard`
`sudo ./install.sh`
`sudo reboot`

until here, the pi and the Respeaker are working fine ...
... and now your script:

`wget https://gist.githubusercontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/e24882e8997730dcf7a308e303b3b88001dbbfa1/slc_download.sh`
`sudo chmod +x slc_download.sh`
`sudo ./slc_download.sh`
set 2 for respeaker 4mic
set 1 for google pattern
answer `no` to not reconfigure the respeaker (testet with answer `yes` with the same result)

Now the pi reboots while it is booting every few, maybe, 10 seconds.

Psychokiller1888, 2019-05-31:
You don't need to install the respeaker before SLC, it does it for you. Anyway, that's not the problem.

You guys running on python 2?

Psychokiller1888, 2019-05-31:
Can you try to unplug the respeaker 4 and start SLC?

Psychokiller1888, 2019-05-31:
I'm installing on a pi zero atm, have to leave for work though. Will continue this evening

onlinepilot, 2019-05-31:
I don´t have changed the default python installation. Version is 2.7.13 .

Psychokiller1888, 2019-05-31:
Ok, I narrowed it down to the powering up the respeaker 4 when the interface is starting

Psychokiller1888, 2019-05-31:
Well... Bad news...

[image]

It works.... But this is on a pi 3B. I get exactly the same problem on a pi zero running a respeaker 4. The problem is the moment when trying to start powering the respeaker (interfaces/apa102.py line 23). Without this the leds are not powered SLC works nicely. But without power, there's no light. With power it crashes from undervoltage which I could measure.

I'll try to get in touch with seeed to see what they say

Psychokiller1888, 2019-05-31:
@lricher1 @onlinepilot 
But I have a good news:

I was able to start SLC with a respeaker 4 on a pi zero... But you need an iron and a capacitor:
[image]

I used a 10kuf capacitor. You find them literally everywhere and even to buy they are like 10 cents. Put it accross +5v and ground so the power needed to boot the respeaker 4 doesn't draw to much for the pi zero

lricher1, 2019-06-01:
which pin is + ? 
the 2 ou 3 ? 
Thanks

Psychokiller1888, 2019-06-01:
The second one (and the first one) are +5v. Third is gnd

onlinepilot, 2019-06-03:
Sorry, I was away for weekend.
@Psychokiller1888 
Thank you for researching a solution, but I am wondering, that the problem is highlighting now. Which change happend since the last version? Or is this the first time for this hardware set?

Psychokiller1888, 2019-06-03:
I never saw anyone plugging a respeaker 4 on a pi zero :) And I never tried myself. The respeaker 2 is much better anyway, though, not so many leds

lricher1, 2019-06-03:
For me, the mic is better with the respeaker4. And i prefer 12 ledd for my snips components

Psychokiller1888, 2019-06-09:
So this is solved?

closed 2019-06-09 by Psychokiller1888