project-alice-assistant/HermesLedControl #38: custom pattern events get out of sync with mqtt events

TeeFiX42, 2019-06-13:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
sudo python main.py --hardware=respeaker4 --pattern=custom --debug=True

**Led device name**
ReSpeaker4

**The bug**
Sometimes, the event functions in my custom pattern are not immediatly run when the mqtt change was detected.
To clarify what I mean, I added a small print("") statement the prints the event function that was started. So when I run the slc with debug option, you can see where it goes wrong.
I marked the moment with ##### where the LED pattern event becomes out of sync

**Debug output**
Starting Snips Led Control v. 1.7.1
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker 4
- Using custom as pattern with 12 leds
Initializing leds controller
Snips Led Control started
Implement me! /ledPatterns/CustomLedPattern.py
On hotword triggered
wake_up
On listen triggered
listen
On think triggered
think
On success received, but it wasn't for me
On say triggered
speak
On listen triggered
listen
On think triggered
think
On say triggered
speak
On hotword toggle on triggered
idle
On hotword triggered
wake_up
On listen triggered
listen
On think triggered
think
On hotword toggle on triggered
idle
On hotword triggered
wake_up
On listen triggered
listen
On think triggered
think
On success received, but it wasn't for me
On say triggered
speak
On hotword toggle on triggered
idle
On hotword triggered ### <-- next line should be "wake_up" because def wakeup(self) of my custompattern  should be triggered because the "On hotword" is triggered. As of here, all events that are triggered are 1 event out of sync.
On listen triggered
wake_up
On think triggered
listen
On success received, but it wasn't for me
On say triggered
think
On hotword toggle on triggered
speak
^CShutting down Snips Led Control

**To Reproduce**
I'll add my custom pattern file (I'm not the best programmer ;-) )
[CustomLedPattern.py.zip](https://github.com/Psychokiller1888/snipsLedControl/files/3287717/CustomLedPattern.py.zip)

Psychokiller1888, 2019-06-14:
Mmh, seems like on hotword does not play the animation for some reason. I'll check your file this evening after work

Psychokiller1888, 2019-06-14:
Not the problem, but your onStart should call self._controller.idle()

TeeFiX42, 2019-06-29:
adding self._controller.idle() to the onstart is not solving the issue.
I solved this by removing the LED actions in the wakeup.
For some reason, the snipsLedcontrol was running the wakeup code and in meantime the On listen was triggered.
After I remove the code in wakeup, the LED actions remain in sync. (NB: it's nice to hear "42" when asking "what is the answer to the universe, the question of live and everything else?")

TeeFiX42, 2019-06-29:
My bad: not solved by removing wakeup actions.
Now I notice it get's aout of sync each time I use the TimeDateCheck app when I ask the current time. The app responds twice the current time. As a result, I notice two times "on say triggered":
On hotword toggle on triggered
idle
On hotword triggered
On listen triggered
wake_up
listen
On think triggered
think
On success received, but it wasn't for me
On say triggered
On say triggered
speak
On hotword toggle on triggered
speak
On hotword triggered
On listen triggered
idle
On think triggered
wake_up

I might fix this in the app, but I still don't understand why the custom led control becomes out of sync.  Can you please check what goes wrong?

Psychokiller1888, 2019-06-30:
Ok, first things of, try to implement the onStart as it is saying:

```
	def onStart(self, *args):
		self._controller.wakeup()
		self._controller.idle()
```

My problem here, in your logs, is that I never see your added "prints", but I see the mqtt capture twice. Is everything fine with Snips? It seems to want to speak twice

Psychokiller1888, 2019-08-01:
Closing as no answer

closed 2019-08-01 by Psychokiller1888