project-alice-assistant/HermesLedControl #39: Support for original ReSpeaker 7mic Array

hipitihop, 2019-06-16:
I know this is superseded by Array V2, but I have one of these. Happy to be a test bunny/tester if you don't have one of these to experiment with.

**Name of the device**
ReSpeaker-Mic-Array-Far-field-w-7-PDM-Microphone

**Website**
https://www.seeedstudio.com/ReSpeaker-Mic-Array-Far-field-w-7-PDM-Microphone-p-2719.html

Psychokiller1888, 2019-06-17:
Can gladly try, will let you know about the advencement

hipitihop, 2019-06-22:
Some feedback. I have tried beta 4 now on RasberryPi 3B with ReSpeaker 7 array. This is setup as a satellite. Install went smoothly, and this satellite continues to work with snips, but nothing from 12 leds, the only led on as normal is the one green power led (sep to the other 12 leds which never light up)

Psychokiller1888, 2019-06-29:
I should get a device soon and I'll be able to help  a little firther

hipitihop, 2019-07-29:
Trace from latest:
```bash
pi@raspberrypi:~ $ sudo ./slc_download.sh
rm: cannot remove 'v1.9.2': No such file or directory
--2019-07-29 17:03:27--  https://api.github.com/repos/Psychokiller1888/snipsLedControl/tarball/v1.9.2
Resolving api.github.com (api.github.com)... 192.30.255.116
Connecting to api.github.com (api.github.com)|192.30.255.116|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.2 [following]
--2019-07-29 17:03:28--  https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.2
Resolving codeload.github.com (codeload.github.com)... 192.30.255.120
Connecting to codeload.github.com (codeload.github.com)|192.30.255.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.9.2’

v1.9.2                         [   <=>                                  ]  53.31K   123KB/s    in 0.4s    

2019-07-29 17:03:30 (123 KB/s) - ‘v1.9.2’ saved [54585]

What device do you wish to control with SLC?
 1) respeaker2
 2) respeaker4
 3) respeakerMicArrayV2
 4) neoPixelsSK6812RGBW
 5) neoPixelsWS2812RGB
 6) matrixvoice
 7) matrixcreator
 8) respeakerCoreV2
 9) respeaker6MicArray
10) respeaker7MicArray
11) googleAIY
12) I'm using simple leds on GPIOs
13) don't overwrite existing parameters
14) cancel
#? 10
What pattern do you want to use?
1) google
2) alexa
3) custom
4) kiboost
5) cancel
#? 4 
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Ign:3 http://apt.mopidy.com stretch InRelease                      
Hit:4 http://apt.mopidy.com stretch Release                                                  
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]                                
Hit:6 https://raspbian.snips.ai/stretch stable InRelease       
Get:7 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Fetched 11.7 MB in 43s (270 kB/s)                                                                         
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u4).
mosquitto is already the newest version (1.4.10-3+deb9u4).
mosquitto-clients is already the newest version (1.4.10-3+deb9u4).
portaudio19-dev is already the newest version (19.6.0-1).
python-numpy is already the newest version (1:1.12.1-3).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Requirement already satisfied: virtualenv in /usr/local/lib/python3.5/dist-packages
Running virtualenv with interpreter /usr/bin/python3.5
Already using interpreter /usr/bin/python3.5
Using base prefix '/usr'
New python executable in /home/pi/snipsLedControl_v1.9.2/venv/bin/python3.5
[... 35 lines cut ...]
1) yes
2) no
3) cancel
#? 1
############### Respeaker 7 Mic Array installation #########################
################ Please run this script with sudo ##########################
Uninstalling gpiozero-1.5.1:
  Successfully uninstalled gpiozero-1.5.1
Uninstalling RPi.GPIO-0.7.0:
  Successfully uninstalled RPi.GPIO-0.7.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting respeaker
  Downloading https://www.piwheels.org/simple/respeaker/respeaker-0.6.2-py2.py3-none-any.whl (491kB)
     |████████████████████████████████| 491kB 169kB/s 
Installing collected packages: respeaker
Successfully installed respeaker-0.6.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python3-numpy
  ERROR: Could not find a version that satisfies the requirement python3-numpy (from versions: none)
ERROR: No matching distribution found for python3-numpy
############################## All done! ##############################
############## Don't forget to turn on the SPI interface! #############
Finished installing Snips Led Control v1.9.2
You may want to copy over your custom led patterns to the new version
```

Psychokiller1888, 2019-07-29:
https://github.com/Psychokiller1888/snipsLedControl/blob/master/installers/respeakers.sh

Remove line 7 to try again

hipitihop, 2019-08-02:
Multiple attempts on dev staelite kit i.e. pi zero w respeaker2 seem to hanf

````bash
pi@snips-sat:~ $ sudo ./slc_download.sh
rm: cannot remove 'v1.9.4': No such file or directory
--2019-08-02 23:07:20--  https://api.github.com/repos/Psychokiller1888/snipsLedControl/tarball/v1.9.4
Resolving api.github.com (api.github.com)... 192.30.255.117
Connecting to api.github.com (api.github.com)|192.30.255.117|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.4 [following]
--2019-08-02 23:07:21--  https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.4
Resolving codeload.github.com (codeload.github.com)... 192.30.255.121
Connecting to codeload.github.com (codeload.github.com)|192.30.255.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.9.4’

v1.9.4                         [   <=>                                  ]  53.31K   109KB/s    in 0.5s    

2019-08-02 23:07:23 (109 KB/s) - ‘v1.9.4’ saved [54594]

What device do you wish to control with SLC?
 1) respeaker2
 2) respeaker4
 3) respeakerMicArrayV2
 4) neoPixelsSK6812RGBW
 5) neoPixelsWS2812RGB
 6) matrixvoice
 7) matrixcreator
 8) respeakerCoreV2
 9) respeaker6MicArray
10) respeaker7MicArray
11) googleAIY
12) I'm using simple leds on GPIOs
13) don't overwrite existing parameters
14) cancel
#? 1
What pattern do you want to use?
1) google	 3) projectalice  5) kiboost
2) alexa	 4) custom	  6) cancel
#? 3
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]                            
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease                                   
Hit:3 https://raspbian.snips.ai/stretch stable InRelease                                                  
Fetched 25.4 kB in 12s (2,073 B/s)                                                                        
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package numpy
Reading package lists... Done
Building dependency tree        
Reading state information... Done
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Requirement already satisfied: virtualenv in /usr/local/lib/python3.5/dist-packages
Running virtualenv with interpreter /usr/bin/python3.5
Already using interpreter /usr/bin/python3.5
Using base prefix '/usr'
New python executable in /home/pi/snipsLedControl_v1.9.4/venv/bin/python3.5
Also creating executable in /home/pi/snipsLedControl_v1.9.4/venv/bin/python
Installing setuptools, pip, wheel...
done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp35-cp35m-linux_armv6l.whl (67kB)
     |████████████████████████████████| 71kB 111kB/s 
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting spidev
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.4-cp35-cp35m-linux_armv6l.whl
Installing collected packages: spidev
Successfully installed spidev-3.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting gpiozero
  Downloading https://files.pythonhosted.org/packages/5a/06/e3440c452a0c2ba1f12527c12c5676b158f2591ac6c43979016f21f71b2a/gpiozero-1.5.1-py2.py3-none-any.whl (138kB)
     |████████████████████████████████| 143kB 321kB/s 
Collecting colorzero (from gpiozero)
  Downloading https://files.pythonhosted.org/packages/cc/a2/718e78485d0a42c3c2971e0d4a0eb9878bd1121cd640f0773de0695c858e/colorzero-1.1-py2.py3-none-any.whl
Installing collected packages: colorzero, gpiozero
Successfully installed colorzero-1.1 gpiozero-1.5.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting paho-mqtt
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.4.0-py3-none-any.whl (48kB)
     |████████████████████████████████| 51kB 78kB/s 
Installing collected packages: paho-mqtt
Successfully installed paho-mqtt-1.4.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting toml
  Downloading https://files.pythonhosted.org/packages/a2/12/ced7105d2de62fa7c8fb5fce92cc4ce66b57c95fb875e9318dba7f8c5db0/toml-0.10.0-py2.py3-none-any.whl
Installing collected packages: toml
Successfully installed toml-0.10.0
WARNING: Skipping pixel-ring as it is not installed.
Do you need to install / configure your respeaker2? This is strongly suggested as it does turn off services that might conflict as well!
1) yes
2) no
3) cancel
#? 1
######################### Respeakers installation #########################
#################### Please run this script with sudo #####################
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/da/32/1b8f2bb5fb50e4db68543eb85ce37b9fa6660cd05b58bddfafafa7ed62da/numpy-1.17.0.zip (6.5MB)
     |████████████████████████████████| 6.5MB 682kB/s 
Building wheels for collected packages: numpy
  Building wheel for numpy (setup.py) ... /
```

Psychokiller1888, 2019-08-02:
It's not hanging, you have a distro without wheels for numpy, takes long to install

Psychokiller1888, 2019-08-02:
Use 1.9.6, should now be much faster for numpy

hipitihop, 2019-08-03:
Thanks, trying with 1.9.6 (install continued so I will report back when complete) but I see this following traceback, Are you recommending an alternate distro to the maker-kit ?  stretch I think.

```
 sudo ./slc_download.sh
rm: cannot remove 'v1.9.6': No such file or directory
--2019-08-03 12:56:16--  https://api.github.com/repos/Psychokiller1888/snipsLedControl/tarball/v1.9.6
Resolving api.github.com (api.github.com)... 192.30.255.117
Connecting to api.github.com (api.github.com)|192.30.255.117|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.6 [following]
--2019-08-03 12:56:18--  https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9.6
Resolving codeload.github.com (codeload.github.com)... 192.30.255.120
Connecting to codeload.github.com (codeload.github.com)|192.30.255.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.9.6’

v1.9.6                         [   <=>                                  ]  53.39K  96.8KB/s    in 0.6s    

2019-08-03 12:56:19 (96.8 KB/s) - ‘v1.9.6’ saved [54674]

What device do you wish to control with SLC?
 1) respeaker2
 2) respeaker4
 3) respeakerMicArrayV2
 4) neoPixelsSK6812RGBW
 5) neoPixelsWS2812RGB
 6) matrixvoice
 7) matrixcreator
 8) respeakerCoreV2
 9) respeaker6MicArray
10) respeaker7MicArray
11) googleAIY
12) I'm using simple leds on GPIOs
13) don't overwrite existing parameters
14) cancel
#? 1
What pattern do you want to use?
1) google	 3) projectalice  5) kiboost
2) alexa	 4) custom	  6) cancel
#? 3
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]                           
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]                        
Hit:3 https://raspbian.snips.ai/stretch stable InRelease                                                  
Fetched 40.3 kB in 9s (4,120 B/s)                                                                         
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u4).
mosquitto is already the newest version (1.4.10-3+deb9u4).
mosquitto-clients is already the newest version (1.4.10-3+deb9u4).
portaudio19-dev is already the newest version (19.6.0-1).
Suggested packages:
  gfortran python-numpy-doc python3-nose python3-numpy-dbg
The following NEW packages will be installed:
  python3-numpy
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 1,693 kB of archives.
After this operation, 9,492 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.org/raspbian stretch/main armhf python3-numpy armhf 1:1.12.1-3 [1,693 kB]
Fetched 1,693 kB in 12s (136 kB/s)                                                                        
Selecting previously unselected package python3-numpy.
(Reading database ... 73342 files and directories currently installed.)
Preparing to unpack .../python3-numpy_1%3a1.12.1-3_armhf.deb ...
Unpacking python3-numpy (1:1.12.1-3) ...
Setting up python3-numpy (1:1.12.1-3) ...
Processing triggers for man-db (2.7.6.1-2) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Collecting virtualenv
  Downloading https://files.pythonhosted.org/packages/db/9e/df208b2baad146fe3fbe750eacadd6e49bcf2f2c3c1117b7192a7b28aec4/virtualenv-16.7.2-py2.py3-none-any.whl (3.3MB)
    100% |████████████████████████████████| 3.3MB 10kB/s 
Installing collected packages: virtualenv
Successfully installed virtualenv-16.7.2
Running virtualenv with interpreter /usr/bin/python3.5
Already using interpreter /usr/bin/python3.5
Using base prefix '/usr'
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/virtualenv.py", line 2635, in <module>
    main()
  File "/home/pi/.local/lib/python3.5/site-packages/virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "/home/pi/.local/lib/python3.5/site-packages/virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "/home/pi/.local/lib/python3.5/site-packages/virtualenv.py", line 1428, in install_python
    mkdir(lib_dir)
  File "/home/pi/.local/lib/python3.5/site-packages/virtualenv.py", line 394, in mkdir
    os.makedirs(at_path)
  File "/usr/lib/python3.5/os.py", line 231, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.5/os.py", line 231, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.5/os.py", line 241, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/pi/snipsLedControl_v1.9.6/venv'
bash: line 3: /home/pi/snipsLedControl_v1.9.6/venv/bin/activate: No such file or directory
```

Psychokiller1888, 2019-08-03:
Can you try to open the installer.sh and add, on line 68: `chown -R ${USER} ${USERDIR}/snipsLedControl_${VERSION}`. This would be how it looks:

[image]

closed 2020-02-09 by Psychokiller1888