project-alice-assistant/HermesLedControl #40: LEDs stay on after speaking

hokascha, 2019-06-18:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
sudo /usr/bin/python main.py --hardware=matrixvoice --pattern=google --debug=true

**Led device name**
Matrix Voice

**The bug**
When Snips is triggered via MQTT, i.e. when a MQTT client publishes a message to 
hermes/tts/say, the LEDs stay in the say pattern after speak has finished. I have to say "hey Snips" and then wait for the error sound feedback after which LEDs turn off as desired.

**Debug output**
```
Starting Snips Led Control v. 1.7.1
Initializing SnipsLedControl
Loading configurations
Loaded 11 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
On say triggered
DEBUG:SnipsLedControl:On say triggered
^CShutting down Snips Led Control
INFO:SnipsLedControl:Shutting down Snips Led Control
```

**To Reproduce**
Publish a message, I use perl (taken from: [https://github.com/Thyraz/Snips-Fhem](https://github.com/Thyraz/Snips-Fhem) )
```
sub say($$) {
    my ($hash, $cmd) = @_;
    my $sendData, my $json;
    my $siteId = "default";
    my $text = $cmd;
    my($unnamedParams, $namedParams) = parseParams($cmd);

    if (defined($namedParams->{'siteId'}) && defined($namedParams->{'text'})) {
        $siteId = $namedParams->{'siteId'};
        $text = $namedParams->{'text'};
    }

    $sendData =  {
        siteId => $siteId,
        text => $text,
        lang => "de",
        id => "0",
        sessionId => "0"
    };

    $json = toJSON($sendData);
    MQTT::send_publish($hash->{IODev}, topic => 'hermes/tts/say', message => $json, qos => 0, retain => "0");
}
```
**Note**
As you can see, sending ^C doesn't end the process, I have to kill it manually. Same happens when shutting down the service. Systemctl waits for the SIG-TERM to timeout for like 30 seconds and then kills the process.

hipitihop, 2019-06-19:
I can confirm the same behaviour both on a maker base and satellites i.e. with ReSpeaker Array 2 hats.

Psychokiller1888, 2019-06-19:
I have updated to python 3 for SLC 1.8. I'm pushing a beta installer and release online. It works flawlessly for me on a respeaker 2 and 4

Psychokiller1888, 2019-06-19:
Beta 4 version is now live and works for me at least. Please update

hipitihop, 2019-06-22:
@Psychokiller1888 I have now tried beta 4 on my maker satellite with respeaker2. It still leaves leds on afters tts say, but I now notice this is on the default google pattern. When I switched pattern to kiboost it does reset to waiting pattern after tts say.

hokascha, 2019-06-26:
@Psychokiller1888 how can I update to the beta version?

Psychokiller1888, 2019-06-26:
Remove the entire SLC directory and install fresh, better

Psychokiller1888, 2019-06-26:
@hipitihop I'll check the google pattern

marcelser, 2019-06-26:
I have the same problem it seems, here's the debug output. I'm also on 1.8 beta 4 and using google pattern.

`Snips Led Control: error: argument --debug: expected one argument
root@snipscontroller:/home/pi/snipsLedControl_v1.8-beta4# /home/pi/snipsLedControl_v1.8-beta4/venv/bin/python3.5 main.py --hardware=matrixvoice --pattern=google --debug=True
Starting Snips Led Control v. 1.8 beta
Initializing SnipsLedControl
Loading configurations
Loaded 12 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Connected to data publisher with port 20024
Counted 18 leds on device
INFO:SnipsLedControl:Counted 18 leds on device
LED count obtained. Disconnecting from data publisher 20024
INFO:SnipsLedControl:LED count obtained. Disconnecting from data publisher 20024
Snips Led Control started
INFO:SnipsLedControl:Snips Led Control started
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On success received but it wasn't for me
DEBUG:SnipsLedControl:On success received but it wasn't for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On success received but it wasn't for me
DEBUG:SnipsLedControl:On success received but it wasn't for me
On say triggered
DEBUG:SnipsLedControl:On say triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
On hotword triggered
DEBUG:SnipsLedControl:On hotword triggered
On listen triggered
DEBUG:SnipsLedControl:On listen triggered
On think triggered
DEBUG:SnipsLedControl:On think triggered
On hotword toggle on triggered
DEBUG:SnipsLedControl:On hotword toggle on triggered
`

hokascha, 2019-06-26:
Exact same behavior here as @marcelser describes. Looks like "hotword toggle" isn't triggered sometimes.

Psychokiller1888, 2019-06-26:
Hmmm, looks like a thread concurrency to me. I'll investigate

Psychokiller1888, 2019-06-28:
I just pushed an update for Matrix devices. This is untested, as I don't have any Matric devices. If you feel brave enough to test and report here it is: https://github.com/Psychokiller1888/snipsLedControl/releases/tag/v1.8-beta5

marcelser, 2019-07-14:
The new beta 1.5 runs really very nice and I've used it now for nearly 2 weeks. Not a single time the LEDs did stay on. Great work. Thanks for this.

Psychokiller1888, 2019-07-15:
Great to hear! Thank you for the feedback!

closed 2019-07-15 by Psychokiller1888