project-alice-assistant/HermesLedControl #41: Python error at startup on Respeaker Core V2

vincegre, 2019-06-26:
**How I did start SLC, with what arguments**
python main.py --debug=True

**Led device name**
Respeaker Core V2

**The bug**
The script doesn't start with a python error (Linux system up-to-date)

**Debug output**

respeaker@respeaker:/home/pi/snipsLedControl_v1.8-beta4$ python main.py --debug=True
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from models.SnipsLedControl import SnipsLedControl
  File "/home/pi/snipsLedControl_v1.8-beta4/models/SnipsLedControl.py", line 232
    def onMessage(self, client, userdata, message: MQTTMessage):
                                                 ^
SyntaxError: invalid syntax
respeaker@respeaker:/home/pi/snipsLedControl_v1.8-beta4$ 

**To Reproduce**
Install Linux image supplied by Seeed in e-MMC memory of Respeaker Core V2, make system updates of the linux, install snips as documented on snips website for respeaker core V2, installed then snipsLedControl as stated here !

Thanks

Vincèn

Psychokiller1888, 2019-06-26:
Python3 only, sry

vincegre, 2019-06-26:
> Python3 only, sry
ah damn, missed that :( May I suggest that the install script checks that and get straight an error message about it ? instead of doing setup as if everything was good ?

vincegre, 2019-06-26:
Side note: after checking more in details it looks like the system on Core V2 has both python 2.7 and 3.5 ! Will check if there is a way to get it to use by default python 3 and not 2.7 ! Will let you know my results ;)

Psychokiller1888, 2019-06-26:
It's weird, because it does check: https://puu.sh/DL3xT.png

vincegre, 2019-06-26:
hum very strange and I removed of system python 2.7, removed all stuffs installed by your script and restarted the install script and it reinstalled python 2.7 during the install process :(

vincegre, 2019-06-26:
Investigated a little more in it and this is what I got up to now:

had to correct the init script automatically created as it was pointing at a virtualenv instead of the real place where it had installed all snipsledcontrol files !
when starting it by hand it complained about missing dependencies and so I installed them one by one by hand in python 3.5 but then I reached that dependency that unless looks to be well installed makes an error:

```
root@respeaker:~/toml# /usr/bin/python3.5 /home/pi/snipsLedControl_v1.8-beta4/main.py --hardware=respeakerCoreV2 --pattern=google
Traceback (most recent call last):
  File "/home/pi/snipsLedControl_v1.8-beta4/main.py", line 21, in <module>
    rotatingHandler = handlers.RotatingFileHandler(filename='./logs/{}-logs.log'.format(date), mode='a', maxBytes=100000, backupCount=5)
  File "/usr/lib/python3.5/logging/handlers.py", line 150, in __init__
    BaseRotatingHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.5/logging/handlers.py", line 57, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.5/logging/__init__.py", line 1009, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.5/logging/__init__.py", line 1038, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/root/toml/logs/20190626-logs.log'
root@respeaker:~/toml#
```

Not sure what I can try now as I'm not really familiar with Python :( Did you get some people with the script running well on Respeaker Core V2 ?

Psychokiller1888, 2019-06-26:
It should point at the venv that should be created in the SLC install dir by the installer

closed 2019-07-15 by Psychokiller1888