project-alice-assistant/HermesLedControl #42: Alexa pattern one led stays on every 2nd request on matrix voice

marcelser, 2019-06-26:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
/home/pi/snipsLedControl_v1.8-beta4/venv/bin/python3.5 main.py --hardware=matrixvoice --pattern=alexa

**Led device name**
"Matrix Voice"

**The bug**
If you use Alexa pattern at the end of the animations when the alexa pattern is disabling the LEDs one of the LEDs stays active on Matrix Voice. If you do another action, that specific LED turns off as soon as the new animations start again (I guess when hotword is recognized) and then it turns on quickly on the "thinking" phase or whatever it is named and then shutdown animations works properly but only until next request then you again have the problem that 1 LED stays on

**Debug output**
Will not help here, it's a pattern problem which does on the logs, but I can create one

**To Reproduce**
use matrix voice, on SLC 1.8 beta 4 and set pattern to alexy and there you go.

Psychokiller1888, 2019-06-28:
I just pushed an update for Matrix devices. This is untested, as I don't have any Matric devices. If you feel brave enough to test and report here it is: https://github.com/Psychokiller1888/snipsLedControl/releases/tag/v1.8-beta5

marcelser, 2019-07-14:
Actually t've been running this version now for day and it works really great. Thanks for fing although I dnd't test alexa pattern yet only google.

Psychokiller1888, 2019-07-15:
Thank you for your feedbackm happy the update did fix the matrix devices problems

closed 2019-07-15 by Psychokiller1888