project-alice-assistant/HermesLedControl #43: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

thundergreen, 2019-06-30:
**How I did start SLC, with what arguments**
/home/pi/snipsLedControl_v1.8-beta6/venv/bin/python3.5 main.py --hardware=respeakerMicArrayV2 --pattern=google

**Led device name**
 "ReSpeaker  Mic Array V2" 

**The bug**
A clear and concise description of what the bug is.
Permission issue after installing the ledscript via sudo. Please see logs

**Debug output**

```
Jun 30 11:27:58 snips systemd[1]: snipsledcontrol.service: Service hold-off time over, scheduling restart.
Jun 30 11:27:58 snips systemd[1]: snips-skill-server.service: Service hold-off time over, scheduling restart.
Jun 30 11:27:58 snips systemd[1]: Stopped Snips Skill Server.
Jun 30 11:27:58 snips systemd[1]: Started Snips Skill Server.
Jun 30 11:27:58 snips systemd[1]: Stopped Snips Led Control.
Jun 30 11:27:58 snips systemd[1]: Started Snips Led Control.
Jun 30 11:27:58 snips snips-skill-server[2149]: ERROR:snips_skill_server: failed to fetch skills
Jun 30 11:27:58 snips snips-skill-server[2149]:  -> caused by: didn't find any skills in root skills directorie(s) ["/var/lib/snips/skills"] nor skill directorie(s) []
Jun 30 11:27:58 snips systemd[1]: snips-skill-server.service: Main process exited, code=exited, status=1/FAILURE
Jun 30 11:27:58 snips systemd[1]: snips-skill-server.service: Unit entered failed state.
Jun 30 11:27:58 snips systemd[1]: snips-skill-server.service: Failed with result 'exit-code'.
Jun 30 11:27:59 snips python3.5[2150]: Starting Snips Led Control v. 1.8 beta
Jun 30 11:27:59 snips python3.5[2150]: Initializing SnipsLedControl
Jun 30 11:27:59 snips python3.5[2150]: Loading configurations
Jun 30 11:27:59 snips python3.5[2150]: Loaded 12 hardware references
Jun 30 11:27:59 snips python3.5[2150]: - Mqtt server set to 192.168.1.41
Jun 30 11:27:59 snips python3.5[2150]: - Mqtt port set to 1883
Jun 30 11:27:59 snips python3.5[2150]: - Mqtt username set to hass
Jun 30 11:27:59 snips python3.5[2150]: - Mqtt password set to "hidden"
Jun 30 11:27:59 snips python3.5[2150]: - Client id set to default
Jun 30 11:27:59 snips python3.5[2150]: - Hardware set to ReSpeaker Mic array v2
Jun 30 11:27:59 snips python3.5[2150]: - Using google as pattern with 12 leds
Jun 30 11:27:59 snips python3.5[2150]: Initializing leds controller
Jun 30 11:27:59 snips python3.5[2150]: Traceback (most recent call last):
Jun 30 11:27:59 snips python3.5[2150]:   File "main.py", line 126, in <module>
Jun 30 11:27:59 snips python3.5[2150]:     main()
Jun 30 11:27:59 snips python3.5[2150]:   File "main.py", line 111, in main
Jun 30 11:27:59 snips python3.5[2150]:     slc = SnipsLedControl(args)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/SnipsLedControl.py", line 151, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     self._ledsController = LedsController(self)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/LedsController.py", line 52, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     self._pattern = GoogleHomeLedPattern(self)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/ledPatterns/GoogleLedPattern.py", line 13, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     super(GoogleHomeLedPattern, self).__init__(controller)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/LedPattern.py", line 14, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     self._animator #011#011#011#011= Animations(self._animation, controller)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/Animations.py", line 14, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     self.new()
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/Animations.py", line 18, in new
Jun 30 11:27:59 snips python3.5[2150]:     self._controller.clearLeds()
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/models/LedsController.py", line 419, in clearLeds
Jun 30 11:27:59 snips python3.5[2150]:     self._interface.clearStrip()
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/interfaces/respeakerMicArrayV2.py", line 42, in clearStrip
Jun 30 11:27:59 snips python3.5[2150]:     self.show()
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/interfaces/respeakerMicArrayV2.py", line 46, in show
Jun 30 11:27:59 snips python3.5[2150]:     self._leds.customize(self._colors)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/libraries/usb_pixel_ring_v2.py", line 45, in show
Jun 30 11:27:59 snips python3.5[2150]:     self.write(6, data)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/libraries/usb_pixel_ring_v2.py", line 74, in write
Jun 30 11:27:59 snips python3.5[2150]:     0, cmd, 0x1C, data, self.TIMEOUT)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/core.py", line 1025, in ctrl_transfer
Jun 30 11:27:59 snips python3.5[2150]:     self._ctx.managed_open()
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
Jun 30 11:27:59 snips python3.5[2150]:     return f(self, *args, **kwargs)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/core.py", line 120, in managed_open
Jun 30 11:27:59 snips python3.5[2150]:     self.handle = self.backend.open_device(self.dev)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 786, in open_device
Jun 30 11:27:59 snips python3.5[2150]:     return _DeviceHandle(dev)
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 643, in __init__
Jun 30 11:27:59 snips python3.5[2150]:     _check(_lib.libusb_open(self.devid, byref(self.handle)))
Jun 30 11:27:59 snips python3.5[2150]:   File "/home/pi/snipsLedControl_v1.8-beta6/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
Jun 30 11:27:59 snips python3.5[2150]:     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
Jun 30 11:27:59 snips python3.5[2150]: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
Jun 30 11:27:59 snips systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Jun 30 11:27:59 snips systemd[1]: snipsledcontrol.service: Unit entered failed state.
Jun 30 11:27:59 snips systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.
```
**To Reproduce**
Steps to reproduce the behavior

Just start ledsnipscontrol :)

Psychokiller1888, 2019-06-30:
Try to start with sudo

thundergreen, 2019-06-30:
Then I get this:

```
pi@snips:~ $ sudo  /home/pi/snipsLedControl_v1.8-beta6/venv/bin/python3.5 /home/pi/snipsLedControl_v1.8-beta6/main.py 
Traceback (most recent call last):
  File "/home/pi/snipsLedControl_v1.8-beta6/main.py", line 21, in <module>
    rotatingHandler = handlers.RotatingFileHandler(filename='./logs/{}-logs.log'.format(date), mode='a', maxBytes=100000, backupCount=5)
  File "/usr/lib/python3.5/logging/handlers.py", line 150, in __init__
    BaseRotatingHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.5/logging/handlers.py", line 57, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.5/logging/__init__.py", line 1009, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.5/logging/__init__.py", line 1038, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/logs/20190630-logs.log'
```

thundergreen, 2019-06-30:
seems like there's something broken?

thundergreen, 2019-06-30:
PR done :)

Psychokiller1888, 2019-06-30:
Thank you! But this messes up the manual start permissions due to logs being written by root and SLC running as pi by default. It would require the users to use sudo to start the service then

Did this instead: https://github.com/Psychokiller1888/snipsLedControl/commit/54746222a1abf0923ce13285fc9876a39e5aef17

closed 2019-06-30 by Psychokiller1888