project-alice-assistant/HermesLedControl #45: The service 'snipsledcontrol' starts, but it does not always work

kmbach, 2019-07-02:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
As service:
/home/pi/snipsLedControl_v1.8-beta6/venv/bin/python3.5 main.py --hardware=matrixvoice --pattern=google

**Led device name**
"Matrix Voice"

**The bug**
The LEDs do not work after a poweroff or reboot. At the start of the Raspi, the LEDs will light up briefly. The service. One

systemctl restart snipsledcontrol

fixes the problem temporarily.

daemon.log
If it does not work:
...
Jul  1 13:37:09 raspberrypi python3.5[699]: INFO: [/dev/matrixio_regmap] was opened
Jul  1 13:37:09 raspberrypi python3.5[699]: Caught exception in on_connect: 'NoneType' object has no attribute 'subscribe'
Jul  1 13:37:09 raspberrypi python3.5[699]: Snips Led Control started

if it works
...
Jul  1 14:39:25 raspberrypi python3.5[698]: Initializing leds controller
Jul  1 14:39:25 raspberrypi python3.5[698]: INFO: [/dev/matrixio_regmap] was opened
Jul  1 14:39:25 raspberrypi python3.5[698]: Snips Led Control started

**Debug output**
Not possible at system startup

**To Reproduce**
It is rather a random behavior. However, it does seem a bit temperature dependent. If the Raspi is switched off for longer, the service usually starts.
Sorry. Could it be that a small wait is missing when describing the registers of the board? It also seems to me different behavior, such to indicate randomly remaining LED's. I think I have read something like this here. Sorry, that's just a guess. I know something like this from my own hardware-related programming.

Psychokiller1888, 2019-07-06:
Can try this: Open the file /models/SnipsLedControl.py

On line 211 you'll find this:

`self._mqttClient.subscribe([`

Just before this line, add this:

`time.sleep(0.5)`

Then go up to line 12 and add this

`import time`

closed 2019-07-15 by Psychokiller1888

kmbach, 2019-08-09:
Hello, I have updated to version v1.9.8. The line with the time.sleep (0.5) is not integrated in this version. Again I had the same problem with the automatic start of the service. I added the line and now it works again. Would be nice if the line would flow into a later version.
Thank you!

Psychokiller1888, 2019-08-15:
I have integrated it, sorry, had forgotten about it