project-alice-assistant/HermesLedControl #46: [Question] Matrix Voice ESP32 Standalone Support?

TheXenocide, 2019-07-03:
**Name of the device**
Matrix Voice ESP32

**Website**
https://matrix-io.github.io/matrix-documentation/matrix-voice/esp32/

I see that you have support for Matrix Voice and Creator, but I was wondering if that support functions when the device is not actively connected to a Pi, or if it currently only runs from the Pi OS. I'm thinking of using something like [this](https://github.com/Romkabouter/Matrix-Voice-ESP32-MQTT-Audio-Streamer) to set up nice small standalone satellites and was hoping this might work

Psychokiller1888, 2019-07-06:
I never tried, sorry, I do not own any matrix device and they are not replying to my inquiries for help

Psychokiller1888, 2019-08-01:
After reading and all, SLC is meant for raspbian and not written in C as required for esp devices. Therefor no, It won't work

closed 2019-08-01 by Psychokiller1888