project-alice-assistant/HermesLedControl #47: dont work after update to Version 1.8.0

xinQuon, 2019-07-22:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
ExecStart=/home/pi/snipsLedControl_v1.8.0/venv/bin/python3.5 main.py --hardware=respeaker4 --pattern=google

**Led device name**
"ReSpeaker 4"

**The bug**
No reaction at the LEDs on the Version 1.8.0. In Version 1.7.0 all your patterns worked fine but after the update i have no function anymore. It is possible to downgrade to 1.7?
The status of the Service ("systemctl status snipsledcontrol") is "active (running)"

**Debug output**
Follow instructions on https://github.com/Psychokiller1888/snipsLedControl/wiki/Run-in-debug-mode
if i follow the instructions i get a syntax error:
pi@192.168.1.73:~/snipsLedControl_v1.8.0$ python main.py --debug=True --hardware=respeaker4 --pattern=google
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from models.SnipsLedControl import SnipsLedControl
  File "/home/pi/snipsLedControl_v1.8.0/models/SnipsLedControl.py", line 236
    def onMessage(self, client, userdata, message: MQTTMessage):
                                                 ^
SyntaxError: invalid syntax

if i put the debug parameter in the /etc/systemd/system/snipsledcontrol.service and start the service, the output is:
2019-07-22 07:03:48,691 [MainThread] - [INFO] - Shutting down Snips Led Control
2019-07-22 07:03:57,350 [MainThread] - [INFO] - Starting Snips Led Control v. 1.8
2019-07-22 07:03:57,363 [MainThread] - [INFO] - Initializing SnipsLedControl
2019-07-22 07:03:57,364 [MainThread] - [INFO] - Loading configurations
2019-07-22 07:03:57,410 [MainThread] - [INFO] - Loaded 12 hardware references
2019-07-22 07:03:57,411 [MainThread] - [INFO] - - Mqtt server set to localhost
2019-07-22 07:03:57,412 [MainThread] - [INFO] - - Mqtt port set to 1883
2019-07-22 07:03:57,413 [MainThread] - [INFO] - - Client id set to Wohnzimmer
2019-07-22 07:03:57,413 [MainThread] - [INFO] - - Hardware set to ReSpeaker 4
2019-07-22 07:03:57,414 [MainThread] - [INFO] - - Using google as pattern with 12 leds
2019-07-22 07:03:57,415 [MainThread] - [INFO] - Initializing leds controller
2019-07-22 07:03:57,637 [MainThread] - [INFO] - Snips Led Control started
2019-07-22 07:05:42,720 [MainThread] - [INFO] - Shutting down Snips Led Control

**To Reproduce**
see **the bug**

Psychokiller1888, 2019-07-22:
Hi!

I just updated the link you gave, it was starting SLC with python 2 and not the virtual env... Try to run in debug mode again

xinQuon, 2019-07-22:
Hi Psycho,
new output with "./venv/bin/python3.5 main.py --debug=True --hardware=respeaker4" and hotword + action triggered:
2019-07-22 19:54:47,007 [MainThread] - [INFO] - Starting Snips Led Control v. 1.8
2019-07-22 19:54:47,019 [MainThread] - [INFO] - Initializing SnipsLedControl
2019-07-22 19:54:47,020 [MainThread] - [INFO] - Loading configurations
2019-07-22 19:54:47,066 [MainThread] - [INFO] - Loaded 12 hardware references
2019-07-22 19:54:47,067 [MainThread] - [INFO] - - Mqtt server set to localhost
2019-07-22 19:54:47,067 [MainThread] - [INFO] - - Mqtt port set to 1883
2019-07-22 19:54:47,068 [MainThread] - [INFO] - - Client id set to Wohnzimmer
2019-07-22 19:54:47,069 [MainThread] - [INFO] - - Hardware set to ReSpeaker 4
2019-07-22 19:54:47,069 [MainThread] - [INFO] - - Using google as pattern with 12 leds
2019-07-22 19:54:47,070 [MainThread] - [INFO] - Initializing leds controller
2019-07-22 19:54:47,290 [MainThread] - [INFO] - Snips Led Control started
2019-07-22 19:54:50,140 [Thread-2] - [DEBUG] - On hotword triggered
2019-07-22 19:54:50,766 [Thread-2] - [DEBUG] - On listen triggered
2019-07-22 19:54:53,145 [Thread-2] - [DEBUG] - On think triggered
2019-07-22 19:54:54,374 [Thread-2] - [DEBUG] - On success received but it wasn't for me
2019-07-22 19:54:54,455 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2019-07-22 19:54:54,505 [Thread-2] - [DEBUG] - On say triggered
2019-07-22 19:55:00,637 [Thread-2] - [DEBUG] - On hotword toggle on triggered

Psychokiller1888, 2019-07-22:
Ok, got it. I really don't know what happened here.... And that you're the first to report after all this time :)

Can you open /home/pi/snipsLedControl_vxxxxxxx/interfaces/apa102.py

On line 22 and 25 you'll find LED(12). Replace with LED(5)

Psychokiller1888, 2019-07-22:
Or update to 1.8.1

xinQuon, 2019-07-22:
Hi again,
great job - replace works :-)

Thanks and have a nice day...

Psychokiller1888, 2019-07-22:
Thanks! Happy it works for you! Have a nice day too!

closed 2019-07-22 by Psychokiller1888