project-alice-assistant/HermesLedControl #49: Pyhton 3.7 on Buster

urbatecte, 2019-07-24:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
$ sudo ./slc_download.sh

**Led device name**
respeaker2 or respeaker4

**The bug**
After upgrading to Buster distro, SLC install seems to require Python 3.5

**Debug output**

pi@raspberrypi:~ $ sudo ./slc_download.sh
rm: cannot remove 'v1.9': No such file or directory
--2019-07-24 13:19:50--  https://api.github.com/repos/Psychokiller1888/snipsLedControl/tarball/v1.9
Resolving api.github.com (api.github.com)... 140.82.118.6
Connecting to api.github.com (api.github.com)|140.82.118.6|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9 [following]
--2019-07-24 13:19:50--  https://codeload.github.com/Psychokiller1888/snipsLedControl/legacy.tar.gz/v1.9
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘v1.9’

v1.9                              [  <=>                                          ]  53.11K   242KB/s    in 0.2s

2019-07-24 13:19:51 (242 KB/s) - ‘v1.9’ saved [54386]

What device do you wish to control with SLC?
1) respeaker2                             8) respeakerCoreV2
2) respeaker4                             9) respeaker6MicArray
3) respeakerMicArrayV2                   10) respeaker7MicArray
4) neoPixelsSK6812RGBW                   11) googleAIY
5) neoPixelsWS2812RGB                    12) I'm using simple leds on GPIOs
6) matrixvoice                           13) don't overwrite existing parameters
7) matrixcreator                         14) cancel
#? 1
What pattern do you want to use?
1) google
2) alexa
3) custom
4) kiboost
5) cancel
#? 4
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://raspbian.snips.ai/stretch stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.20.1-2).
mosquitto is already the newest version (1.5.7-1).
mosquitto-clients is already the newest version (1.5.7-1).
portaudio19-dev is already the newest version (19.6.0-1).
python-numpy is already the newest version (1:1.16.2-1).
The following packages were automatically installed and are no longer required:
  libdrm-amdgpu1 libllvm7 rpi.gpio-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Please make sure you have Python 3.5 installed

**To Reproduce**
Upgrade to buster with:
$ sudo nano /etc/apt/sources.list
$ deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
and the second one source file
$ sudo nano /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ buster main ui
then
$ sudo apt update
$ sudo apt-get dist-upgrade

Thanks a lot for your gorgeous app !
If you can also add a function to turn down the heat in France for today, it will be welcome ;o)

Psychokiller1888, 2019-07-24:
It is python 3.5 yes, since 1.8. Is 3.5 not available on buster? I can make it use 3.7 if on buster

urbatecte, 2019-07-24:
Knowing that Python deprecated rate seems to increase it could be worth ;o)

Thanks.

Le 24/07/2019 ?? 15:00, Psycho a ??crit??:

>
> It is python 3.5 yes, since 1.8. Is 3.5 not available on buster? I can 
> make it use 3.7 if on buster
>
> ???

---
L'absence de virus dans ce courrier ??lectronique a ??t?? v??rifi??e par le logiciel antivirus Avast.
https://www.avast.com/antivirus

Psychokiller1888, 2019-07-24:
https://github.com/Psychokiller1888/snipsLedControl/releases/tag/v1.9.1

urbatecte, 2019-07-24:
Nearly ;o)

/etc/systemd/system/snipsledcontrol.service was was still pointing at python3.5 instead 3.7
This one was easy for me ;o)
But this following one is too tricky:

$ cd /home/pi/snipsLedControl_v1.9.1
$ ./venv/bin/python3.7 main.py --debug=True
Starting Snips Led Control v. 1.9.1
Initializing SnipsLedControl
Loading configurations
Loaded 12 hardware references
- Mqtt server set to 10.0.4.48
- Mqtt port set to 1883
- Mqtt username set to homeassistant
- Mqtt password set to "hidden"
- Client id set to default
- Hardware set to ReSpeaker 2
- Using google as pattern with 3 leds
Initializing leds controller
Traceback (most recent call last):
  File "main.py", line 127, in <module>
    main()
  File "main.py", line 112, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.9.1/models/SnipsLedControl.py", line 151, in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl_v1.9.1/models/LedsController.py", line 46, in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl_v1.9.1/models/LedsController.py", line 104, in initHardware
    self._interface = APA102(numLed=self._hardware['numberOfLeds'], endFrame=self._hardware['endFrame'], hardware=self._hardware)
  File "/home/pi/snipsLedControl_v1.9.1/interfaces/apa102.py", line 34, in __init__
    if hardware and hardware['doa']:
KeyError: 'doa'

$

Psychokiller1888, 2019-07-24:
Can you check my commit and try it? I did not push a new version, but if needed tell me

urbatecte, 2019-07-24:
I'm not used to use commit.
I've replace code in original files.
restart SLC service => KO
reboot => KO
then run debug
$ cd /home/pi/snipsLedControl_v1.9.1
$ ./venv/bin/python3.7 main.py --debug=True
Starting Snips Led Control v. 1.9.2
Initializing SnipsLedControl
Loading configurations
Loaded 12 hardware references
- Mqtt server set to 10.0.4.48
- Mqtt port set to 1883
- Mqtt username set to homeassistant
- Mqtt password set to "hidden"
- Client id set to default
- Hardware set to ReSpeaker 2
- Using google as pattern with 3 leds
Initializing leds controller
Traceback (most recent call last):
  File "main.py", line 127, in <module>
    main()
  File "main.py", line 112, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.9.1/models/SnipsLedControl.py", line 151, in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl_v1.9.1/models/LedsController.py", line 46, in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl_v1.9.1/models/LedsController.py", line 104, in initHardware
    self._interface = APA102(numLed=self._hardware['numberOfLeds'], endFrame=self._hardware['endFrame'], hardware=self._hardware)
TypeError: __init__() got an unexpected keyword argument 'numLed'

Psychokiller1888, 2019-07-24:
Lemme push a new version then :)

Psychokiller1888, 2019-07-24:
https://github.com/Psychokiller1888/snipsLedControl/releases/tag/v1.9.2

urbatecte, 2019-07-24:
you rock it !
Thanks a lot

closed 2019-07-24 by urbatecte