project-alice-assistant/HermesLedControl #50: Error in sys log after adding ./venv/bin/pip3 install numpy

LazzaWA, 2019-07-29:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments
usually just reboot my Pi and dont use a command line. however when i tried to use something like 

> pi@snips:~/snipsLedControl_v1.9.2 $ sudo python main.py --hardware=respeaker4 --pattern=alexa --debug=True
 i get the following 

```
Traceback (most recent call last):
  File "main.py", line 10, in <module>
    from models.SnipsLedControl import SnipsLedControl
  File "/home/pi/snipsLedControl_v1.9.2/models/SnipsLedControl.py", line 236
    def onMessage(self, client, userdata, message: MQTTMessage):
```
Regardless thats not the difficulties i'm having :)

**Led device name**
The name of the device such as "ReSpeaker 2" or "Matrix Voice"

> Respeaker 4

**The bug**
I had SnipsLEDcontrol script somewhat working in  previous version then recently updated to version v1.9.2 and the script failed to work from then on. Originally i had a error as stated in this post [https://forum.snips.ai/t/snips-led-control/351/162?u=lazza](url) and followed the advise to add numpy as described in the response. I now get the following errors that i can't work out on my own unfortunately.
( because i'm learning linux stuff as i go so not fluent in commands etc)

using this command
`tail -f /var/log/syslog`
i get the following output 

```
> Jul 29 16:16:16 snips systemd[1]: Started Snips Led Control.
> Jul 29 16:16:16 snips python3[29166]: Starting Snips Led Control v. 1.9.2
> Jul 29 16:16:16 snips python3[29166]: Initializing SnipsLedControl
> Jul 29 16:16:16 snips python3[29166]: Loading configurations
> Jul 29 16:16:16 snips python3[29166]: Loaded 12 hardware references
> Jul 29 16:16:16 snips python3[29166]: - Mqtt server set to snips.local
> Jul 29 16:16:16 snips python3[29166]: - Mqtt port set to 1883
> Jul 29 16:16:16 snips python3[29166]: - Mqtt username set to snips
> Jul 29 16:16:16 snips python3[29166]: - Mqtt password set to "hidden"
> Jul 29 16:16:16 snips python3[29166]: - Client id set to master
> Jul 29 16:16:16 snips python3[29166]: - Hardware set to ReSpeaker 4
> Jul 29 16:16:16 snips python3[29166]: - Using alexa as pattern with 12 leds
> Jul 29 16:16:16 snips python3[29166]: Initializing leds controller
> Jul 29 16:16:17 snips python3[29166]: Hardware is DOA capable
> Jul 29 16:16:17 snips python3[29166]: Traceback (most recent call last):
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/venv/lib/python3.7/site-packages/numpy/core/__init__.py", line 17, in <module>
> Jul 29 16:16:17 snips python3[29166]:     from . import multiarray
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/venv/lib/python3.7/site-packages/numpy/core/multiarray.py", line 14, in <module>
> Jul 29 16:16:17 snips python3[29166]:     from . import overrides
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/venv/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
> Jul 29 16:16:17 snips python3[29166]:     from numpy.core._multiarray_umath import (
> Jul 29 16:16:17 snips python3[29166]: ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /home/pi/snipsLedControl_v1.9.2/venv/lib/python3.7/site-packages/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so)
> Jul 29 16:16:17 snips python3[29166]: During handling of the above exception, another exception occurred:
> Jul 29 16:16:17 snips python3[29166]: Traceback (most recent call last):
> Jul 29 16:16:17 snips python3[29166]:   File "main.py", line 127, in <module>
> Jul 29 16:16:17 snips python3[29166]:     main()
> Jul 29 16:16:17 snips python3[29166]:   File "main.py", line 112, in main
> Jul 29 16:16:17 snips python3[29166]:     slc = SnipsLedControl(args)
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/models/SnipsLedControl.py", line 151, in __init__
> Jul 29 16:16:17 snips python3[29166]:     self._ledsController = LedsController(self)
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/models/LedsController.py", line 46, in __init__
> Jul 29 16:16:17 snips python3[29166]:     if not self.initHardware():
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/models/LedsController.py", line 104, in initHardware
> Jul 29 16:16:17 snips python3[29166]:     self._interface = APA102(hardware=self._hardware, endFrame=self._hardware['endFrame'])
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/interfaces/apa102.py", line 35, in __init__
> Jul 29 16:16:17 snips python3[29166]:     from libraries.seeedstudios.channel_picker import ChannelPicker
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/libraries/seeedstudios/channel_picker.py", line 8, in <module>
> Jul 29 16:16:17 snips python3[29166]:     import numpy as np
> Jul 29 16:16:17 snips python3[29166]:   File "/home/pi/snipsLedControl_v1.9.2/venv/lib/python3.7/site-packages/numpy/__init__.py", line 142, in <module>
[... 36 lines cut ...]
Loaded 12 hardware references
- Mqtt server set to snips.local
- Mqtt port set to 1883
- Mqtt username set to snips
- Mqtt password set to "hidden"
- Client id set to master
- Hardware set to ReSpeaker 2
- Using google as pattern with 3 leds
Initializing leds controller
Caught exception in on_connect: 'NoneType' object has no attribute 'subscribe'
Button 1 not implemented, override it in CustomLedPattern
Snips Led Control started
Button 1 not implemented, override it in CustomLedPattern
Button 1 not implemented, override it in CustomLedPattern
`

Curiosity question in regards to the debug. it says hardware set to respeaker2 but that's not the option chosen or displayed in tail -f /var/log/syslog
not that i think it matters was just pointing it out in case it means something :)

**To Reproduce**
Steps to reproduce the behavior

Not sure. perhaps be a noob like me and know enough to be dangerous but not competent :)

FYI.. i did recently manually install python 3.7 on stretch because i thought that might help. not sure if that caused the issue also or at least contributes to it ?

Psychokiller1888, 2019-07-29:
It's normal the hardware is not the same, as starting with

`./venv/bin/python3.7 main.py --debug=True` defaults to respeaker2 as you don't set it to respeaker4

Past that, I can say that yes, python 3.7 really doesn't like numpy and it's the problem most likely. I'd try with python 3.5 if I were you

LazzaWA, 2019-07-29:
I've removed python 3.7 and installed 3.5.7 as per git page [https://gist.github.com/BMeu/af107b1f3d7cf1a2507c9c6429367a3b](url)

i then deleted SLC directories and the service , rebooted the pi and then re installed SLC. I ended up with exact same initial error of 

`import numpy as np
Jul 28 09:46:27 snips python3[7421]: ImportError: No module named ‘numpy’`

so then i done the 

`cd  /home/pi/snipsLedControl_v1.9.3
./venv/bin/pip3 install numpy`

And i end up with the exact same result as original post. I'm not too sure how to proceed from here so any assistance in pointing me in the right direction would be appreciated

Hermesfrank, 2019-07-30:
I tried to install SLC...the script ran except that this error (the only one I saw while I was monitoring):
[image]

While I was monitoring the installation, I saw earlier references in the script to _numpy_ and they seemed to run through OK and I seem to remember it saying it was already installed...

Anyway, I did:
`cd /home/pi/snipsLedControl_v1.9.3`
`./venv/bin/pip3 install numpy`

It appears to be hanging up trying to install...I suspect I should have first removed whatever numpy was already installed, but not sure just how to do that...

LazzaWA, 2019-07-30:
i do get the exact same error as you hermesfrank, how ever mine doesnt hang trying to install it always defaults to a cached version of numpy. i'm wondering in my case could the cached version just be junk ? *shrug*

[image]

Hermesfrank, 2019-07-30:
Sorry -I am too ignorant...maybe Psycho will have an answer when he checks in...my second attempt at _numpy_ install hung also...Frank

Psychokiller1888, 2019-07-30:
Hum... Let's try this:

```
sudo apt-get purge -y python-numpy
cd /home/pi/snipsLedControl_v1.9.3
./venv/bin/pip3 uninstall -y python-numpy
./venv/bin/pip3 --no-cache-dir install python-numpy
```
I suspect python3-numpy was removed now that python 2 is ending its life

Hermesfrank, 2019-07-30:
[image]
Same basic error...

Psychokiller1888, 2019-07-30:
Try with `./venv/bin/pip3 --no-cache-dir install numpy`, that's my fault, python numpy is for apt

Hermesfrank, 2019-07-30:
OK, that's running...then do re-install of SLC?

Psychokiller1888, 2019-07-30:
no, you shouldn't need to

Hermesfrank, 2019-07-30:
Not sure what's going on - it ran for awhile (maybe 5 minutes) with symbol turning intermittently, then hung...I'll try again...

Hermesfrank, 2019-07-30:
I have snips running (well sort of) - should I stop something?

Psychokiller1888, 2019-07-30:
nope, you should have snips ans SLC  running. I need logs to debug :)

Hermesfrank, 2019-07-30:
I ran Sam Status and saw that the audio server is not running...a look at the syslog and saw that there are no audio devices...arecord and aplay show same...all was running OK before SLC install...I think I will punt for now...thanks for trying...Frank

Psychokiller1888, 2019-07-30:
You can't mix everything. You just install Snips, then run SLC installer and chose wisely what you do. If you want or not SLC to install your hardware etc etc

Hermesfrank, 2019-07-30:
OK...I have apps installed, i.e., I ran the SLC install after installing Snips and the respeaker install via Sam Setup Audio...I will go back and do a clean install of Snips, without the setup audio, then install SLC and see how it goes...

LazzaWA, 2019-07-31:
Hi psycho. I'm not back home to test until tomorrow but will your responce be for my issue as well ? Cheers

LazzaWA, 2019-08-01:
For me, i tried the above suggestion and now when i do 
`tail -f /var/log/syslog
`
i get the following errors 

>Aug  1 12:26:33 snips python3[1384]: Starting Snips Led Control v. 1.9.3
Aug  1 12:26:33 snips python3[1384]: Initializing SnipsLedControl
Aug  1 12:26:33 snips python3[1384]: Loading configurations
Aug  1 12:26:33 snips python3[1384]: Traceback (most recent call last):
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 456, in loads
Aug  1 12:26:33 snips python3[1384]:     multibackslash)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 725, in load_line
Aug  1 12:26:33 snips python3[1384]:     value, vtype = self.load_value(pair[1], strictly_valid)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 805, in load_value
Aug  1 12:26:33 snips python3[1384]:     self.load_inline_object(v, inline_object)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 621, in load_inline_object
Aug  1 12:26:33 snips python3[1384]:     raise ValueError("Invalid inline table value encountered")
Aug  1 12:26:33 snips python3[1384]: ValueError: Invalid inline table value encountered
Aug  1 12:26:33 snips python3[1384]: During handling of the above exception, another exception occurred:
Aug  1 12:26:33 snips python3[1384]: Traceback (most recent call last):
Aug  1 12:26:33 snips python3[1384]:   File "main.py", line 127, in <module>
Aug  1 12:26:33 snips python3[1384]:     main()
Aug  1 12:26:33 snips python3[1384]:   File "main.py", line 112, in main
Aug  1 12:26:33 snips python3[1384]:     slc = SnipsLedControl(args)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/models/SnipsLedControl.py", line 53, in __init__
Aug  1 12:26:33 snips python3[1384]:     self._snipsConfigs #011= self.loadConfigs()
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/models/SnipsLedControl.py", line 175, in loadConfigs
Aug  1 12:26:33 snips python3[1384]:     configs = toml.load(confFile)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 134, in load
Aug  1 12:26:33 snips python3[1384]:     return loads(f.read(), _dict, decoder)
Aug  1 12:26:33 snips python3[1384]:   File "/home/pi/snipsLedControl_v1.9.3/venv/lib/python3.7/site-packages/toml/decoder.py", line 458, in loads
Aug  1 12:26:33 snips python3[1384]:     raise TomlDecodeError(str(err), original, pos)
Aug  1 12:26:33 snips python3[1384]: toml.decoder.TomlDecodeError: Invalid inline table value encountered (line 74 column 1 char 1886)
Aug  1 12:26:33 snips systemd[1]: snipsledcontrol.service: Main process exited, code=exited, status=1/FAILURE
Aug  1 12:26:33 snips systemd[1]: snipsledcontrol.service: Unit entered failed state.
Aug  1 12:26:33 snips systemd[1]: snipsledcontrol.service: Failed with result 'exit-code'.

I tried reinstalling SLC after this error and i ended up with the same error as above again

Psychokiller1888, 2019-08-01:
To me it looks like your /etc/snips.toml file is corrupted

LazzaWA, 2019-08-01:
Here's my latest status.

/etc/snips.toml only had a couple of bridge items (mqtt settings) un commented and also customtts enabled cause i'm using your other script also ( snipsSuperTTS.sh )

Snips itself works fine,sam status shows everything running, responds to voice , plays sound etc. the only problems im currently experiencing is the SLC script wont run and therefore i have no respeaker lights.

I have re commented the MQTT settings in the snips.toml ( because the bridge didn't work anyway, so im using node red on another pi as the bridge)  

Tested Snips operation again and all is good.
Tested SLC still not running
Deleted the SLC folder 
Deleted the SLC service from /etc/systemd/
Rebooted the pi
Done a tail -f /var/log/syslog and i get no errors other than a alsa error

Re downloaded the latest SLC using your WGET command noted in the install procedure and installed SLC. Selecting "#2" for respeaker 4 option and "#2" for alexa pattern.

I also allow your script to re install my re speaker but it still brings up the following error when doing so ..

> Collecting python3-numpy
  ERROR: Could not find a version that satisfies the requirement python3-numpy (from versions: none)
ERROR: No matching distribution found for python3-numpy

**** side note **** Installation of SLC still shows on my PI as v1.9.3 ? and /snipsLedControl_v1.9.3/installers/respeakers.sh doesn't seem to reflect the numpy changes you made yesterday according to your Git page. possibly the way its meant to be but thought id mention it,

**From this point i have now ...**
1. Removed SLC directories and service
2. Placed my snips.toml back to standard other than the customtts part
3. tested snips still works and is all good
4. downloaded today SLC from your git page and run all options as described above

doing a tail -f /var/log/syslog displays the original error of 
>import numpy as np
>Aug  1 15:44:39 snips python3[12785]: ModuleNotFoundError: No module named 'numpy'

So now adding 
>sudo apt-get purge -y python-numpy
>cd /home/pi/snipsLedControl_v1.9.3
>./venv/bin/pip3 uninstall -y python-numpy
>/venv/bin/pip3 --no-cache-dir install numpy`

and rebooting brings me back to one of my other original errors of a few days ago

> Importing the numpy c-extensions failed.

*takes a deepth breath * :)

The SLC debug log fails now when doing the suggested 
> ./venv/bin/python3.5 main.py --debug=True 

_comes up with error of_ 
>pi@snips:~ $ cd snipsLedControl_v1.9.3
>pi@snips:~/snipsLedControl_v1.9.3 $ sudo chown -R pi logs
>pi@snips:~/snipsLedControl_v1.9.3 $ ./venv/bin/python3.5 main.py --debug=True
>bash: ./venv/bin/python3.5: No such file or directory

looking in /venv/bin/ i only see reference to 3.7 items despite my pi having python 3.5.7 installed ??

I'm going to leave this novel there and try not to touch anything until i hear back as i'm not sure what else to do to get my snips lights working :) but hopefully i've given some feedback on debug info

PS: a full log of the above is attached
[full_SLC_log.txt](https://github.com/Psychokiller1888/snipsLedControl/files/3455176/full_SLC_log.txt)

Psychokiller1888, 2019-08-01:
Wow... Don't start burning on every ends but rather try to understand the issues. We've been talking about numpy a couple of posts just above yours as you mention it.

You have many problems which are easily understandable.
- Numpy doesn't install as SLC 1.9.3 doesn't have the fix for it, as I was waiting for confirmation for it to work. I have pushed 1.9.4 that now should install numpy properly though I have no confirmation it works on wide usage.
- You have an error stating python 3.5 doesn't exist so there's no need to try to start SLC with python 3.5. The installer checks and take the first python it finds to create the virtual environement, starting with python 3.7, then 3.6 and finally 3.5. If none is found, it will fail and tell you to install at least 3.5. You are using Python 3.7 because you either compiled it yourself or your are using Raspbian Buster. So instead of trying to start SLC with `./venv/bin/python3.5 main.py blablabla`try to start it with `./venv/bin/python3 main.py blablabla`

One step at a time trying to understand the failures is to only way to a proper debug when you know that many are using SLC with respeakers devices at this very moment on updated versions, including me.

So from here I'm gonna assume that:
- You have snips installed and running
- Your snips.toml is clean and is not indented with a mix of spaces and tabs as this is prolly the source of your toml error log
- You have deleted SLC whatever version(s) you have
- You have deleted the service file
- You have, or not, your respeaker device installed, which doesn't matter

From here:
- Download SLC using the wget method
- Install it, chosing your respeaker, the pattern you want and saying yes for respeaker install

Past that point, after a restart it should work

LazzaWA, 2019-08-01:
Thanks for the reply psycho. I wasn't trying to be "that annoying person" i was just trying to give info i thought that may help with my issue. :)

Your assumptions were all correct apart from the fact it's a raspberry pi on stretch not buster and i have python 3.5 installed not 3.7.

Unfortunately i still get the following errors

>Traceback (most recent call last):
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/core/__init__.py", line 17, in <module>
Aug  1 18:07:05 snips python3[1526]:     from . import multiarray
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/core/multiarray.py", line 14, in <module>
Aug  1 18:07:05 snips python3[1526]:     from . import overrides
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
Aug  1 18:07:05 snips python3[1526]:     from numpy.core._multiarray_umath import (
Aug  1 18:07:05 snips python3[1526]: ImportError: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so)
Aug  1 18:07:05 snips python3[1526]: During handling of the above exception, another exception occurred:
Aug  1 18:07:05 snips python3[1526]: Traceback (most recent call last):
Aug  1 18:07:05 snips python3[1526]:   File "main.py", line 127, in <module>
Aug  1 18:07:05 snips python3[1526]:     main()
Aug  1 18:07:05 snips python3[1526]:   File "main.py", line 112, in main
Aug  1 18:07:05 snips python3[1526]:     slc = SnipsLedControl(args)
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/models/SnipsLedControl.py", line 151, in __init__
Aug  1 18:07:05 snips python3[1526]:     self._ledsController = LedsController(self)
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/models/LedsController.py", line 46, in __init__
Aug  1 18:07:05 snips python3[1526]:     if not self.initHardware():
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/models/LedsController.py", line 104, in initHardware
Aug  1 18:07:05 snips python3[1526]:     self._interface = APA102(hardware=self._hardware, endFrame=self._hardware['endFrame'])
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/interfaces/apa102.py", line 35, in __init__
Aug  1 18:07:05 snips python3[1526]:     from libraries.seeedstudios.channel_picker import ChannelPicker
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/libraries/seeedstudios/channel_picker.py", line 8, in <module>
Aug  1 18:07:05 snips python3[1526]:     import numpy as np
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/__init__.py", line 142, in <module>
Aug  1 18:07:05 snips python3[1526]:     from . import core
Aug  1 18:07:05 snips python3[1526]:   File "/home/pi/snipsLedControl_v1.9.4/venv/lib/python3.7/site-packages/numpy/core/__init__.py", line 47, in <module>
Aug  1 18:07:05 snips python3[1526]:     raise ImportError(msg)
Aug  1 18:07:05 snips python3[1526]: ImportError:
Aug  1 18:07:05 snips python3[1526]: IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Aug  1 18:07:05 snips python3[1526]: Importing the numpy c-extensions failed.
 
and debug log of 

`Starting Snips Led Control v. 1.9.4
Initializing SnipsLedControl
Loading configurations
Loaded 12 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to ReSpeaker 2
- Using google as pattern with 3 leds
Initializing leds controller
Button 1 not implemented, override it in CustomLedPattern
Snips Led Control started
Button 1 not implemented, override it in CustomLedPattern

I'm only about 3 months into using raspberry pi's and learning linux commands and that sort of thing. I'm happy and capable of learning but i'm at a lost on the debug side of things if it's its not documented around the net :)

Will certainly shout a beer if you can point me in the right direction

Psychokiller1888, 2019-08-01:
I never said you were annoying ;) Just saying that one step after the other is better :)

For some reason, as you said, your venv shows traces of python3.7

So we gonna do this old school, manually to see where it hangs:

In your terminal, type the following:
`which python3.7`

If it returns something your have 3.7 running

Then try the same, but with 3.6
`which python3.6`

and 3.5
`which python3.5`

If you get a path, note it, for the first one you get. This tells you the path to any python you might have installed

Now, got to the snipsledcontrol directory:
```bash
cd /home/pi/SnipsLedControl_v1.9.4
sudo rm -rf venv
```

This removes the faulty virtual environement. Dunno if you are aware of it, but we use virtual envs for python, it copies the executable to you project and we add libraries to that copy, so you can have multiple dependencies that don't conflict across projects.
We now need to create back the venv:

`virtualenv -p /usr/bin/python3.XXXX /home/pi/SnipsLedControl_v1.9.4`

I expect you to replace the first path to the path you discovered using the `which`command and the SLC path. This will copy python to SLC directory and install a few needed things for it to run.

Now we need to activate that virtual env:

`./venv/bin/activate`

If it's ok, you'll see your terminal line prefixed with `venv`

Now let's try to install some dependencies to this venv:

`pip3 --no-cache-dir install numpy`

This should install numpy

Now, as further step, because we did this manually, you'll need to add the others deps:

```bash
pip3 --no-cache-dir install RPi.GPIO
pip3 --no-cache-dir install spidev
pip3 --no-cache-dir install gpiozero
pip3 --no-cache-dir install paho-mqtt
pip3 --no-cache-dir install toml
```

Now the deps are installed

Let's try to start SLC. Remember, your terminal line starts with `venv`? that means we are actively using the virtual environement, so we can simply:

`python3 main.py --hardware=respeaker2 --debug=True`

This will start SLC with respeaker 2 and google pattern

If you want to quit the venv, you can type `deactivate`. You'd then start SLC using

`./venv/bin/python3 main.py --hardware=respeaker2 --debug=True`

Follow all these steps and stop at the first error you encounter, don't continue, but post it here

LazzaWA, 2019-08-01:
awesome reply thanks. love to learn. Quick question though 
i did have 3.7 installed at one stage then i believe i uninstalled it and installed 3.5 . However i do get the below response to the which python commands

> pi@snips:~ $ which python3.7
/usr/local/bin/python3.7
pi@snips:~ $ which python3.6
pi@snips:~ $ which python3.5
/usr/bin/python3.5

Question is... for your comment on 
`virtualenv -p /usr/bin/python3.XXXX /home/pi/SnipsLedControl_v1.9.4`
 which address would i use cause theres a /usr/local/bin and a /usr/bin/ .... bare in mind i suspect 3.7 was un installed but has maybe left some references floating around ?

Psychokiller1888, 2019-08-01:
This shows that you tried to install python3.7 but not system wide.

I would use python3.5 in that case

LazzaWA, 2019-08-01:
ok thanks, is it worth me at a later stage to try something like `sudo apt-get remove python3.7 --purge` to get rid of maybe some redundant references ? anyway will continue with the above , thanks

Psychokiller1888, 2019-08-01:
You should try `sudo apt-get purge python3.7` which would get rid of it entirely, including configs and whatever else. `sudo apt-get remove xxxx`uninstalls the package but keeps everything related to it

Psychokiller1888, 2019-08-01:
And if you do purge python3.7 before trying to go further, run a `which python3.7` to make sure it's not there anymore. In which case the installer will work all nicely again

urbatecte, 2019-08-01:
Hi Psycho,

In install.sh

numpy appear within  apt-get install instead pip3 then to have a warning during install (E: Unable to locate package numpy)

"apt-get install -y git mosquitto mosquitto-clients portaudio19-dev numpy"

Rgds

Psychokiller1888, 2019-08-01:
The apt-get install is for those who have distros without wheels which makes numpy take approx. 30 minutes to install through pip

LazzaWA, 2019-08-01:
It didn't aera to make a venv folder as per below 

>pi@snips:~/snipsLedControl_v1.9.4 $ sudo rm -rf venv
pi@snips:~/snipsLedControl_v1.9.4 $ virtualenv -p /usr/bin/python3.5 /home/pi/snipsLedControl_v1.9.4
Running virtualenv with interpreter /usr/bin/python3.5
Already using interpreter /usr/bin/python3.5
Using base prefix '/usr'
New python executable in /home/pi/snipsLedControl_v1.9.4/bin/python3.5
Also creating executable in /home/pi/snipsLedControl_v1.9.4/bin/python
Installing setuptools, pip, wheel...
done.
pi@snips:~/snipsLedControl_v1.9.4 $ ./venv/bin/activate
**-bash: ./venv/bin/activate: No such file or directory**
pi@snips:~/snipsLedControl_v1.9.4 $ ls -l
total 120
drwxr-xr-x 2 pi pi    4096 Aug  1 19:35 bin
-rw-rw-r-- 1 pi root    32 Jul 31 05:34 _config.yml
-rw-rw-r-- 1 pi root  2440 Jul 31 05:34 hardware.json
drwxr-xr-x 2 pi pi    4096 Aug  1 19:33 include
drwxrwxr-x 2 pi root  4096 Jul 31 05:34 installers
-rwxrwxr-x 1 pi root  6062 Jul 31 05:34 install.sh
drwxrwxr-x 3 pi root  4096 Aug  1 18:04 interfaces
drwxrwxr-x 3 pi root  4096 Aug  1 18:04 ledPatterns
drwxr-xr-x 3 pi pi    4096 Aug  1 19:33 lib
drwxrwxr-x 4 pi root  4096 Aug  1 18:04 libraries
-rw-rw-r-- 1 pi root 35149 Jul 31 05:34 LICENSE
drwxr-xr-x 2 pi root  4096 Aug  1 18:04 logs
-rw-r--r-- 1 pi pi    2001 Aug  1 18:09 logs.log
-rw-rw-r-- 1 pi root  5779 Jul 31 05:34 main.py
drwxrwxr-x 3 pi root  4096 Aug  1 18:04 models
drwxrwxr-x 2 pi root  4096 Jul 31 05:34 MqttSpyScripts
drwxrwxr-x 2 pi root  4096 Jul 31 05:34 plugins
-rw-rw-r-- 1 pi root   911 Jul 31 05:34 README.md
-rw-rw-r-- 1 pi root   638 Jul 31 05:34 slc_download.sh
-rw-rw-r-- 1 pi root   281 Jul 31 05:34 snipsledcontrol.service
pi@snips:~/snipsLedControl_v1.9.4 $

Psychokiller1888, 2019-08-01:
` virtualenv -p /usr/bin/python3.5 /home/pi/snipsLedControl_v1.9.4/venv` ;)

LazzaWA, 2019-08-01:
thanks 
and as suspected 3.7 isnt there 

>pi@snips:~ $ sudo apt-get purge python3.7
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python3.7
E: Couldn't find any package by glob 'python3.7'
E: Couldn't find any package by regex 'python3.7'

LazzaWA, 2019-08-01:
i tried working out the next step before disturbing you again but im stuck again 

> pi@snips:~/snipsLedControl_v1.9.4 $ ./venv/bin/activate
-bash: ./venv/bin/activate: Permission denied

Psychokiller1888, 2019-08-01:
Try with

`source venv/bin/activate`

LazzaWA, 2019-08-01:
i got to this stage `pip3 --no-cache-dir install numpy` and then got flooded with a  multitude of red errors.
I'm not the guru here but i think i really need to get rid of all reference's to /usr/local/bin/python3.7 so that it uses /usr/bin/python3.5  , can i just delete that path ? or is it something ive read called a symlink that i can remove ?

Psychokiller1888, 2019-08-01:
and what are the errors saying?

LazzaWA, 2019-08-01:
[numpyFail.txt](https://github.com/Psychokiller1888/snipsLedControl/files/3456215/numpyFail.txt)

im just scrolling through it now to see. have attached a copy for reference

Psychokiller1888, 2019-08-01:
Hmmm, I hate numpy, always said so. Let's try to get rid of it:

```bash
sudo apt-get purge python-numpy
sudo apt-get purge python2-numpy
sudo apt-get purge python3-numpy
pip uninstall numpy
pip3 uninstall nump
```

Some of these will fail, I know, I'm just not sure what you have installed and what not anymore.

Psychokiller1888, 2019-08-01:
Also, it seems you don't have the python-dev package for some reason

`sudo apt-get install python3.5-dev`

I think you uninstalled the default python 3.5 from your stretch distro and when reinstalling you haven't installed the dev pacakge?

LazzaWA, 2019-08-01:
ok two of the numpy purge lines uninstalled stuff the other lines couldn't find packages so think that's clean now. The installation of the dev package im not sure, i just followed a git page on installing it but in the process now of doing a 
`sudo apt-get install python3.5-dev`

i think this stretch only came with 2.7 ? i didnt use buster cause it says snips doesnt work on buster yet according to the snips install page 

definately be doing more reading and learning and less cut and pasting once this is working :)

Psychokiller1888, 2019-08-01:
No, stretch comes with 2.7 and 3.5 :)

LazzaWA, 2019-08-01:
learning every minute :)

LazzaWA, 2019-08-01:
i managed to get through all those steps with no errors :) i dont seem to have any lights just yet but at least nothing broke

LazzaWA, 2019-08-01:
all good i changed python3 main.py --hardware=respeaker2 --debug=True to python3 main.py --hardware=respeaker4 --debug=True

Ok it's official you are a legend . thank you for your support tonight a teaching. i really appreciated it. Ive spent like 2 weeks trying to get that going on my own :)

closed 2019-08-01 by LazzaWA

Psychokiller1888, 2019-08-01:
Happy it worked out for you! And happy to be of help! Thank you so much ;)

LazzaWA, 2019-08-11:
just a FYI for other readers that have a similar issue to mine above. For me the headache came from having a local install of 3.7 which was later uninstalled but still showed up when doing a
`which python3.7`

i ended up renaming the python3.7 file in /usr/local/ by doing the following 
`sudo mv /usr/local/bin/python3.7 /usr/local/bin/python3.7.bak`
then checked that 
`python3 --version` returned only my python3.5 in /usr/bin/python3.5

and i now no longer have to go through the above steps everytime i break something that stops SnipsLEDcontrol script ( or update the script) :)

**On a separate note :**

Two questions for pyscho... :) 
1. is there a shortcut command to changing led patterns ? or is it a case of running the full install.sh ?

2. any chance you can make available the "ok" and "yes" sound wav's you use for your snips ? they seem to be sooo much clearer than anything i can find so far. The wav files i've tried end up sounding like shes done a line of crack or something :)

Psychokiller1888, 2019-08-11:
For sure not reinstalling! Just edit the .service file and change the pattern there, then reload the daemon

`sudo systemctl stop snipsledcontrol`

Edit the .service file and change the pattern there

```bash
sudo systemctl daemon-reload
sudo systemctl start snipsledcontrol
```

Re the sound files they'll be available in a couple of weeks with Project Alice public release

LazzaWA, 2019-08-11:
awesome thanks