project-alice-assistant/HermesLedControl #51: Install errors with 1.9.7 for Respeaker Mic Array V2

Platup, 2019-08-07:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
sudo ./slc_download.sh

**Led device name**
Respeaker Mic Array V2

**The bug**
Latest installer(s) 1.9.6 and 1.9.7 give the below chmod and sudo errors when performing the hardware install/config.
Note: This is on a pretty fresh Stretch install with only snips satellite service installed before installing SLC.

**Debug output**
Cut from the install output.  I can add more, but I think the applicable errors are shown.
...
Do you need to install / configure your respeakerMicArrayV2? This is strongly suggested as it does turn off services that might conflict as well!
1) yes
2) no
3) cancel
#? 1
chmod: cannot access 'pi': No such file or directory
chmod: cannot access '/home/pi/snipsLedControl_v1.9.7/venv': No such file or directory
############### Respeaker Mic Array V2 installation ########################
################ Please run this script with sudo ##########################
How many channels do you want to use?
1) 1
2) 6
#? 2
sudo: unknown user: bash
sudo: unable to initialize policy plugin
############################## All done! ##############################
############## Don't forget to turn on the SPI interface! #############
Finished installing Snips Led Control v1.9.7
You may want to copy over your custom led patterns to the new version

**To Reproduce**
Run the install option 3 for respeakerMicArrayV2

Psychokiller1888, 2019-08-07:
Hi!
Yep, this was fixed, I was waiting on confirmation of the fix which didn't come before releasing another version. Should be ok now, with 1.9.8

Platup, 2019-08-07:
I tried 1.9.8 and it got all the way to the end and then had a permission error.
FYI, for others installing, on a PiZero it took around 2 hours for the numpy install.

Do you need to install / configure your respeakerMicArrayV2? This is strongly suggested as it does turn off services that might conflict as well!
1) yes
2) no
3) cancel
#? 1
############### Respeaker Mic Array V2 installation ########################
################ Please run this script with sudo ##########################
How many channels do you want to use?
1) 1
2) 6
#? 2
bash: line 1: /home/pi/snipsLedControl_v1.9.8/venv/bin/activate: No such file or directory
Collecting click
  Downloading https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl (81kB)
    100% |████████████████████████████████| 81kB 207kB/s
Installing collected packages: click
Successfully installed click-7.0
Collecting pyusb
  Downloading https://www.piwheels.org/simple/pyusb/pyusb-1.0.2-py3-none-any.whl (57kB)
    100% |████████████████████████████████| 61kB 179kB/s
Installing collected packages: pyusb
Successfully installed pyusb-1.0.2
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/da/32/1b8f2bb5fb50e4db68543eb85ce37b9fa6660cd05b58bddfafafa7ed62da/numpy-1.17.0.zip (6.5MB)
    100% |████████████████████████████████| 6.5MB 1.0MB/s
Installing collected packages: numpy
  Running setup.py install for numpy ... done
Successfully installed numpy-1.17.0
Cloning into 'usb_4_mic_array'...
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 79 (delta 9), reused 15 (delta 7), pack-reused 59
Unpacking objects: 100% (79/79), done.
Traceback (most recent call last):
  File "dfu.py", line 229, in <module>
    main()
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "dfu.py", line 218, in main
    dev = XMOS_DFU()
  File "dfu.py", line 198, in __init__
    super(XMOS_DFU, self).__init__()
  File "dfu.py", line 71, in __init__
    devices = self.find()
  File "dfu.py", line 61, in find
    configuration = device.get_active_configuration()
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 875, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 236, in get_active_configuration
    self.managed_open()
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
############################## All done! ##############################
############## Don't forget to turn on the SPI interface! #############
Finished installing Snips Led Control v1.9.8
You may want to copy over your custom led patterns to the new version

LazzaWA, 2019-08-12:
I just bit the bullet and started with a fresh install of snips on my pi and ended up with the same error trying to install respeaker4. to reproduce.....
* i have a fresh install of stretch, 
* then did a sudo apt-get update and sudo apt-get upgrade
* Then installed NPM and GIt so that i could install Snips and sam.
* Then installed sam and snips as platform and then assistant.
* Didn't bother installing my respeaker via seedstudio process as i know the LED script does it too.

During the install of SnipsLEDcontrol v1.9.8........

Install errors worth noting are :

> Collecting virtualenv
  Downloading https://files.pythonhosted.org/packages/db/9e/df208b2baad146fe3fbe750eacadd6e49bcf2f2c3c1117b7192a7b28aec4/virtualenv-16.7.2-py2.py3-none-any.whl (3.3MB)
    100% |████████████████████████████████| 3.3MB 41kB/s
Installing collected packages: virtualenv
Successfully installed virtualenv-16.7.2
bash: line 2: virtualenv: command not found
bash: line 3: /home/pi/snipsLedControl_v1.9.8/venv/bin/activate: No such file or directory
Collecting RPi.GPIO

and further on in the install ( installing the respeaker part...

> ######################### Respeakers installation #########################
#################### Please run this script with sudo #####################
bash: line 1: /home/pi/snipsLedControl_v1.9.8/venv/bin/activate: No such file or directory
Collecting numpy
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.17.0-cp35-cp35m-linux_armv7l.whl (9.0MB)
    100% |████████████████████████████████| 9.1MB 1.1MB/s
Installing collected packages: numpy
Successfully installed numpy-1.17.0

After the install has finished i have no venv folder in /snipsLedControl_v1.9.8/ 
Just as well i can fix that from my previous issue :)

Hope that helps add some info to the install issue

LazzaWA, 2019-08-12:
FYI
In my instance it was because a fresh install of stretch appears not to come with "virtualenvwrapper"
Running `sudo pip install virtualenv virtualenvwrapper` and then reinstalling snipsLEDcontrol works 

Perhaps add a check in the install to check if virtualenvwrapper is installed and if not install it ??

Psychokiller1888, 2019-08-12:
Hmmm, that's weird! I'll check that, thank you!

Psychokiller1888, 2019-08-12:
You used it with sudo, maybe that's what I'm missing, maybe virtualenv should be pip installed with sudo, not sure, gotta do some researches

Platup, 2019-08-13:
Don't think it was supposed to be fixed yet, but I noticed an update, so I tried the install.
FYI, I'm still seeing the permissions issue when installing the Respeaker MicV2 on SLC 1.9.10

Do you need to install / configure your respeakerMicArrayV2? This is strongly suggested as it does turn off services that might conflict as well!
1) yes
2) no
3) cancel
#? 1
############### Respeaker Mic Array V2 installation ########################
################ Please run this script with sudo ##########################
How many channels do you want to use?
1) 1
2) 6
#? 2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting click
  Using cached https://files.pythonhosted.org/packages/fa/37/45185cb5abbc30d7257104c434fe0b07e5a195a6847506c074527aa599ec/Click-7.0-py2.py3-none-any.whl
Installing collected packages: click
Successfully installed click-7.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyusb
  Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.0.2-py3-none-any.whl
Installing collected packages: pyusb
Successfully installed pyusb-1.0.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.17.0-cp35-cp35m-linux_armv6l.whl (9.0MB)
     |████████████████████████████████| 9.1MB 794kB/s
Installing collected packages: numpy
Successfully installed numpy-1.17.0
Cloning into 'usb_4_mic_array'...
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 79 (delta 9), reused 15 (delta 7), pack-reused 59
Unpacking objects: 100% (79/79), done.
Traceback (most recent call last):
  File "dfu.py", line 229, in <module>
    main()
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "dfu.py", line 218, in main
    dev = XMOS_DFU()
  File "dfu.py", line 198, in __init__
    super(XMOS_DFU, self).__init__()
  File "dfu.py", line 71, in __init__
    devices = self.find()
  File "dfu.py", line 61, in find
    configuration = device.get_active_configuration()
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/core.py", line 875, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/core.py", line 236, in get_active_configuration
    self.managed_open()
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/pi/snipsLedControl_v1.9.10/venv/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
############################## All done! ##############################
############## Don't forget to turn on the SPI interface! #############
Finished installing Snips Led Control v1.9.10
You may want to copy over your custom led patterns to the new version

Platup, 2019-08-20:
FYI, I tried v1.9.11, but it still downloaded at v1.9.10.

I received the same error as above regarding the permissions.

Platup, 2019-08-30:
So it seems the issue with the USB Mic Array V2 is Python 3. At least that's the best I could figure out.

> pi@snips2:~/usb_4_mic_array $ sudo python3 dfu.py --download 6_channels_firmware.bin
> Traceback (most recent call last):
>   File "dfu.py", line 15, in <module>
>     import usb.core
> ImportError: No module named 'usb'
> pi@snips2:~/usb_4_mic_array $ sudo python dfu.py --download 6_channels_firmware.bin
> entering dfu mode
> found dfu device
> downloading
> 149056 bytes
> done

Psychokiller1888, 2019-09-02:
Well the error states no module named usb in your python 3 test

Platup, 2019-09-03:
Yeah, I was trying to determine where the install was failing.
This was from a base stretch install with only snips_satellite installed and letting SLC install the Mic Array V2.
pyusb get's installed, but when running under python3, it fails when trying to import the USB module.
If I run the same command with python it works fine.

Platup, 2019-09-04:
ok, so after:
sudo apt-get install python3-usb
sudo apt-get install python3-click
sudo apt-get install python3-numpy
I was able to get dfu.py to run with python3.

However, SLC still get's a permissions error when starting. 
A re-install after all of the above throws this error:
```
Collecting numpy
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.17.1-cp35-cp35m-linux_armv6l.whl (9.1MB)
     |████████████████████████████████| 9.1MB 708kB/s
Installing collected packages: numpy
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/home/pi/snipsLedControl_v1.9.11/venv/lib/python3.5/site-packages/numpy/__pycache__/_globals.cpython-35.pyc'
Consider using the `--user` option or check the permissions.
```
and then the install ends with the same error I've posted before.

Platup, 2019-09-19:
I've got some time to fiddle with this again.  
Is there a specific version of Stretch I should start with?  
Or should I try to alter permissions on something?

closed 2020-02-09 by Psychokiller1888