project-alice-assistant/HermesLedControl #52: ImportError: No module named 'matrix_lite'

Blackir62, 2019-08-18:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py --hardware=matrixvoice --pattern=google

**Led device name**
Matrix Voice

**The bug**
A clear and concise description of what the bug is.
I'm unable to have led ring functioning, I have ImportError: No module named 'matrix_lite' issue when lunch in debug mode.

**Debug output**
Follow instructions on https://github.com/Psychokiller1888/snipsLedControl/wiki/Run-in-debug-mode
And paste the output

pi@raspberrypi:~/snipsLedControl_v1.9.10 $ ./venv/bin/python3 main.py --hardware=matrixvoice --pattern=google --debug=True
Starting Snips Led Control v. 1.9.10
Initializing SnipsLedControl
Loading configurations
Loaded 12 hardware references
- Mqtt server set to localhost
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Matrix Voice
- Using google as pattern with 18 leds
Initializing leds controller
Traceback (most recent call last):
  File "main.py", line 127, in <module>
    main()
  File "main.py", line 112, in main
    slc = SnipsLedControl(args)
  File "/home/pi/snipsLedControl_v1.9.10/models/SnipsLedControl.py", line 152, in __init__
    self._ledsController = LedsController(self)
  File "/home/pi/snipsLedControl_v1.9.10/models/LedsController.py", line 46, in __init__
    if not self.initHardware():
  File "/home/pi/snipsLedControl_v1.9.10/models/LedsController.py", line 115, in initHardware
    from interfaces.matrixvoice import MatrixVoice
  File "/home/pi/snipsLedControl_v1.9.10/interfaces/matrixvoice.py", line 3, in <module>
    from matrix_lite import led
ImportError: No module named 'matrix_lite'

**To Reproduce**
Steps to reproduce the behavior
Install on a fresh raspberry pi 3 B+ raspbian stretch lite then snips with this tutorial https://www.hackster.io/matrix-labs/matrix-voice-and-matrix-creator-running-snips-ai-b48344
And install SLC with slc_download.sh matrix voice option and do you need to install your matrix voice to no.

Psychokiller1888, 2019-08-19:
The installer install a virtual environement, you cannot start with python 2 as you do and not without the venv either

`./venv/bin/python3 main.py --hardware=matrixvoice --pattern=google --debug=true`

Blackir62, 2019-08-19:
Thanks for specifying, I execute well the main.py with python3 and not python2. 
My mistake is to have forgotten that we were in a venv, I installed the module outside the venv. 
For those who might have the same concerns, here is the command to install the matrix_lite packet in the python3 venv.

cd snipsLedControl_v*******
./venv/bin/python3 -m pip install matrix-lite

closed 2019-08-19 by Blackir62