project-alice-assistant/HermesLedControl #55: High CPU aorund 28%

Toxminn, 2019-09-07:
ich habe das tool erfolgreich installiert. Leider benutzt python3 um die 28% für SnipsLedControll. Welche log datein muss ich auslesen? Danke im vorraus

Psychokiller1888, 2019-09-08:
I guess it's the DoA that eats up some cpu. Try to disable it for respeaker 4 in hardware.json

Toxminn, 2019-09-08:
Thank you :)
I have commented out DOA and now my cpu is not so busy anymore. Thanks

closed 2019-09-08 by Toxminn

markusressel, 2019-09-27:
@Psychokiller1888 Thx for this tip.

I am not sure if I have the same issue, but after a fresh install of snips and SLC the led animations were unusably slow and cpu load spiked to 100% on one core without decreasing ever again. After I removed the `"doa": "doa_respeaker_4mic_array",` line this did not happen anymore. LED animations are smooth and cpu load is near 0%.

Also in `journalctl -U snipsledcontrol` I had this line:
`arecord: main:788: Fehler beim Öffnen des Gerätes: Das Gerät oder die Ressource ist belegt`
which roughly translates to `arecord: main:788: error opening device: device or resource is busy`
which disappeared after I removed the `doa` line.

I can live without DOA, so this is not a huge issue for me, but the default behaviour leads to high load and through that *very* high device temperatures for me, which is not good. Let me know if I can provide anything to help debug this.

Psychokiller1888, 2019-09-27:
I will turn DoA off by default, better :)