project-alice-assistant/HermesLedControl #57: Support for Jabra Speak 410 (USB)

federico2015, 2019-10-11:
Thank for your work, is it possible to control this device?

Psychokiller1888, 2019-10-11:
? There's no led on that device

federico2015, 2019-10-11:
There are 12 LEDs, 1 LED (just over Jabra Logo) stay always white and I'd to use it for HotWord found feedback, when you switch mute ON all LEDs stay red.
There are also some buttons: Mute On/Off, Vol. +, Vol. -, Green Phone (answer), Red Phone (End Call), Jabra logo and Tringle (custom use).
[image]
[image]
I know that Jabra has SDK for connecting to it.
Hope this can help, thank you.

closed 2020-02-09 by Psychokiller1888