project-alice-assistant/HermesLedControl #69: Document how to actually use

probonopd, 2020-02-02:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I follow (copy- and-paste) instructions and then nothing works

**Describe the solution you'd like**
I follow the instructions on https://github.com/project-alice-assistant/HermesLedControl/wiki/Installation-&-update and then when I say "Snowboy" in Rhasspy the Neopixel ring animates blue like on an Alexa device

**Describe alternatives you've considered**
I followed the instructions (copy-and-paste) up until https://github.com/project-alice-assistant/HermesLedControl/wiki/Installation-&-update at which point the instructions become vague. What _exactly_ do I need to do so that when I say "Snowboy" in Rhasspy the Neopixel ring animates blue like on an Alexa device?

**Additional context**

This is _exactly_ what I did:

```
What assistant engine are you using?
1) projectalice
2) rhasspy
3) snips
4) cancel
#? 2
What's the path to your assistant config file?
./install.sh: line 42: read: `Path: (/etc/snips.toml)': not a valid identifier
Path: /etc/snips.toml
What device do you wish to control with SLC?
1) respeaker2				  6) matrixvoice			  11) googleAIY
2) respeaker4				  7) matrixcreator			  12) I'm using simple leds on GPIOs
3) respeakerMicArrayV2			  8) respeakerCoreV2			  13) don't overwrite existing parameters
4) neoPixelsSK6812RGBW			  9) respeaker6MicArray			  14) cancel
5) neoPixelsWS2812RGB			 10) respeaker7MicArray
#? 5
What pattern do you want to use?
1) google
2) alexa
3) projectalice
4) custom
5) kiboost
6) cancel
#? 2
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]                                                            
Get:3 https://download.docker.com/linux/raspbian buster InRelease [29.7 kB]                                                          
Hit:4 https://deb.nodesource.com/node_12.x buster InRelease                                                                          
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf Packages [274 kB]
Fetched 13.4 MB in 17s (765 kB/s)                                                                                                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u1).
portaudio19-dev is already the newest version (19.6.0-1).
The following additional packages will be installed:
  libev4 libmosquitto1 libuv1 libwebsockets8
Suggested packages:
  apparmor python-numpy-doc python3-pytest python3-numpy-dbg
The following NEW packages will be installed:
  libev4 libmosquitto1 libuv1 libwebsockets8 mosquitto mosquitto-clients python3-numpy
0 upgraded, 7 newly installed, 0 to remove and 65 not upgraded.
Need to get 2,358 kB of archives.
After this operation, 11.7 MB of additional disk space will be used.
Get:1 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libev4 armhf 1:4.25-1 [34.5 kB]
Get:2 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libmosquitto1 armhf 1.5.7-1+deb10u1 [57.6 kB]
Get:3 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster/main armhf libuv1 armhf 1.24.1-1 [96.7 kB]
Get:4 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libwebsockets8 armhf 2.0.3-3 [85.6 kB]
Get:5 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf mosquitto armhf 1.5.7-1+deb10u1 [143 kB]                 
Get:6 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf mosquitto-clients armhf 1.5.7-1+deb10u1 [66.8 kB]
Get:7 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf python3-numpy armhf 1:1.16.2-1 [1,873 kB]
Fetched 2,358 kB in 2s (1,296 kB/s)      
Selecting previously unselected package libev4:armhf.
(Reading database ... 103263 files and directories currently installed.)
[... 83 lines cut ...]
2) no
3) cancel
#? 1
######################### Neopixels installation ##########################
#################### Please run this script with sudo #####################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
swig is already the newest version (3.0.12-2).
0 upgraded, 0 newly installed, 0 to remove and 65 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi-ws281x
  Downloading rpi_ws281x-4.2.3-cp37-cp37m-linux_armv7l.whl (104 kB)
     |████████████████████████████████| 104 kB 1.8 MB/s 
Installing collected packages: rpi-ws281x
Successfully installed rpi-ws281x-4.2.3
###################################### All done! #######################################
##################### Don't forget to turn on the SPI interface! #######################
Created symlink /etc/systemd/system/multi-user.target.wants/hermesledcontrol.service → /etc/systemd/system/hermesledcontrol.service.
Finished installing Hermes Led Control v2.0.3
You may want to copy over your custom led patterns to the new version
(reverse-i-search)`ser': wget https://gist.githubu^Crcontent.com/Psychokiller1888/a9826f92c5a3c5d03f34d182fda1ce4c/raw/cbb53252dd55dc4e9f5f6064a493f0981cf133fb/hlc_download.sh
```

And now? Which pin do I have to attach the Neopixel ring to? Where am I supposed to know this from?

Psychokiller1888, 2020-02-02:
Attach it to whatever pin you want, usuall GPiO 10 is a good idea and is the one declared for neopixels... Or override the pin with --gpioPin argument...

And no, it's not done if you run it manually, the config sets up the service file, not what you are typing with your fingers....

Psychokiller1888, 2020-02-02:
I guess your debug command would be much better like this: ./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=neoPixelsWS2812RGB --gpioPin=10 --leds=12 --pattern=alexa

probonopd, 2020-02-03:
Looks like it is also exiting immediately.

```
pi@raspberrypi:~/hermesLedControl_v2.0.3 $  ./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=neoPixelsWS2812RGB --gpioPin=10 --leds=12 --pattern=alexa
Starting Hermes Led Control v. 2.0.3
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
No Rhasspy config found but debug mode, allow to continue
pi@raspberrypi:~/hermesLedControl_v2.0.3 $ 
```

probonopd, 2020-02-03:
Examining it with `strace` I can see that it is trying to load a file `/etc/snips.toml` which I do not have (I do not have snips at all - will not be relevant anymore given that snips is factually closed down by Sonos).

Psychokiller1888, 2020-02-03:
Well, because you told it so:

[image]

Maybe use --pathToConfig argument to change this to the rhasspy config file?

probonopd, 2020-02-03:
> Well, because you told it so

Well, I just pressed enter assuming the default would work:

```
pi@raspberrypi:~ $ sudo ./hlc_download.sh
(...)

What assistant engine are you using?
1) projectalice
2) rhasspy
3) snips
4) cancel
#? 2
What's the path to your assistant config file?
Path: (For Snips and ProjectAlice = /etc/snips.toml)
```

I don't have a config file (that I know of - and if I do it is hidden in some Docker container). What do I need to enter here?

Psychokiller1888, 2020-02-03:
I'm not using rhasspy, but a rapid check on their website gave this: `/.config/rhasspy/profiles/en/profile.json`

probonopd, 2020-02-03:
This works for me. _It would be cool if this would be documented._

```
pi@raspberrypi:~/hermesLedControl_v2.0.4 $ ./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=neoPixelsWS2812RGB --gpioPin=10 --leds=12 --pattern=alexa --pathToConfig=/home/pi/.config/rhasspy/profiles/de/profile.json
Starting Hermes Led Control v. 2.0.4
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
Loaded 12 hardware references
- Mqtt server set to 192.168.0.10
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Adafruit NeoPixels WS2812 RGB
- Using alexa as pattern with 12 leds
Using pin #10
Initializing leds controller
Hermes Led Control started
```

When I say the wakeword, the animation starts. But it never seems to stop, neither when I say a correct (working) command nor when I say garbage.

Log says:

```
pi@raspberrypi:~/hermesLedControl_v2.0.4 $ ./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=neoPixelsWS2812RGB --gpioPin=10 --leds=12 --pattern=alexa --pathToConfig=/home/pi/.config/rhasspy/profiles/de/profile.json
Starting Hermes Led Control v. 2.0.4
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
Loaded 12 hardware references
- Mqtt server set to 192.168.0.10
- Mqtt port set to 1883
- Client id set to default
- Hardware set to Adafruit NeoPixels WS2812 RGB
- Using alexa as pattern with 12 leds
Using pin #10
Initializing leds controller
Hermes Led Control started
On think triggered
On error received but it wasn't for me
On think triggered
On error received but it wasn't for me
On think triggered
On think triggered
On error received but it wasn't for me
```

Psychokiller1888, 2020-02-03:
Looks like your assistant is not ok, or doesn't communicate on hermes protocol correctly

Psychokiller1888, 2020-02-04:
I'm not gonna try Rhasspy, sorry, I'm the main dev of ProjectAlice. Was using Snips since very long and can still use it for Project Alice in parts....

probonopd, 2020-02-04:
Thanks, I didn't know about ProjectAlice yet. Will look into it. Anyhow, the original issue has been solved for me, although I think it would be nice to have it documented a bit more prominently (e.g., in the Wiki).

closed 2020-02-04 by probonopd