project-alice-assistant/HermesLedControl #71: Misunderstanding Question "What assistant engine are you using?"

Pittermaennchen, 2020-03-27:
First of all: **Thank you very much for this HermesLEDControl! It's great!**

**Led device name**
**Respeaker 4-Mic-Array**

**The bug**
A clear and concise description of what the bug is.
**It's not a bug. Only that I do not understand this question when it is asked: "What assistant engine are you using?
1) projectalice
2) rhasspy
3) snips
4) cancel"

Point is: I have snips installed and want to use at least the "hey snips"-hotword. But I have also Rhasspy installed and want to use it to understand my intents and send it to home assistant for handling.

So: Maybe there could be a slightly better explanation for noobs like me?
**

Psychokiller1888, 2020-03-27:
You're welcome!
Your engine is then Rhasspy. But you'll need hermes support for it, as I know it's not yet officially supported.

maxbachmann, 2020-03-27:
@Pittermaennchen Yes Rhasspy will support Hermes starting v2.5. Right now there is only a pre-release for v2.5 https://community.rhasspy.org/t/rhasspy-2-5-pre-release/643/56.
For HermesLedControl it is only relevant that Hermes is used

closed 2020-04-07 by Psychokiller1888