project-alice-assistant/HermesLedControl #81: Docker support

Daenara, 2020-07-18:
**Is your feature request related to a problem? Please describe.**
I build my smart home system in docker containers so I can easily move them to more powerful hardware if needed. Since this doesn't have an official docker file I wrote one. 

**Describe the solution you'd like**
I would like it if this project somehow incorporates the docker file, either by integrating it, or leaving this issue here for others to find if they want to run this in a docker. My Dockerfile might not be perfect but it runs.

**Additional context**
I used the download script and the install script to write this so it always gets the latest release. I did omit the version number in the folder name and the removal of the last version because the docker is always built from scratch anyway. I did not test if the GPIO led support works but in theory it should, because all the devices I encountered so far for GPIO are in the docker-compose and the user I add for the docker gets the gpio group.

Dockerfile:
```
FROM python:latest

ARG USER=docker
ARG FOLDER=/home/$USER
ARG VENV=venv
ARG FOLDER_NAME=hermesLedControl

RUN set -ex && adduser --uid 1000 --disabled-password --gecos '' $USER && \
	addgroup --gid 997 --system gpio && \
	addgroup --gid 999 --system spi && \
	usermod -a -G gpio,spi $USER

WORKDIR /home/docker
RUN set -ex && \
	export VERSION=$(curl --silent "https://api.github.com/repos/project-alice-assistant/HermesLedControl/releases/latest" | grep -Po '"tag_name": "\K.*?(?=")') && \
	apt-get update && apt-get install -y \
	git \
	mosquitto \
	mosquitto-clients \
	portaudio19-dev \
	python3-numpy \
	python3-pip && \
	rm -rf /var/lib/apt/lists/* && \
	pip3 --no-cache-dir install virtualenv && \
	mkdir -p ${FOLDER_NAME} && \
	wget $(curl --silent "https://api.github.com/repos/project-alice-assistant/HermesLedControl/releases/latest" | grep -Po '"tarball_url": "\K.*?(?=")') && \
	tar -xzf ${VERSION} -C ${FOLDER_NAME} --strip-components=1 && \
	rm ${VERSION} && \
	mkdir -p ${FOLDER}/${FOLDER_NAME}/logs && \
	chown -R ${USER} ${FOLDER}/${FOLDER_NAME}

USER $USER

RUN set -ex && \
	virtualenv -p python3 ${FOLDER}'/'${FOLDER_NAME}'/'${VENV}

ENV PATH="${FOLDER}'/'${FOLDER_NAME}'/'${VENV}/bin:$PATH"

RUN	pip3 --no-cache-dir install RPi.GPIO && \
	pip3 --no-cache-dir install spidev && \
	pip3 --no-cache-dir install gpiozero && \
	pip3 --no-cache-dir install paho-mqtt && \
	pip3 --no-cache-dir install toml
	
WORKDIR ${FOLDER}/${FOLDER_NAME}

ENTRYPOINT python3 main.py ${HLC_ARGUMENTS}
```

docker-compose integration:

```
version: '3'
services:
  hermes-led:
    container_name: hermes-led
    build: 
      context: ./hermes-led/
      dockerfile: ./Dockerfile
    volumes:
     - /opt/smart-home/rhasspy_2.5/profiles/de:/tmp/rhasspy
    environment:
      - TZ=Europe/Berlin
      - HLC_ARGUMENTS=--hardware=respeaker4 --pathToConfig=/tmp/rhasspy/profile.json --engine=rhasspy
    devices:
      - /dev/gpiomem:/dev/gpiomem
      - /dev/mem:/dev/mem
      - /dev/spidev0.0:/dev/spidev0.0
      - /dev/spidev0.1:/dev/spidev0.1
    restart: unless-stopped
    privileged: true
    network_mode: host
```

JonahKr, 2020-07-19:
You could save a lot of container size by using one of the slim images.
https://hub.docker.com/_/python
3.8-slim for example

Daenara, 2020-07-28:
I did that while testing but it took 3 times longer just to install the apt packages needed so I decided I personally don't have a space issue and I prefer it to be built faster instead.

michelcve, 2021-02-14:
Based on @Daenara 's Dockerfile, I created an alternative, based on Alpine Linux. Tested on a Raspberry Pi Zero W with Respeaker 2 and Rhasspy.

```
FROM arm32v6/python:3.9.1-alpine3.12

ARG USER=docker
ARG FOLDER=/home/$USER
ARG VENV=venv
ARG FOLDER_NAME=hermesLedControl

RUN set -ex && adduser --uid 1000 --disabled-password --gecos '' $USER && \
    delgroup ping && \
    addgroup --gid 997 --system gpio && \
    addgroup --gid 999 --system spi && \
    addgroup $USER gpio && \
    addgroup $USER spi

WORKDIR /home/docker
RUN set -ex && \
        apk add --no-cache \
        curl \
        grep && \
        export VERSION=$(curl --silent "https://api.github.com/repos/project-ali                                                                                                             ce-assistant/HermesLedControl/releases/latest" | grep -Po '"tag_name": "\K.*?(?=                                                                                                             ")') && \
        apk update && apk add \
        git \
        mosquitto \
        mosquitto-clients \
        portaudio-dev && \
        rm -rf /var/lib/apt/lists/* && \
        pip3 --no-cache-dir install virtualenv && \
        mkdir -p ${FOLDER_NAME} && \
        wget $(curl --silent "https://api.github.com/repos/project-alice-assista                                                                                                             nt/HermesLedControl/releases/latest" | grep -Po '"tarball_url": "\K.*?(?=")') &&                                                                                                              \
        tar -xzf ${VERSION} -C ${FOLDER_NAME} --strip-components=1 && \
        rm ${VERSION} && \
        mkdir -p ${FOLDER}/${FOLDER_NAME}/logs && \
        chown -R ${USER} ${FOLDER}/${FOLDER_NAME}

RUN set -ex && \
        apk add --no-cache \
        gcc \
        musl-dev \
        linux-headers

USER $USER

RUN set -ex && \
        virtualenv -p python3 ${FOLDER}'/'${FOLDER_NAME}'/'${VENV}

ENV PATH="${FOLDER}'/'${FOLDER_NAME}'/'${VENV}/bin:$PATH"

RUN     pip3 --no-cache-dir install RPi.GPIO && \
        pip3 --no-cache-dir install spidev && \
        pip3 --no-cache-dir install gpiozero && \
        pip3 --no-cache-dir install paho-mqtt && \
        pip3 --no-cache-dir install toml

WORKDIR ${FOLDER}/${FOLDER_NAME}

ENTRYPOINT python3 main.py ${HLC_ARGUMENTS}
```

JonahKr, 2021-02-14:
@michelcve out of curiosity: whats your container image size? Total or even for each step?

marecabo, 2021-04-10:
Thanks a lot for these Dockerfiles! Is it possible to use them with `--enableDOA=True` in theory?