project-alice-assistant/HermesLedControl #83: Syntax error in HermesLedControl.py

jyvern, 2020-11-14:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2 --pattern=google --leds=12 --pathToConfig=/home/respeaker/.config/rhasspy/profiles/fr/profile.json

**Led device name**
respeakerCoreV2

**The bug**
It simply stopped

**Debug output**
_Traceback (most recent call last):
  File "main.py", line 8, in <module>
    from models.HermesLedControl import HermesLedControl
  File "/home/respeaker/hermesLedControl_v2.0.9/models/HermesLedControl.py", line 594
    self._logger.warning(f"Missing white channel in '{attributeName}' attribute (RGBW format), appending default value")
                                                                                                                      ^
SyntaxError: invalid syntax_

**To Reproduce**
Just launch the command

Psychokiller1888, 2020-11-14:
Hmmm, wrong version of python?

jyvern, 2020-11-14:
Python 3.5.3

Psychokiller1888, 2020-11-14:
You need python 3.7

jyvern, 2020-11-20:
Ok, I've had some difficulties to update Respeaker Core V2 to python 3.7, but it is now ok. I still fail to make it working.
Hera are the debug traces where I made in bold 2 things which should pay attention 

```
Starting Hermes Led Control v. **2.0.6**
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
Loaded 14 hardware references
- Mqtt server set to xxxxxx
- Mqtt port set to 1883
- Mqtt username set to xxxxxx
- Mqtt password set to "hidden"
- Client id set to rhasat1
- Hardware set to Respeaker Core v2
- Using google as pattern with 12 leds
Initializing leds controller
**Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined**
Hermes Led Control started
On leds toggle triggered
On leds toggle triggered
On leds toggle off triggered
On leds toggle on triggered
```

Psychokiller1888, 2020-11-20:
Try to install mraa in the hlc venvSent from my Galaxy

closed 2021-02-05 by Psychokiller1888