project-alice-assistant/HermesLedControl #84: Unable to start un rhasspy system

LeducH, 2020-11-21:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py xxxxxxxxx
```
./venv/bin/python3 main.py --engine=rhasspy --pathToConfig=/home/pi/.config/rhasspy/profiles/en/profile.json --hardware=respeaker2 --pattern=google
```

**Led device name**
The name of the device such as "ReSpeaker 2" 

**The bug**
The script does not start 
**Debug output**

```
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
Error loading configurations: 'host'
```

**To Reproduce**
try to run start the `main.py`

```
cat /home/pi/.config/rhasspy/profiles/en/profile.json
{
    "dialogue": {
        "system": "rhasspy"
    },
    "intent": {
        "system": "fsticuffs"
    },
    "microphone": {
        "pyaudio": {
            "device": "0"
        },
        "system": "pyaudio"
    },
    "mqtt": {
        "enabled": "true"
    },
    "sounds": {
        "aplay": {
            "device": "sysdefault:CARD=seeed2micvoicec"
        },
        "error": "/profiles/wav/error_bleep.wav",
        "recorded": "/profiles/wav/google_pixelwakeupLow.wav",
        "system": "aplay",
        "wake": "/profiles/wav/google_pixelwakeup2.wav"
    },
    "speech_to_text": {
        "system": "kaldi"
    },
    "text_to_speech": {
        "marytts": {
            "voice": "dfki-spike-hsmm"
        },
        "nanotts": {
            "language": "en-GB"
        },
        "picotts": {
            "language": "en-US"
        },
        "system": "picotts"
    },
    "wake": {
        "porcupine": {
            "keyword_path": "bumblebee.ppn"
        },
        "system": "porcupine"
    }
```

philipp2310, 2020-11-21:
Hermes is an MQTT-based protocol, so it needs a host name for connecting to the mqtt-broker. Did rhasspy change their config structure maybe?
```    "mqtt": {
        "enabled": "true"
    },```
should have more fields for at least the host.

Of course it would be possible to assume if there is no host, take localhost, but somebody from rhasspy would need to do that, as I don't know about that.

Psychokiller1888, 2020-11-21:
Try to add the --host argument if rhasspy changed the config structure

LeducH, 2020-11-21:
@Psychokiller1888  Do you mean in this form?
```
...
    "system": "porcupine",
    "host": "localhost"
}
```

Psychokiller1888, 2020-11-21:
No idea what that is....

I mean by adding the argument to the start command as in `--host localhost`

LeducH, 2020-11-21:
for the command 
```
./venv/bin/python3 main.py --engine=rhasspy --pathToConfig=/home/pi/.config/rhasspy/profiles/en/profile.json --host localhost --hardware=respeaker2 --pattern=google 
```
I get an error with 

```
usage: Hermes Led Control [-h] [--engine {projectalice,rhasspy,snips}]
                          [--pathToConfig PATHTOCONFIG]
                          [--mqttServer MQTTSERVER] [--mqttPort MQTTPORT]
                          [--mqttUsername MQTTUSERNAME]
                          [--mqttPassword MQTTPASSWORD] [--clientId CLIENTID]
                          [--hardware {respeaker2,respeaker4,respeakerMicArrayV2,respeakerMicArrayV1,respeakerCoreV2,respeaker6MicArray,respeaker7MicArray,matrixvoice,matrixcreator,neoPixelsSK6812RGBW,neoPixelsWS2812RGB,googleAIY,puregpio,dummy}]
                          [--leds LEDS]
                          [--defaultBrightness DEFAULTBRIGHTNESS]
                          [--endFrame ENDFRAME]
                          [--pattern {google,alexa,kiboost,projectalice,pgas,custom}]
                          [--offListener {hermes/hotword/toggleOn,hermes/tts/sayFinished,hermes/audioServer/playFinished}]
                          [--enableDoA ENABLEDOA]
                          [--startPattern STARTPATTERN]
                          [--stopPattern STOPPATTERN]
                          [--offPattern OFFPATTERN]
                          [--idlePattern IDLEPATTERN]
                          [--wakeupPattern WAKEUPPATTERN]
                          [--speakPattern SPEAKPATTERN]
                          [--thinkPattern THINKPATTERN]
                          [--listenPattern LISTENPATTERN]
                          [--errorPattern ERRORPATTERN]
                          [--successPattern SUCCESSPATTERN]
                          [--updatingPattern UPDATINGPATTERN]
                          [--callPattern CALLPATTERN]
                          [--setupModePattern SETUPMODEPATTERN]
                          [--conErrorPattern CONERRORPATTERN]
                          [--messagePattern MESSAGEPATTERN]
                          [--dndPattern DNDPATTERN] [--defaultState {on,off}]
                          [--gpioPin GPIOPIN] [--vid VID] [--pid PID]
                          [--matrixIp MATRIXIP] [--everloopPort EVERLOOPPORT]
                          [--pureGpioPinout PUREGPIOPINOUT]
                          [--activeHigh ACTIVEHIGH] [--debug DEBUG]
Hermes Led Control: error: unrecognized arguments: --host localhost
```

Psychokiller1888, 2020-11-21:
Yeah, so use the argument --mqttServer....

LeducH, 2020-11-21:
Then I get same error with `host`

```
[pi@raspberrypi hermesLedControl_v2.0.9]$ ./venv/bin/python3 main.py --engine=rhasspy --pathToConfig=/home/pi/.config/rhasspy/profiles/en/profile.json --mqttServer localhost --mqttPort 1883 --hardware=respeaker2 --pattern=google
Starting Hermes Led Control v. 2.0.6
Initializing HermesLedControl
Initializing Rhasspy settings
Loading configurations
Error loading configurations: 'host'
```

Psychokiller1888, 2020-11-21:
So rhasspy changed something in your conf file as it is missing the mqtt host info

Psychokiller1888, 2020-11-21:
#TODO
```
- Fallback to some values if engine changes format/structure/data (models/engines/*
```

KiboOst, 2020-11-28:
In rhasspy UI, settings, MQTT
try to put the host IP instead of localhost

I had to do so to get HLC running fine.

Xenira, 2021-01-14:
It seems rhasspy removes the default values from the profile on saving changes.

It would be great if this project would fall back to localhost:1883 when no host / port is present in the profile json.

Psychokiller1888, 2021-02-05:
I think this was fixed

closed 2021-02-05 by Psychokiller1888