project-alice-assistant/HermesLedControl #92: Led stripes  how ti connect

dwarfwalker, 2021-01-26:
Hello, im new ti gpio, i can use 2 led stripes (chinese)
How i can connect ti gpio?
I have build a magic mirror, whit homeassistant, rhasspy, nodered and mqtt, i want 2 side of mirror whit led stripes, looks likes Alexa animation.
Thanks

Xenira, 2021-02-04:
Could you give some details about the led stripe?
Which chipset does it use? One or two data lines?
Do you want to connect the led stripe directly to the Pi or do you want to use something like a nodemcu? If so you might want to have a look at https://esphome.io/components/light/fastled.html

Please keep in mind, that the stripes might need an additional power supply.

I could imagine that something linke fastled (or python equivalent) could be used to address the led strip.
I might look into that.

Also, please stick to the issue templates.

dwarfwalker, 2021-02-05:
> Could you give some details about the led stripe?
> Which chipset does it use? One or two data lines?
> Do you want to connect the led stripe directly to the Pi or do you want to use something like a [...]

The led strips are 2, only 1 color, so whit only + and - connect cable, i found another magic mirror project where the led strip are connected whit Power supply and a mosfet irfz44n to Raspberry gpio likes this:
[image]
Can It work whit HermesLedControl?
Thx

Psychokiller1888, 2021-02-05:
You connect to the gpio of your choice and select the option "gpio leds" in the installer

closed 2021-04-11 by Psychokiller1888