project-alice-assistant/HermesLedControl #93: sed: can’t read /home/pi/.config/hermesLedControl/configuration.yml from install script

jazzy-r, 2021-01-30:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
Via the hlc_download.sh script using sudo permissions after using chmod +x as stated in the documentation

**Led device name**
Respeaker 4

**The bug**
When the installer is used using sudo permissions as stated in the documentation it fails to create the configuration file.

**Debug output**
sed: can’t read /home/pi/.config/hermesLedControl/configuration.yml

**To Reproduce**
Raspberry pi4 with respeaker 4

sudo chmod +x hlc_download.sh
sudo ./hlc_download.sh

jazzy-r, 2021-02-01:
https://community.rhasspy.org/t/hermesledcontrol-cannot-create-configuration-yml-file/2279

Added how I resolved this here.

Psychokiller1888, 2021-02-03:
@Xenira This seems to be linked to your recent changes adding a config yaml?

Xenira, 2021-02-04:
Ill have a look

Xenira, 2021-02-04:
Sorry about that. The config directory was already present while testing.
Added the missing mkdir.

PR #94 (2021-02-05): Fix config directory #93
Added missing mkdir to create config directory
Resolves #93
closed 2021-02-05 by Psychokiller1888