project-alice-assistant/HermesLedControl #95: please help me with this

coolkkmirror, 2021-02-04:
pi@raspberrypi:~ $ nano /home/pi/.asoundrc
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ sudo systemctl enable assistant.service
(env) pi@raspberrypi:~ $ sudo systemctl start assistant.service
(env) pi@raspberrypi:~ $ sudo systemctl status assistant.service
● assistant.service - Google Assistant
   Loaded: loaded (/lib/systemd/system/assistant.service; enabled; vendor preset
   Active: failed (Result: exit-code) since Fri 2021-02-05 02:23:24 IST; 5s ago
  Process: 18818 ExecStart=/bin/bash /home/pi/start_assistant.sh (code=exited, s
 Main PID: 18818 (code=exited, status=1/FAILURE)

Feb 05 02:23:24 raspberrypi bash[18818]:     **_remove_self(locals()))
Feb 05 02:23:24 raspberrypi bash[18818]:   File "/home/pi/env/lib/python3.7/site
Feb 05 02:23:24 raspberrypi bash[18818]:     samplerate)
Feb 05 02:23:24 raspberrypi bash[18818]:   File "/home/pi/env/lib/python3.7/site
Feb 05 02:23:24 raspberrypi bash[18818]:     info = query_devices(device)
Feb 05 02:23:24 raspberrypi bash[18818]:   File "/home/pi/env/lib/python3.7/site
Feb 05 02:23:24 raspberrypi bash[18818]:     raise PortAudioError('Error queryin
Feb 05 02:23:24 raspberrypi bash[18818]: sounddevice.PortAudioError: Error query
Feb 05 02:23:24 raspberrypi systemd[1]: assistant.service: Main process exited, 
Feb 05 02:23:24 raspberrypi systemd[1]: assistant.service: Failed with result 'e

Xenira, 2021-02-04:
Are you sure, this is the right project?
This project is about controlling LEDs for the voice assistant only.

Also please stick to the issue templates.

coolkkmirror, 2021-02-04:
i want to start google assistant on magicmirror
can u healp me?

coolkkmirror, 2021-02-04:
> Are you sure, this is the right project?
> This project is about controlling LEDs for the voice assistant only.
> 
> Also please stick to the issue templates.

i got new error on 

● assistant.service - Google Assistant
   Loaded: loaded (/lib/systemd/system/assistant.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-02-05 02:37:33 IST; 1min 18s ago
  Process: 29081 ExecStart=/bin/bash /home/pi/assistant-sdk-python/google-assistant-sdk/googlesamples/assistant/library/hotword.sh (code=exited, status=127)
 Main PID: 29081 (code=exited, status=127)

Feb 05 02:37:33 raspberrypi systemd[1]: Started Google Assistant.
Feb 05 02:37:33 raspberrypi bash[29081]: /bin/bash: /home/pi/assistant-sdk-python/google-assistant-sdk/googlesamples/assistant/library/hotword.sh: No such file or directory
Feb 05 02:37:33 raspberrypi systemd[1]: assistant.service: Main process exited, code=exited, status=127/n/a
Feb 05 02:37:33 raspberrypi systemd[1]: assistant.service: Failed with result 'exit-code'.

Xenira, 2021-02-04:
You probably want to ask your question here: https://github.com/googlesamples/assistant-sdk-python
As already stated this project is only about controlling LEDs on devices like the respeaker.
This has nothing to do with the voice assistant project you are using.

coolkkmirror, 2021-02-04:
> You probably want to ask your question here: https://github.com/googlesamples/assistant-sdk-python
> As already stated this project is only about controlling LEDs on devices like the respeaker.
> This has nothing to do with the voice assistant project you are using.

thanks. if i get some problem ill help to you. thanks again

closed 2021-02-05 by Psychokiller1888