project-alice-assistant/HermesLedControl #100: MQTT port broken since v2.0.13

urbatecte, 2021-04-10:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
Default launch
ExecStart=/home/pi/hermesLedControl_v2.0.13/venv/bin/python3 main.py --hermesLedControlConfig=/home/pi/.config/hermesLedControl/configuration.yml

**Led device name**
"ReSpeaker 2" and or "ReSpeaker 2"

**The bug**
new Rhasspy.py file change MQTT port to 12183 that may match to the internal Rhasspy MQTT one but not for external one.
I had to change it manually to get back to the 1883.
Another issue I've find out is, still in the Rhasspy.py file the Rhasspy profile is hard coded and by the way is pointing to a english profile where as the link to the Rhappy profile is asked during HLC config.
After updating both MQTT port and link to Rhasspy profile (form my case /home/pi/.config/rhasspy/profiles/fr/profile.json) everything work flawlessly !

**Debug output**
Unfortunatly I don't have this output anymore but 1st issue n debug was the link to Rhasspy profile that lead to an error.
2nd one was the MQTT port that lead to unable join MQTT server error.

**To Reproduce**
Steps to reproduce the behavior
In the Rhasspy config, set a external MQTT with port different than 12183.
In Rhasspy as well, select a profle different than 'en'

Psychokiller1888, 2021-04-11:
Ok, I need some explanation here, as I have never used Rhasspy...

- What is the profile file doing?
- What is the default mqtt port rhasspy is using?

If you have some non common params, why don't you override the port with the arguments?

urbatecte, 2021-04-11:
The profle.json give the Rhasspy params. 
The Rhasspy.py in HLC parse it to get the appropriate param (instead of launchng HLC with dedicated param).
I rely on the issue I've open on Rhasspy side to get back to you with a PR (t will be my 1st one !
In between I will have a look regarding the profile path hardcoded instead of usng the requested path during HLC config.
Cheers

kevinbischof, 2021-07-10:
I'm facing the same issue in point of mqtt server. I'm also using rhasspy. My MQTT port is also 1883 and not 12183.
It would be good if a custom port could be set for example in the configuration.yml.

Chreece, 2021-09-27:
> I'm facing the same issue in point of mqtt server. I'm also using rhasspy. My MQTT port is also 1883 and not 12183.
> It would be good if a custom port could be set for example in the configuration.yml.

I hope you have already resolved this, since there is an option to add Mqtt Port in configuration.yml
But I haven't yet figured out why the port given in rhasspy isn't the one that HermesLedControl is using.
@Psychokiller1888 can you find out why mqtt user and password are extracted from profile.json but not the port?
I also use a non en profile in rhasspy...

Chreece, 2021-09-27:
ok the problem is on rhasspy's site... I just saw that there wasn't any port saved in profile.json but in the web ui there was the 1883 standing... Maybe that's the default for rhasspy and not needed to be saved in profile.json?

Psychokiller1888, 2021-09-27:
I cannot find it out, I'm not using Rhasspy.... But 1883 is the default mqtt port. My problem is that the guy that implemented Rhasspy configs, implemented it with another default port:

`configs['mqttPort'] = conf['mqtt'].get('port', 12183)`

So, what does Rhasspy use?

donburch888, 2021-10-03:
I have seen both 1883 and 12183 used by Rhasspy.  Docker installs seem to use 12183, and also when rhasspy is booting (but not restarting).  I don't see advantage to adding extra processing on a dedicated low-powered satellite, so I have personally done the Debian install which seems to use 1883. 

I note that Rhasspy installs and enables mosquitto service (on port 12183) at the beginning of its initial configuration; and mosquitto service remains enabled even after the user changes to use an External MQTT server ... leading me to wonder if this was an oversight; or does rhasspy still use mosquitto on localhost internally even after an External MQTT is defined ?

Yes, in Rhasspy's web interface under MQTT section, the MQTT port field displays default value 1883 - but it doesn't get written to profile.json. 
I have manually added Port: 1883 to mqtt section in Rhasspys ~/.config/rhasspy/profiles/en/profile.json
`    "mqtt": {
        "enabled": "true",
        "host": "192.168.1.98",
        "port": "1883",
        "username": "rhasspy"
        "password": "passwd",
        "site_id": "sat-2",
    },`

closed 2022-05-30 by philipp2310