project-alice-assistant/HermesLedControl #102: On play finished received but it wasn't for me

kuumaur, 2021-04-23:
Iam using HLC together with Rhasspy 2.5.10.
This Pi3a+ is configured as satellite.
GPIO 14/15/18 are connected to simple LED's.
I experimented with clientId, but it seems to have no effect.
What Iam doing wrong?

The LED's are not working as of pattern, the following messages in debug output:
2021-04-23 11:18:52,264 [MainThread] - [INFO] - Initializing HermesLedControl
2021-04-23 11:18:52,334 [MainThread] - [INFO] - Initializing Rhasspy settings
2021-04-23 11:18:52,335 [MainThread] - [INFO] - Loading configurations
2021-04-23 11:18:52,339 [MainThread] - [INFO] - Loaded 14 hardware references
2021-04-23 11:18:52,340 [MainThread] - [INFO] - - Mqtt server set to mqtt-prod.fritz.box
2021-04-23 11:18:52,341 [MainThread] - [INFO] - - Mqtt port set to 1883
2021-04-23 11:18:52,343 [MainThread] - [INFO] - - Client id set to pi4
2021-04-23 11:18:52,344 [MainThread] - [INFO] - - Hardware set to Pure GPIO
2021-04-23 11:18:52,345 [MainThread] - [INFO] - - Using kiboost as pattern with 3 leds
2021-04-23 11:18:52,346 [MainThread] - [INFO] - Initializing leds controller
2021-04-23 11:18:52,632 [MainThread] - [INFO] - Hermes Led Control started
2021-04-23 11:18:55,058 [Thread-2] - [DEBUG] - On hotword triggered
2021-04-23 11:18:55,704 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2021-04-23 11:18:55,707 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2021-04-23 11:18:55,715 [Thread-2] - [DEBUG] - On listen triggered
2021-04-23 11:18:57,207 [Thread-2] - [DEBUG] - On think triggered
2021-04-23 11:18:58,138 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2021-04-23 11:18:58,142 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2021-04-23 11:18:58,147 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2021-04-23 11:18:58,152 [Thread-2] - [DEBUG] - On success triggered
2021-04-23 11:18:58,213 [Thread-2] - [DEBUG] - On say triggered
2021-04-23 11:19:02,071 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2021-04-23 11:19:02,073 [Thread-2] - [DEBUG] - On tts finished triggered
2021-04-23 11:19:25,777 [Thread-2] - [DEBUG] - On hotword toggle on triggered

kuumaur, 2021-04-23:
I think, that I need a little aid, to start with the customLed Pattern.
Is there a bit more documentation, on how to create custom pattern for GPIO?
Thanx in advance

Psychokiller1888, 2021-04-23:
Hi!

The doc available is the only one. For any type of leds, the syntax is exactly the same, the creator of a led pattern doesn't need to care the hardware others use. To understand how it works, you should check at the existing patterns, like the google/alexa/project alice/kiboost ones.

As of how to connect the leds, you should use the `--pureGpioPinout` argument as a list of the **broadcom** (BCM) gpio pins (https://roboticsbackend.com/raspberry-pi-3-pins/ => the ones in orange)

By reading your pin list, you either took the pin logical numbers, in which case 14 is ground, 15 would be BCM 22 and 18 would be BCM 24 or you are using BCM numbering, in which case BCM 14 and BCM 15 don't exist.

Psychokiller1888, 2021-04-23:
I'd say connect your leds to pins #3, #5, #7 and #6 for the ground. Declare your `--pureGpioPinout` as [3, 5, 7]

kuumaur, 2021-04-23:
Well, I have connected the pins #8 #10 #12 (GPIO 14,15,18) that is by documentation the default.
The pins are connected to switch anode.
Now pattern is working at all.
I tried to investigate the existing pattern scripts and added some ideas to the customled pattern script.
However only the "onStart" is showing some reaction.
All other like "listen", "idle", etc. seem to be not triggert.
I will try pureGpioPinout 8,10,12
DefaultState=Off/On is working, so I thing the wiring is ok.

CustomLedPattern.py
from models.LedPattern import LedPattern
class CustomLedPattern(LedPattern):
        def __init__(self, controller):
                super(CustomLedPattern, self).__init__(controller)

        def wakeup(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def listen(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def think(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def speak(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def idle(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def off(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def updating(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=300, repeat=3)

        def call(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

        def message(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

        def onError(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

        def onSuccess(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

        def onButton1(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

        def onStart(self, *args):
                self._animator.blink(color=[255, 0, 0, 2], minBrightness=2, maxBrightness=20, speed=100, repeat=1)

Psychokiller1888, 2021-04-24:
Are you sure you are getting the mqtt messages correctly?

Have you tried manually sending a message? Or by using the MqttSpy scripts you can quick try them by setting your mqtt broker to localhost

kuumaur, 2021-04-24:
I cant belive, why this setup is so tricky. I have already 2 respeaker-2 up and running with HLC without any hassle. I want to understand, why this particular setup is different. It might be an issue with the mqtt messages. Rhasspy is working and is exchanging messages via mqtt. I can see that HLC on this device is listening to the messages, but it feels not responsible to react on these messages.

2021-04-24 21:38:22,472 [MainThread] - [INFO] - Starting Hermes Led Control v. 2.0.13
2021-04-24 21:38:22,473 [MainThread] - [DEBUG] - Reading command line arguments
2021-04-24 21:38:22,484 [MainThread] - [DEBUG] - Applying configuration file
2021-04-24 21:38:22,485 [MainThread] - [INFO] - Trying to load configuration from "/home/pi/.config/hermesLedControl/configuration.yml"
2021-04-24 21:38:22,499 [MainThread] - [INFO] - Configuration file read successfully
2021-04-24 21:38:22,500 [MainThread] - [DEBUG] - Applying default values
2021-04-24 21:38:22,501 [MainThread] - [DEBUG] - Namespace(activeHigh=True, callPattern=None, clientId=None, conErrorPattern=None, debug='true', defaultBrightness=50, defaultState='off', dndPa
ttern=None, enableDoA=False, endFrame=None, engine='rhasspy', errorPattern=None, everloopPort=20021, gpioPin=None, hardware='puregpio', hermesLedControlConfig='/home/pi/.config/hermesLedContro
l/configuration.yml', idlePattern=None, leds=3, listenPattern=None, matrixIp='127.0.0.1', messagePattern=None, mqttPassword=None, mqttPort='1883', mqttServer='mqtt-prod.fritz.box', mqttUsernam
e=None, offListener='hermes/hotword/toggleOn', offPattern=None, pathToConfig='/home/pi/.config/rhasspy/profiles/de/profile.json', pattern='custom', pid=None, pureGpioPinout=[], setupModePatter
n=None, speakPattern=None, startPattern=None, stopPattern=None, successPattern=None, thinkPattern=None, timeout=15, updatingPattern=None, vid=None, wakeupPattern=None)
2021-04-24 21:38:22,503 [MainThread] - [INFO] - Initializing HermesLedControl
2021-04-24 21:38:22,534 [MainThread] - [INFO] - Initializing Rhasspy settings
2021-04-24 21:38:22,535 [MainThread] - [INFO] - Loading configurations
2021-04-24 21:38:22,536 [MainThread] - [INFO] - Loaded 14 hardware references
2021-04-24 21:38:22,537 [MainThread] - [INFO] - - Mqtt server set to mqtt-prod.fritz.box
2021-04-24 21:38:22,538 [MainThread] - [INFO] - - Mqtt port set to 1883
2021-04-24 21:38:22,538 [MainThread] - [INFO] - - Client id set to pi4
2021-04-24 21:38:22,539 [MainThread] - [INFO] - - Hardware set to Pure GPIO
2021-04-24 21:38:22,539 [MainThread] - [INFO] - - Using custom as pattern with 3 leds
2021-04-24 21:38:22,539 [MainThread] - [INFO] - Initializing leds controller
2021-04-24 21:38:22,675 [MainThread] - [INFO] - Hermes Led Control started
2021-04-24 21:39:01,375 [Thread-2] - [DEBUG] - On success triggered
2021-04-24 21:39:01,476 [Thread-2] - [DEBUG] - On say triggered
2021-04-24 21:39:05,146 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2021-04-24 21:39:05,149 [Thread-2] - [DEBUG] - On tts finished triggered

Psychokiller1888, 2021-04-25:
"Not for me" is decided by checking if the Mqtt message siteId property corresponds to the "deviceName" config or to a string "all"