project-alice-assistant/HermesLedControl #103: Toggle LED On when WAV is played

sadmin91, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
No LED are turn on when I play a WAV file on my Rhasspy instance by calling play-wav endpoint API

**Describe the solution you'd like**
Activate LEDs when a WAV file is played on my Rhasspy instance

Psychokiller1888, 2021-04-27:
Hi

What's the mqtt topic sent when playing a wav file?

sadmin91, 2021-04-27:
Hello,
By reading the Rhasspy documentation (https://rhasspy.readthedocs.io/en/latest/reference/#audio-server), I think that the mqtt topic is `hermes/audioServer/<siteId>/playBytes/<requestId>`

Psychokiller1888, 2021-04-28:
No, playBytes is not connected to HLC, you should use the higher API of Hermes hermes/tts/say