project-alice-assistant/HermesLedControl #106: Custom pattern for systems without a speaker

fake-name, 2021-08-16:
I'm experimenting with Rhasspy, and while this hermes led control is very nice, all the modes in the included pattern are basically impossible to distinguish from each other. This means if your application doesn't have a speaker, you basically just have meaningless flashing lights that tell you nothing.

Here's a pattern file that has clear trivially distinguishable states:

```python

###########################################################################################################
# SUBMIT YOUR OWN CUSTOM PATTERN AND SHARE WITH THE WORLD YOUR LED ANIMATIONS!
# Visit https://github.com/project-alice-assistant/HermesLedControl/issues/new?template=custom-pattern-proposal.md
# for more informations
#
# Check models/LedPattern.py for the available functions
# Do NEVER have a function call a super class function directly!!
# It could cause a deadlock! Instead, call self._controller.THE_METHOD_YOU_WANT
#
# @author: Fake-Name
# @weblink: https://github.com/fake-name
# @email: github@fake-url.com (yes, this is a real email address)
#
###########################################################################################################

import math
import time
import colorsys

from models.LedPattern import LedPattern

def hue_angle_to_rgb(angle, saturation=1, value=1):
	'''
	Given an hue angle, return the RGB triplet
	that represents the 100% saturated rgb value
	for that Hue from the HSV color model.

	Input value is 0-1.
	Return value is normalized to 0-255 integer range

	'''
	# Get fully saturated HSV value
	r, g, b = colorsys.hsv_to_rgb(angle % 1, saturation, value)
	ret = int(r*255), int(g*255), int(b*255)
	return ret

RED_THETA     =   0.0
ORANGE_THETA  =  30.0/360
YELLOW_THETA  =  60.0/360
GREEN_THETA   = 120.0/360
CYAN_THETA    = 180.0/360
BLUE_THETA    = 240.0/360
MAGENTA_THETA = 300.0/360

class RainbowLedPattern(LedPattern):

	def __init__(self, controller):
		super(CustomLedPattern, self).__init__(controller)
		self._image 			= []

	def _newRainbow(self):
		# Drawing the Google color scheme by default
		self._image = []
		for i in range(self._numLeds):
			r, g, b = hue_angle_to_rgb((i-1)/self._numLeds)
			self._image.append([r, g, b, self._controller.defaultBrightness])

	def _newSingleColor(self, hue_angle):
		# Drawing the Google color scheme by default
		self._image = []
		step = math.pi / self._numLeds
		for i in range(self._numLeds):

			r, g, b = hue_angle_to_rgb(hue_angle, value=math.sin(i*step))
			self._image.append([r, g, b, self._controller.defaultBrightness])

	def _rotateImage(self, steps):
		if steps < 0:
			for _ in range(0, steps, -1):
				insertBack = self._image.pop(0)
				self._image.insert(len(self._image), insertBack)
		else:
			for _ in range(steps):
				insertBack = self._image.pop()
				self._image.insert(0, insertBack)

	def _displayImage(self):
		for i, led in enumerate(self._image):
			self._controller.setLedRGB(i, led)
		self._controller.show()

	def wakeup(self):
		self._newRainbow()

		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)

	def listen(self):
		self._newSingleColor(ORANGE_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)

	def think(self):
		self._newSingleColor(BLUE_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(-1)
			self._displayImage()
			time.sleep(0.05)

	def speak(self):
		self._newSingleColor(YELLOW_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)

	def onSuccess(self):
		self._newSingleColor(GREEN_THETA)
		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)
		self.off()

	def onError(self):
		self._newSingleColor(RED_THETA)
		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)
		self.off()

	def idle(self, *args):
		self.off()

	def onStart(self, *args):
		self._controller.wakeup()
		self._controller.idle()

```

Psychokiller1888, 2021-08-19:
Thanks for that!

I don't know what your hardware is, but the included animations don't just flash for half a second :) Google and Alexa are copied over animations like their devices. Project Alice has a smooth rolling theme with with pulsation when thinking. Etc etc

As for the use of the animator, you wouldn't have had the need to create your own "_rotateImage", because the animator has already 3 different rotation system on place :)

If you take another pattern as exemple:

```python
	def think(self, *args):
		self._animator.rotate(color=[0, 0, 255, 25], speed=20, trail=int(self.numLeds / 3))
```

Psychokiller1888, 2021-08-19:
As for the animation flag, the flag is cleared automatically when a new animation is pushed in per exemple, or when HLC goes down, or another animation is called, so you don't need to bother breaking your pattern everywhere across HLC

fake-name, 2021-08-19:
> I don't know what your hardware is, but the included animations don't just flash for half a second :)

At least for my system running Rhasspy, the "thinking" stage generally takes ~0.5-1 second. It's basically impossible to figure out which blink pattern you're seeing when the state only lasts for such a short period of time.

Having the entire color change per-state is a LOT easier to follow.

fake-name, 2021-08-22:
Updates:

```python

###########################################################################################################
# SUBMIT YOUR OWN CUSTOM PATTERN AND SHARE WITH THE WORLD YOUR LED ANIMATIONS!
# Visit https://github.com/project-alice-assistant/HermesLedControl/issues/new?template=custom-pattern-proposal.md
# for more informations
#
# Check models/LedPattern.py for the available functions
# Do NEVER have a function call a super class function directly!!
# It could cause a deadlock! Instead, call self._controller.THE_METHOD_YOU_WANT
#
# @author:
# @weblink:
# @email:
#
###########################################################################################################

import math
import time
import colorsys

from models.LedPattern import LedPattern

def hue_angle_to_rgb(angle, saturation=1, value=1):
	'''
	Given an hue angle, return the RGB triplet
	that represents the 100% saturated rgb value
	for that Hue from the HSV color model.

	Input value is 0-1.
	Return value is normalized to 0-255 integer range

	'''
	# Get fully saturated HSV value
	r, g, b = colorsys.hsv_to_rgb(angle % 1, saturation, value)
	ret = int(r*255), int(g*255), int(b*255)
	return ret

RED_THETA     =   0.0
ORANGE_THETA  =  30.0/360
YELLOW_THETA  =  60.0/360
GREEN_THETA   = 120.0/360
CYAN_THETA    = 180.0/360
BLUE_THETA    = 240.0/360
MAGENTA_THETA = 300.0/360

class CustomLedPattern(LedPattern):

	def __init__(self, controller):
		super(CustomLedPattern, self).__init__(controller)
		self._image 			= []

	def _newRainbow(self):
		# Generate a full rotation of the HSV colorspace across the available
		# LEDs.
		self._image = []
		for i in range(self._numLeds):
			r, g, b = hue_angle_to_rgb((i-1)/self._numLeds)
			self._image.append([r, g, b, self._controller.defaultBrightness])

	def _newSingleColor(self, hue_angle):
		# Generate a nice full-brightness-off-full-brightness cycle of a
		# specific color across the available LEDs
		self._image = []
		step = math.pi / self._numLeds
		for i in range(self._numLeds):

			r, g, b = hue_angle_to_rgb(hue_angle, value=math.sin(i*step))
			self._image.append([r, g, b, self._controller.defaultBrightness])

	def _rotateImage(self, steps):
		if steps < 0:
			for _ in range(0, steps, -1):
				insertBack = self._image.pop(0)
				self._image.insert(len(self._image), insertBack)
		else:
			for _ in range(steps):
				insertBack = self._image.pop()
				self._image.insert(0, insertBack)

	def _displayImage(self):
		for i, led in enumerate(self._image):
			self._controller.setLedRGB(i, led)
		self._controller.show()

	def wakeup(self):
		self._newRainbow()

		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)

	def listen(self):
		self._newSingleColor(ORANGE_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)

	def think(self):
		self._newSingleColor(BLUE_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(-1)
			self._displayImage()
			time.sleep(0.05)

	def speak(self):
		self._newSingleColor(YELLOW_THETA)
		self._animation.set()
		while self._animation.isSet():
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)
		self.off()

	def onSuccess(self):
		self._newSingleColor(GREEN_THETA)
		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)
		self.off()

	def onError(self):
		self._newSingleColor(RED_THETA)
		for _ in range(self._numLeds):
			self._rotateImage(1)
			self._displayImage()
			time.sleep(0.05)
		self.off()

	def onStart(self, *args):
		self._controller.wakeup()
		self.off()

```

Now it doesn't leave the LEDs on after speaking.

Psychokiller1888, 2021-08-22:
Yeah, no problem, was just answering the different points :)
Re animations, they came in at a later point, when the basics were already made, and I did not take the time to update google, alexa, alice to using them. One pattern is fully using them, the PGAS: https://github.com/project-alice-assistant/HermesLedControl/blob/master/ledPatterns/PgasPattern.py

You don't see a "pushing a new animation in" because it's made at a higher level. HLC uses Queue as a FIFO system on the main loop to push animations in the queue when recieving infos over mqtt.

Thanks for your pattern by the way, I will publish it asap

Psychokiller1888, 2021-08-22:
Ok, done!
https://github.com/project-alice-assistant/HermesLedControl/commit/c7f4c003db7a9d8ee036cb9d0946268218cc805c

As for changes made:
- Compiled headers with your credits
- Moved static vars to class
- Moved your awesome hueToRgb to led pattern, it's very useful

No other changes :)

closed 2023-04-08 by Psychokiller1888