project-alice-assistant/HermesLedControl #114: Install script not working with python3.9

zwometer, 2021-11-17:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
I could not start it because the installation aborted with the remark, that I need at least python3.5
I have python3.9 installed.

**Led device name**
ReSpeaker 2

**The bug**
The installation aborted with the remark, that I need at least python3.5
I have python3.9 installed.

**Debug output**
Follow instructions on https://github.com/project-alice-assistant/HermesLedControl/wiki/Run-in-debug-mode
And paste the output
I did not get it to run at all.

**To Reproduce**
Install python3.9
Try to install HLC as described on github.

Psychokiller1888, 2021-11-18:
Hi!

You can try to run on 3.0.0, which is not yet released: https://github.com/project-alice-assistant/HermesLedControl/tree/3.0.0

zwometer, 2021-11-18:
Thank you for your reply. I tried but could not get it to work. Would you mind sharing some details on how one would go about installing that version?

My workaround for now is to edit the install.sh so that it not only checks for 3.5, 3.6 and 3.7 but also for 3.9 and then I execute the install.sh manually. (installation is still running, but it looks good so far.)

Psychokiller1888, 2021-11-18:
you can `cd`to HLC directory and then `git checkout 3.0.0` and `git pull`

zwometer, 2021-11-18:
I tried both ways now. my initial try with 2.0.15 and modified install.sh led to a service that was crashing after a few seconds after being started:
`hermesledcontrol.service - Hermes Led Control
     Loaded: loaded (/etc/systemd/system/hermesledcontrol.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2021-11-18 15:45:11 CET; 2s ago
    Process: 2554 ExecStart=/home/pi/hermesLedControl_v2.0.15/venv/bin/python3 main.py --hermesLedControlConfig=/home/pi/.config/hermesLedControl/configuration.yml (code=exited, status=1/F>
   Main PID: 2554 (code=exited, status=1/FAILURE)
        CPU: 3.227s

Nov 18 15:45:11 AudioPi1 systemd[1]: hermesledcontrol.service: Failed with result 'exit-code'.
Nov 18 15:45:11 AudioPi1 systemd[1]: hermesledcontrol.service: Consumed 3.227s CPU time.`

Then I installed version 3.0.0 but this didn't even create a service that I could start. So I'm not sure how I would run it.

timothybuchanan, 2022-06-02:
I am also trying to install HLC using Python3.9. I get a gcc failure with this output:

Building wheels for collected packages: matrix-lite
  Building wheel for matrix-lite (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      /home/voice/HermesLedControl/venv/lib/python3.9/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
        warnings.warn(
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-39
      creating build/lib.linux-armv7l-cpython-39/matrix_lite
      copying matrix_lite/colour.py -> build/lib.linux-armv7l-cpython-39/matrix_lite
      copying matrix_lite/__init__.py -> build/lib.linux-armv7l-cpython-39/matrix_lite
      copying matrix_lite/sensors.py -> build/lib.linux-armv7l-cpython-39/matrix_lite
      copying matrix_lite/led.py -> build/lib.linux-armv7l-cpython-39/matrix_lite
      running build_ext
      building '_matrix_hal' extension
      creating build/temp.linux-armv7l-cpython-39
      creating build/temp.linux-armv7l-cpython-39/hal_wrapper
      creating build/temp.linux-armv7l-cpython-39/hal_wrapper/drivers
      gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/tmp/pip-install-goxyu68q/matrix-lite_0791c9804bc145f8a08085674fa9e761/.eggs/pybind11-2.9.2-py3.9.egg/pybind11/include -I/tmp/pip-install-goxyu68q/matrix-lite_0791c9804bc145f8a08085674fa9e761/.eggs/pybind11-2.9.2-py3.9.egg/pybind11/include -I/home/voice/HermesLedControl/venv/include -I/usr/include/python3.9 -c hal_wrapper/drivers/everloop.cpp -o build/temp.linux-armv7l-cpython-39/hal_wrapper/drivers/everloop.o -O3 -DVERSION_INFO=\"0.0.8\" -std=c++11
      hal_wrapper/drivers/everloop.cpp:2:10: fatal error: ./everloop.h: No such file or directory
       #include "./everloop.h"
                ^~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for matrix-lite

Where is everloop.h?