project-alice-assistant/HermesLedControl #115: Systemctl 'No Modul named RPi' | Debug works

ChrizZz90, 2021-12-04:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
python main.py --hermesLedControlConfig=/home/pi/.config/hermesLedControl/configuration.yml

```
engine: "rhasspy"
mqttServer: "IP"
mqttPort: "PORT"
mqttUsername: "USER"
mqttPassword: "PASSWORD"
clientId: "pi0sat"
pathToConfig: "/.config/rhasspy/profiles/de/profile.json"
hardware: "respeaker2Mics"
pattern: "projectalice"
enableDoA: false
offListener: "hermes/audioServer/playFinished"
```

**Led device name**
ReSpeaker 2 on a Raspberry Pi Zero 2. I use it with Rhasspy and Rhasspy is running in a docker container.

**The bug**
I have the issue that with based on systemctl Hermes LED Control doesn't work. But if I run HLC in debug mode, everything works fine. 

```
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ sudo systemctl status hermesledcontrol.service
● hermesledcontrol.service - Hermes Led Control
     Loaded: loaded (/etc/systemd/system/hermesledcontrol.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-12-04 16:13:37 CET; 976ms ago
   Main PID: 19449 (python3)
      Tasks: 3 (limit: 409)
        CPU: 964ms
     CGroup: /system.slice/hermesledcontrol.service
             └─19449 /home/pi/hermesLedControl_v2.0.15/venv/bin/python3 main.py --hermesLedControlConfig=/home/pi/.config/hermesLedControl/configuration.yml

Dez 04 16:13:38 raspberrypi python3[19449]: Traceback (most recent call last):
Dez 04 16:13:38 raspberrypi python3[19449]:   File "/home/pi/hermesLedControl_v2.0.15/main.py", line 63, in <module>
Dez 04 16:13:38 raspberrypi python3[19449]:     main()
Dez 04 16:13:38 raspberrypi python3[19449]:   File "/home/pi/hermesLedControl_v2.0.15/main.py", line 48, in main
Dez 04 16:13:38 raspberrypi python3[19449]:     slc = HermesLedControl(args)
Dez 04 16:13:38 raspberrypi python3[19449]:   File "/home/pi/hermesLedControl_v2.0.15/models/HermesLedControl.py", line 134, in __init__
Dez 04 16:13:38 raspberrypi python3[19449]:     self._ledsController = LedsController(self)
Dez 04 16:13:38 raspberrypi python3[19449]:   File "/home/pi/hermesLedControl_v2.0.15/models/LedsController.py", line 74, in __init__
Dez 04 16:13:38 raspberrypi python3[19449]:     import RPi.GPIO
Dez 04 16:13:38 raspberrypi python3[19449]: ModuleNotFoundError: No module named 'RPi'
```

```
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ pip freeze | grep RPi
RPi.GPIO==0.7.0
```

**Debug output**
Debug mode works as it should.

**To Reproduce**
Well, good question. As in debug everything works, I don't know why I have this kind of issue.

Psychokiller1888, 2021-12-04:
You need to make sure rpi.gpio in install in HLC venv

```
cd ~/hermesLedControl_v2.0.15
./venv/bin/pip install rpi.gpio
```

ChrizZz90, 2021-12-04:
that is causing issues. Do you have any idea if this is HLC related?

```
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ ./venv/bin/pip3 install rpi.gpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi.gpio
  Using cached RPi.GPIO-0.7.0.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: rpi.gpio
  Building wheel for rpi.gpio (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/hermesLedControl_v2.0.15/venv/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-2u5j3yle/rpi-gpio_beb7ec2026f8436a80d1638030145c1a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-2u5j3yle/rpi-gpio_beb7ec2026f8436a80d1638030145c1a/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-e_3ycsyv
       cwd: /tmp/pip-install-2u5j3yle/rpi-gpio_beb7ec2026f8436a80d1638030145c1a/
  Complete output (87 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.9
  creating build/lib.linux-armv7l-3.9/RPi
  copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
  creating build/lib.linux-armv7l-3.9/RPi/GPIO
  copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
  running build_ext
  building 'RPi._GPIO' extension
  creating build/temp.linux-armv7l-3.9
  creating build/temp.linux-armv7l-3.9/source
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-armv7l-3.9/source/event_gpio.o
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/pi/hermesLedControl_v2.0.15/venv/include -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-armv7l-3.9/source/py_gpio.o
  source/py_gpio.c: In function ‘PyInit__GPIO’:
  source/py_gpio.c:1046:4: warning: ‘PyEval_ThreadsInitialized’ is deprecated [-Wdeprecated-declarations]
   1046 |    if (!PyEval_ThreadsInitialized())
        |    ^~
  In file included from /usr/include/python3.9/Python.h:145,
                   from source/py_gpio.c:23:
  /usr/include/python3.9/ceval.h:129:36: note: declared here
    129 | Py_DEPRECATED(3.9) PyAPI_FUNC(int) PyEval_ThreadsInitialized(void);
        |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
  source/py_gpio.c:1047:7: warning: ‘PyEval_InitThreads’ is deprecated [-Wdeprecated-declarations]
   1047 |       PyEval_InitThreads();
        |       ^~~~~~~~~~~~~~~~~~
  In file included from /usr/include/python3.9/Python.h:145,
                   from source/py_gpio.c:23:
  /usr/include/python3.9/ceval.h:130:37: note: declared here
[... cut ...] ;f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-_crz043a/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/hermesLedControl_v2.0.15/venv/include/site/python3.9/rpi.gpio Check the logs for full command output.

```

Psychokiller1888, 2021-12-04:
No it's not, it's rpi.gpio install related, that is saying that some functions are deprecated in python 3.9

Start by uninstalling rpi.gpio from your global site package and make sure you only have it in venv. Also make sure you never `sudo pip`

ChrizZz90, 2021-12-04:
```
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ pip uninstall RPi.GPIO
WARNING: Skipping RPi.GPIO as it is not installed.
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ pip freeze | grep RPi
pi@raspberrypi:~/hermesLedControl_v2.0.15 $ cd ./venv/lib/python3.9/site-packages/
colorzero/                   gpiozero/                    numpy/                       paho_mqtt-1.6.1.dist-info/   pkg_resources/               setuptools/                  toml/                        wheel-0.37.0.dist-info/
colorzero-2.0.dist-info/     gpiozero-1.6.2.dist-info/    numpy-1.21.4.dist-info/      pip/                         __pycache__/                 setuptools-58.3.0.dist-info/ toml-0.10.2.dist-info/       _yaml/
_distutils_hack/             gpiozerocli/                 paho/                        pip-21.3.1.dist-info/        PyYAML-6.0.dist-info/        spidev-3.5.dist-info/        wheel/                       yaml/

```

I guess rpi.gpio is uninstalled now. But should there be a rpi.gpio folder in venv side-packages? I still have the same issue if I try to install it

ChrizZz90, 2021-12-07:
it's an issue related to rpi.gpio and Raspbian Bullseye. Installing an alpha version solved the issue.

`pip3 install RPi.GPIO==0.7.1a4`

closed 2021-12-07 by ChrizZz90