project-alice-assistant/HermesLedControl #116: Play Animation on customized event, e.g. play a music, apart from standard methods in patttern

daiqing2009, 2021-12-23:
**Is your feature request related to a problem? Please describe.**
I just wonder specific let animation(s) can be shown  when certain event, apart from standard , is triggered,e.g. playing music locally. 

**Describe the solution you'd like**
When playing (certain piece of) music, e.g. Christmas songs, it will be fascinating if specific led animation(s) can be shown on the respeaker leds

**Describe alternatives you've considered**
I got some hint from #103, which suggest hook on hermes/audioServer/<siteId>/playBytes/<requestId>. Though, i agree that it is NOT a standard feature, just wonder if some kind of plugin architecture can be well-defined so that we can implement a specific plugin for that. 

**Additional context**
Add any other context or screenshots about the feature request here.

Psychokiller1888, 2021-12-23:
Hi there!

Nice idea, love it!

As of now, you could use [hermes/leds/manual/animations](https://github.com/project-alice-assistant/HermesLedControl/blob/master/models/HermesLedControl.py#L454) to call animations.

This calls for premade animations, but they are easily extensible? Otherwise, we could also think about a "partition" architecture, a way to send a pattern over mqtt that HLC would play?

daiqing2009, 2021-12-24:
Thanks, @Psychokiller1888 , let me try this idea today or tomorrow. Will feedback the result.

daiqing2009, 2021-12-29:
I try to make most of the animation available, but i found one problem, once the parameter is not correct(logs below), leds fails to respond to further instructions. 
`2021-12-29 08:53:26,348 [Thread-24] - [DEBUG] - New animation <bound method AlexaLedPattern.idle of <ledPatterns.AlexaLedPattern.AlexaLedPattern object at 0x75ebf9b0>> has id 7f106b4c-7bc0-4f32-9b44-7195f5efef67
2021-12-29 08:53:47,812 [Thread-2] - [ERROR] - Caught exception in on_message: 'colors'`