project-alice-assistant/HermesLedControl #117: Error during 3.0.0 installation and Python 3.9

Aurel-14, 2021-12-30:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
./hlc_download.sh 3.0.0
[HermesLedControl-3.0.0-Installation Isue.txt](https://github.com/project-alice-assistant/HermesLedControl/files/7792026/HermesLedControl-3.0.0-Installation.Isue.txt)

**Led device name**
ReSpeaker 3

**The bug**
Error during installation

**Debug output**
see attached file

**To Reproduce**
Install the last version of raspberrypi OS Lite
install Respeaker 4mics drivers
install 3.0.0 HermesLedControl

Psychokiller1888, 2021-12-30:
Hey there! 3.0.0 is not yet out and I did not test Bullseye that brings many many issues. It's noted, thanks, but we'll see down the road when releasing. I did not have these issues installing on Buster. Also, the install script changed for 3.0.0, you can find it on the repo for the latest version, just in case

Your bug files states an error with GPIO. Which means, wheels aren't yet available on Bullseye.

Aurel-14, 2021-12-30:
Hy,

Ok thanks for your reply !
I see that for the install script and i used the 3.0.0 version
so you advise me to reinstall Buster and stable version of HLC ?

Psychokiller1888, 2021-12-30:
3.0.0 works flawless on buster, that's for sure

the hlc_download.sh script is just a little different:

```
#!/usr/bin/env bash

if [[ "$EUID" -ne 0 ]]
  then echo "Please run as root"
  exit
fi

apt-get install git

if [[ -z "$1" ]]; then
    echo "No version supplied"
    latest=$(curl --silent "https://api.github.com/repos/project-alice-assistant/HermesLedControl/releases/latest" | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/')
    echo "Will download $latest"
    VERSION=$latest
else
    VERSION=$1
    echo "Supplied version $VERSION"
fi

dest="/home/$(logname)/HermesLedControl"

rm -rf "$dest"
git clone https://github.com/project-alice-assistant/HermesLedControl.git "$dest"
cd "$dest" || exit
git fetch
git checkout "$VERSION"
git pull
chown -R "$(logname)" "$dest"

chmod +x install.sh
./install.sh
```

Using git directly and not manually downloading releases enables easy updating then. To start install you can just `sudo ./hlc_download.sh 3.0.0` as 3.0.0 is not released and is not the "latest" on github

Aurel-14, 2021-12-30:
Thanks ! I will try with Buster

Aurel-14, 2021-12-30:
All is installed correctly !
But i have  new issue with the connection to mqtt server, in service log. there is: 
`[CRITICAL] - Couldn't connect to mqtt, aborting`
I my Rhasspy i have configured external mqtt server with the ip and port of my brocker.
I don't know with i ahve this error ...

Psychokiller1888, 2021-12-30:
The only thing I know is that rhasspy doesn't use the default mqtt port or something like that. Try to start HLC specifying the broker and port manually

Aurel-14, 2021-12-30:
What is the command to  specifying the broker and port ?
or config file tu put them ?

Aurel-14, 2021-12-30:
Ok I have add the port in my file profile.json of rhasspy

Aurel-14, 2021-12-30:
Resolved by réinstalling Buster version of Raspberry pi os
Don't not work on last "Bullseye" version

closed 2021-12-30 by Aurel-14