project-alice-assistant/HermesLedControl #125: led's only lite up when simutainuesly running (python ~/pixel_ring/examples/respeaker_v2_pixel_ring.py)

rooferdave, 2023-02-10:
**Do not remove fields, answer them, otherwise your bug will be reported as invalid and closed**

**How I did start SLC, with what arguments**
./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2  --mqttServer=10.0.2.108  --mqttPort=1883 --mqttUsername=homefix --mqttPassword=Flash1 --clientId=corev2 --leds=12 --pattern=google --defaultState=on

**Led device name**

ReSpeaker Core V2 boot from sdcard...also tried with booting from mmc(using updated corev2 linux respeaker-debian-9-iot-flasher-20180801-4gb.img.xz and also used:respeaker-debian-9-iot-sd-20180801-4gb.img

**The bug**

Obviously not a bug I cannot find anyone else with this problem, Sure somthing i, not doing right.
The patterns work if i use "screen -S pixelring" and execute "python ~/pixel_ring/examples/respeaker_v2_pixel_ring.py" then "cntrl a-d" to exit screen then "./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2  --mqttServer=10.0.2.108  --mqttPort=1883 --mqttUsername=homefix --mqttPassword=Flash1 --clientId=corev2 --leds=12 --pattern=google --defaultState=on". The Lights Dance and  Shine (with the interferance of the python script running at the same time messing up the google pattern)

If I skip the "python ~/pixel_ring/examples/respeaker_v2_pixel_ring.py" step.....No Lights

**Debug output**

This is with respeaker_v2_pixel_ring.py running in the background:................

~/HermesLedControl$ ./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2  --mqttServer=10.0.2.108  --mqttPort=1883 --mqttUsername=homefix --mqttPassword=Flash1 --clientId=corev2 --leds=12 --pattern=google --defaultState=on
2023-02-10 13:56:48,508 [MainThread] - [INFO] - Starting Hermes Led Control version 3.0.4
2023-02-10 13:56:48,510 [MainThread] - [INFO] - Reading command line arguments
2023-02-10 13:56:48,534 [MainThread] - [DEBUG] - Applying configuration file
2023-02-10 13:56:48,537 [MainThread] - [INFO] - Trying to load configuration from '/home/respeaker/.config/HermesLedControl/configuration.yml'
2023-02-10 13:56:48,549 [MainThread] - [INFO] - Configuration file read successfully
2023-02-10 13:56:48,550 [MainThread] - [DEBUG] - 'engine' was provided as command line argument. Ignoring config file.
2023-02-10 13:56:48,552 [MainThread] - [DEBUG] - 'hardware' was provided as command line argument. Ignoring config file.
2023-02-10 13:56:48,554 [MainThread] - [DEBUG] - 'pattern' was provided as command line argument. Ignoring config file.
2023-02-10 13:56:48,555 [MainThread] - [DEBUG] - Applying default values
2023-02-10 13:56:48,557 [MainThread] - [DEBUG] - Namespace(activeHigh=True, callPattern=None, clientId='corev2', conErrorPattern=None, debug=True, defaultBrightness=50, defaultState='on', dndPattern=None, enableDoA=False, endFrame=None, engine='rhasspy', errorPattern=None, everloopPort=20021, gpioPin=None, hardware='respeakerCoreV2', hermesLedControlConfig='~/.config/HermesLedControl/configuration.yml', idlePattern=None, leds=12, listenPattern=None, matrixIp='127.0.0.1', messagePattern=None, mqttPassword='Flash1', mqttPort='1883', mqttServer='10.0.2.108', mqttUsername='homefix', offListener='hermes/hotword/toggleOn', offPattern=None, pathToConfig='', pattern='google', pid=None, pureGpioPinout=[], setupModePattern=None, speakPattern=None, startPattern=None, stopPattern=None, successPattern=None, thinkPattern=None, timeout=120, updatingPattern=None, vid=None, wakeupPattern=None)
2023-02-10 13:56:48,560 [MainThread] - [INFO] - Initializing HermesLedControl
2023-02-10 13:56:48,568 [MainThread] - [INFO] - Initializing Rhasspy settings
2023-02-10 13:56:48,570 [MainThread] - [INFO] - Loading configurations
2023-02-10 13:56:48,573 [MainThread] - [INFO] - Loaded 14 hardware references
2023-02-10 13:56:48,575 [MainThread] - [INFO] - - Mqtt server set to 10.0.2.108
2023-02-10 13:56:48,577 [MainThread] - [INFO] - - Mqtt port set to 1883
2023-02-10 13:56:48,578 [MainThread] - [INFO] - - Mqtt username set to homefix
2023-02-10 13:56:48,580 [MainThread] - [INFO] - - Mqtt password set!
2023-02-10 13:56:48,581 [MainThread] - [INFO] - - Client id set to corev2
2023-02-10 13:56:48,583 [MainThread] - [INFO] - - Hardware set to Respeaker Core v2
2023-02-10 13:56:48,584 [MainThread] - [INFO] - - Using google as pattern with 12 leds
2023-02-10 13:56:48,586 [MainThread] - [INFO] - Initializing leds controller
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpigpio: This module can only be run on a Raspberry Pi!
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from native: unknown old-style revision "0"
  'Falling back from %s: %s' % (name, str(e))))
2023-02-10 13:56:48,855 [MainThread] - [INFO] - Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined
2023-02-10 13:56:48,890 [MainThread] - [INFO] - Hermes Led Control started
2023-02-10 13:56:49,391 [Thread-3] - [DEBUG] - New animation "wakeup" has id 52077286-12dd-4ea7-8a98-c22e3b9ca5e9
2023-02-10 13:56:49,395 [Thread-3] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-02-10 13:56:49,404 [Thread-3] - [DEBUG] - New animation "idle" has id 07cfdedd-21bf-457c-8b1b-4dd68fff2b04
2023-02-10 13:57:32,079 [Thread-2] - [DEBUG] - On hotword triggered
2023-02-10 13:57:32,084 [Thread-2] - [DEBUG] - New animation "wakeup" has id 90867f42-6b90-465e-996a-22ff30a8c570
2023-02-10 13:57:32,094 [Thread-2] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-02-10 13:57:32,233 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2023-02-10 13:57:32,238 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2023-02-10 13:57:32,241 [Thread-2] - [DEBUG] - New animation "idle" has id 7e2f1b15-4ba3-4004-8b2c-851789d6ccf2
2023-02-10 13:57:32,282 [Thread-2] - [DEBUG] - On listen triggered
2023-02-10 13:57:32,286 [Thread-2] - [DEBUG] - New animation "listen" has id 56316273-af2e-4123-9490-f0356656b3c8
2023-02-10 13:57:32,290 [Thread-2] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-02-10 13:57:34,550 [Thread-2] - [DEBUG] - On think triggered
2023-02-10 13:57:34,555 [Thread-2] - [DEBUG] - New animation "think" has id f5fd5e77-1c85-460d-b5bb-ce79b16195b2
2023-02-10 13:57:34,559 [Thread-2] - [DEBUG] - Timeout is setting duration from 0 to 120
[... 60 lines cut ...]
2023-02-10 13:59:05,550 [Thread-2] - [DEBUG] - On think triggered
2023-02-10 13:59:05,553 [Thread-2] - [DEBUG] - New animation "think" has id c0f2dbad-518b-48bd-a829-938c59af0103
2023-02-10 13:59:05,556 [Thread-2] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-02-10 13:59:05,676 [Thread-2] - [DEBUG] - On play finished received but it wasn't for me
2023-02-10 13:59:05,680 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2023-02-10 13:59:05,684 [Thread-2] - [DEBUG] - New animation "idle" has id 67cf44e8-cadd-46b7-b493-60d865abefb5
2023-02-10 13:59:05,692 [Thread-2] - [DEBUG] - On hotword toggle on triggered
2023-02-10 13:59:05,699 [Thread-2] - [DEBUG] - New animation "idle" has id ce345abb-e954-4646-88dc-8db848e38e46
2023-02-10 13:59:05,702 [Thread-2] - [DEBUG] - On success triggered
2023-02-10 13:59:05,705 [Thread-2] - [DEBUG] - New animation "onSuccess" has id 8d836636-ed99-4858-b97b-a36d68e15db1
2023-02-10 13:59:05,708 [Thread-2] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-02-10 13:59:05,767 [Thread-2] - [DEBUG] - On say received but it wasn't for me
2023-02-10 13:59:07,222 [Thread-2] - [DEBUG] - On tts finished received but it wasn't for me

NOTE: This is not my first attempt at this. I just cant get the light to work unless i Jump-Start with the pixelring python script.

**To Reproduce**
Followed the wiki answered all questions correctly

Engine: rhasspy
Path: Kept default also tried full path /home/respeaker/.config/rhasspy/profiles/en/profile.json
path to assistant: default also tried full path /home/respeaker/.config/HermesLedControl
Device: respeakerCoreV2
Pattern: google
DOA tried all options on different install retries

DivanX10, 2023-03-27:
I join, a similar problem with Raspeaker Core V2. [Pixel Ring](https://github.com/respeaker/pixel_ring) it works fine for me, but HermesLedControl can't control the LEDs

OS Debian 10
Python 3.7.3

```
./venv/bin/python3 main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2 --mqttServer=localhost --mqttPort=1883 --clientId=respeaker --leds=12 --pattern=google --gpioPin=12 --pathToConfig=/home/respeaker/.config/rhasspy/profiles/ru/profile.json
```
[image]

```

2023-03-28 02:33:22,594 [MainThread] - [INFO] - Initializing HermesLedControl
2023-03-28 02:33:22,601 [MainThread] - [INFO] - Initializing Rhasspy settings
2023-03-28 02:33:22,603 [MainThread] - [INFO] - Loading configurations
2023-03-28 02:33:22,606 [MainThread] - [INFO] - Loaded 14 hardware references
2023-03-28 02:33:22,608 [MainThread] - [INFO] - - Mqtt server set to localhost
2023-03-28 02:33:22,609 [MainThread] - [INFO] - - Mqtt port set to 1883
2023-03-28 02:33:22,611 [MainThread] - [INFO] - - Client id set to respeaker
2023-03-28 02:33:22,612 [MainThread] - [INFO] - - Hardware set to Respeaker Core v2
2023-03-28 02:33:22,613 [MainThread] - [INFO] - - Using google as pattern with 12 leds
2023-03-28 02:33:22,615 [MainThread] - [INFO] - Initializing leds controller
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpigpio: This module can only be run on a Raspberry Pi!
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/respeaker/HermesLedControl/venv/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from native: unknown old-style revision "0"
  'Falling back from %s: %s' % (name, str(e))))
2023-03-28 02:33:22,839 [MainThread] - [INFO] - Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined
2023-03-28 02:33:22,866 [MainThread] - [INFO] - Hermes Led Control started
2023-03-28 02:33:23,367 [Thread-3] - [DEBUG] - New animation "wakeup" has id c50bf5c7-ca64-4c82-984b-7fefb92830e0
2023-03-28 02:33:23,370 [Thread-3] - [DEBUG] - Timeout is setting duration from 0 to 120
2023-03-28 02:33:23,377 [Thread-3] - [DEBUG] - New animation "idle" has id 32e8dfaa-b5a2-4401-a24b-a2c5c45bda2b
^C2023-03-28 02:33:30,086 [MainThread] - [INFO] - Shutting down Hermes Led Control

```

DivanX10, 2023-04-05:
I managed to start the LEDs through hermesledcontrol, but for some reason it works through debugging mode, and if you run it through the service, it doesn't work and there is clearly something with rights and with launching from a virtual environment. How to make LEDs work? Interestingly, if you run the python file to turn on the LEDs via the console and simultaneously access the column, the hermesledcontrol color effects will work. It's as if hermesledcontrol lacks permissions or access to `spidev0.0` and `spidev0.1'. What are your thoughts on this? What are the possible solutions to the issue?

How do I start the LEDs?

There are 12 LEDs on the Raspeaker Core v2 board and they can be turned on. These LEDs are controlled via the files `spidev0.0` and `spidev0.1` which are located in `/dev'.

In order for the LEDs to turn on, you need to run them as `root` or via `sudo`, since there are two files `spidev0.0` and `spidev0.1` have a `user`:`group` `root`:`spi`

[image]
[image]

In order not to run the LEDs through root, but through the user `respeaker`, you can do the following

View User groups
```
groups respeaker
```

Add the user `respeaker` to the group `spi` and `sudo`
```
sudo useradd respeaker -G spi,sudo
```

Allow the 'respeaker` user to run python scripts with sudo without entering a password

Let's create our `sudoers.d` with the name `respeaker`
```
sudo visudo -f /etc/sudoers.d/respeaker 
```

Let's add the user `respeaker` and indicate that only python scripts can be run without a password
```
respeaker  ALL=NOPASSWD: /usr/bin/python, /usr/bin/python2, /usr/bin/python3
```

[image]

[Download python scripts for turning on LEDs](https://github.com/DivanX10/ReSpeaker-Core-v2-Rus/tree/main/leds) 

Now let's check the launch of the python script c sudo without input to turn on the LEDs

If we have Debian 11 installed and the python script needs to be run with python version 2, then we need to install python version 2 and we will run with python version 2

```
sudo apt install python
```

You can run the python version 2 script. The password will not be requested when entering `sudo`
```
sudo python /home/respeaker/led_wakeup.py
```
or

```
sudo python2 /home/respeaker/led_wakeup.py
```

Next, stop hermesledcontrol

```
sudo systemctl stop hermesledcontrol
```

Edit the file HermesLedControl.py
```
nano /home/respeaker/HermesLedControl/models/HermesLedControl.py
```
Add  `import subprocess` and `subprocess.call("sudo python2 /home/respeaker/leds/led_wakeup.py", shell=True)`

[image]
[image]

Let's go to the HermesLedControl folder
```
cd /home/respeaker/HermesLedControl
```

Let's run it in debug mode
```
./venv/bin/python main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2 --mqttServer=localhost --mqttPort=1883 --clientId=respeakerv2 --leds=12 --pattern=google --pathToConfig=/home/respeaker/.config/rhasspy/profiles/ru/profile.json
```

And the LEDs light up. I can't understand why the LEDs don't light up without these manipulations and how can they be started normally?

For reference, I have compiled new images from debian 10 and 11 for raspeaker core v2, you can download [here](https://github.com/DivanX10/ReSpeaker-Core-v2-Rus/wiki/Образы)

DivanX10, 2023-04-05:
### The solution to the problem has been found

I found the cause of the problem and understood why HermesLedControl could not control the LEDs. After I fixed it, HermesLedControl worked perfectly and the LEDs glow as they should.  HermesLedControl cannot turn on the LEDs itself.

To turn on the LEDs, I used this template [respeaker_v2_pixel_ring.py](https://github.com/respeaker/pixel_ring/blob/master/examples/respeaker_v2_pixel_ring.py ) and the LEDs lit up and this pattern was the culprit that HermesLedControl could not control the LEDs. I discovered this purely by chance when I was creating several files with different effects to turn the LEDs on and off, and here I discovered that if I turn on the LEDs, HermesLedControl will be able to control the LEDs. I double-checked and my accidental detection confirmed it.

### What needs to be done so that HermesLedControl can control ReSpeaker Core v2 LEDs?

Download the archive file leds_ring.zip
```
wget https://github.com/DivanX10/ReSpeaker-Core-v2-Rus/raw/main/files/leds_ring.zip
```

Unpack the archive file leds_ring.zip
```
unzip leds_ring.zip
```

Launch `turn_on_leds.py ` you can use the command. Specify the path to the file
```
sudo /home/respeaker/leds_ring/python2 turn_on_leds.py
```

The LEDs should turn on for 3 seconds and go out.

In the HermesLedControl settings, we do not touch anything, we leave it as it is, even if you specify a different host and port in the rhasspy settings, all settings are pulled from rhasspy automatically thanks to this line of code

```
pathToConfig: "/home/respeaker/.config/rhasspy/profiles/ru/profile.json"

```

This is how the `configuration' settings should look like.yml` and they do not need to be changed, we do not prescribe anything there, except for pattern. You can specify:

- pattern: "google"
- pattern: "alexa"
- pattern: "projectalice"
- pattern: "kiboost"

```
engine: "rhasspy"
pathToConfig: "/home/respeaker/.config/rhasspy/profiles/ru/profile.json"
hardware: "respeakerCoreV2"
pattern: "google"
enableDoA: false
```

You can open configuration.yml like this

```
nano /home/respeaker/.config/HermesLedControl/configuration.yml
```

If you changed the `pattern`, then you definitely need to restart HermesLedControl

```
sudo systemctl restart hermesledcontrol
```

Psychokiller1888, 2023-04-07:
What I don't get is that you use piyel_ring, which is apa102, what respeakerCoreV2 uses so basically it's the same

Psychokiller1888, 2023-04-07:
I think Respeaker Core v2 mraa uses gpio 12 to power instead of gpio 5. Can you check in `interfaces\apa102.py`, line 26 and change `mraa.Gpio(5)`to `mraa.Gpio(12)` and restart HLC

Psychokiller1888, 2023-04-07:
Another thing, in your logs, it shows

`2023-03-28 02:33:22,839 [MainThread] - [INFO] - Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined`

where it should be using it... Try installing it in your HLC venv

DivanX10, 2023-04-07:
> I think Respeaker Core v2 mraa uses gpio 12 to power instead of gpio 5. Can you check in `interfaces\apa102.py`, line 26 and change `mraa.Gpio(5)`to `mraa.Gpio(12)` and restart HLC

Thank you for responding. I tried to do as you said, but it didn't work out.

[image]

I have python-mraa installed и я использую mraa according to this [instruction](https://wiki.seeedstudio.com/ReSpeaker_Core_v2.0/#play-with-gpio). If you enter the command `mraa-gpio list', the pins are displayed

[image]

***
`where it should be using it... Try installing it with apt-get install python-mraa but it should be done by the respeaker installer`

[image]

as I understood, the LEDs need to be turned on every time after a debian reboot. For some reason, if you turn on the LEDs through the pixel ring script, then HermesLedControl can control the LEDs

```
import time

from pixel_ring import pixel_ring
import mraa
import os

en = mraa.Gpio(12)
if os.geteuid() != 0 :
    time.sleep(0)
 
en.dir(mraa.DIR_OUT)
en.write(0)

pixel_ring.set_brightness(50)

if __name__ == '__main__':
	pixel_ring.on()
	time.sleep(3)
en.write(1)
```

Psychokiller1888, 2023-04-07:
What I can see from your logs is that: `2023-03-28 02:33:22,839 [MainThread] - [INFO] - Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined` which means power is ignored. So the interface doesn't write the leds as on in `interfaces\Interface.py`on line 58 and/or 61

Psychokiller1888, 2023-04-07:
After thinking a little, it might also be because HLC is running in a venv, the venv is created before python-mraa is installed, so it's not available for it...

DivanX10, 2023-04-07:
> Or install anything python related using sudo?

Yes, I installed python-mraa via sudo as specified in [documentation]([image])
```
sudo apt install -y python-mraa python-upm libmraa1 libupm1 mraa-tools
```

[image]

I will try to install python-mraa without sudo, and in interfaces\apa102.py, line 26 and change mraa.Gpio(5)to mraa.Gpio(12) and restart HLC and I will unsubscribe as I check

DivanX10, 2023-04-07:
It didn't help. I installed debian from 0, installed Rhasspy, HermesLedControl, python-mraa did not install from root, but did it through sudo, as a result I can run python examples/respeaker_v2_pixel_ring.py without sudo. Running the template `python examples/respeaker_v2_pixel_ring.py` the LEDs light up.

[image]

It is impossible to install apt install -y python-mraa without sudo, it does not give

[image]

In the file apa102.py, which is located `/home/respeaker/HermesLedControl/interfaces/apa102.py`, I specified in the line mara.Gpio(12)

Launched HermesLedControl, there is still "Device not using gpiozero or mraa, ignore power: name 'mraa' is not defined`"

[image]

Psychokiller1888, 2023-04-08:
you run pixel-ring from outside HLC venv, which gives it access to the global mraa package. HLC tries to load it but fails as it runs in its own virtual environment. The solution is simple, get HLC to be able to load mraa, in other words, make it available in HLC venv. Pixel-ring is just a wrapper around the apa102.py driver, and does the same HLC does. I'll try to find a solution to get mraa in you venv

Psychokiller1888, 2023-04-08:
Ok, I tried something...
https://github.com/project-alice-assistant/HermesLedControl/tree/fix-respeakerCoreV2

Installing mraa before venv creation and making the venv creation inherit from system packages to confirm my theory. You'll have to fresh install HLC to try it. It's not optimal as it forces pythn-mraa install for every users even those not needing it, but well, give it a shot.

If not working, do change the GPIO pin from 5 to 12 and try again

Psychokiller1888, 2023-04-08:
So, here 2 things for you to try:

First: https://github.com/project-alice-assistant/HermesLedControl/commit/42404126d7a7414d7fcb9386e6cccec6da2fb6d7

This tries to install python-mraa and then create the virtual-environement, with the global packages flag, which should take it with into HLC venv. I don't like it, because it forces everybody to install mraa which is only used by respeaker Core

Second:
https://github.com/project-alice-assistant/HermesLedControl/commit/0ae9cfda2123d323045edf2fdbcc8d8ba0fca020 and
https://github.com/project-alice-assistant/HermesLedControl/commit/39c0d23a95486298aaff0a63183b34e00e687adf

This doesn't install mraa on HLC install, but only when respeaker core v2 is selected. It is then copied from system packages into the venv itself before activating it. As I'm not home and don't have access to any debian os, I cannot confirm the copy command to work as I don't know if it's installed there

DivanX10, 2023-04-08:
It didn't work. I repeated the installation, put it from scratch and did everything as usual. I still can't get HLC to work with your method, and by running the script, the LEDs work. I even deleted pixel_ring as indicated by your command `pip3 uninstall -y pixel_ring`

[image]
[image]
[image]
[image]
[image]

Moved to the HermesLedControl folder
```
cd /home/respeaker/HermesLedControl
```

and started debugging
```
./venv/bin/python main.py --debug=True --engine=rhasspy --hardware=respeakerCoreV2 --mqttServer=localhost --mqttPort=1883 --clientId=respeakerv2 --leds=12 --pattern=google --pathToConfig=/home/respeaker/.config/rhasspy/profiles/ru/profile.json
```

[image]

Psychokiller1888, 2023-04-08:
Have you tried to copy python-mraa to the HLC venv?

DivanX10, 2023-04-08:
> Have you tried to copy python-mraa to the HLC venv?

I couldn't stand it, I didn't understand what needed to be done. You can describe the detailed steps, I will repeat.

Возможно вы имеете ввиду установить `python-mraa python-upm libmraa1 libupm1 mraa-tools `в виртуальную среду python? Я не умею работать с виртуальной средой python и не знаю как там нужно правильно работать.

Или возможно вы имеете ввиду перенести папку pixel_ring в venv? 

[image]

**Не знаю, поможет ли вам эта информация**
Опытным путем было выявлено, что  GPIO будут доступны, если установить пакеты `python-mraa python-upm libmraa1 libupm1 mraa-tools` на debian 9, а если поставить на debian 10, то GPIO будут недоступны, поэтому прежде чем обновляться до Debian 10 или 11, обязательно нужно установить  пакеты `python-mraa python-upm libmraa1 libupm1 mraa-tools` на debian 9 и как только GPIO будут активны и они будут видны через команду `mraa-gpio list`, то можно смело обновляться до до Debian 10 или 11. Rhasspy работает только на Debian 10, можно поставить и на Debian 11, но тогда нужно пересобирать пакет из-за пакета libgfortran5, так как в сборке стоит libgfortran4, а нужен libgfortran5. Поэтому для работы Rhasspy Debian необходимо обновить до 10 или 11 версии. Вот тут и возникает проблема, что для работы светодиодов пакеты `python-mraa python-upm libmraa1 libupm1 mraa-tools` надо ставить на debian 9, а после обновив до до Debian 10 или 11, ставим Rhasspy и HLC.

Psychokiller1888, 2023-04-08:
Check my second solution I gave earlier. Locate where python-mraa is installed and make a copy of it in your HLC venv

DivanX10, 2023-04-09:
> Проверьте мое второе решение, которое я дал ранее. Найдите, где установлен python-mraa, и создайте его копию в вашем HLC venv

I tried, this option does not work, I even copied to different places, tried with respeaker and root rights. There is no python-mraa folder, there are only files mraa.py, mraa.pyc, _mraa.arm-linux-gnueabihf.so

[image]
[image]
[image]
[image]

***

But this option worked, but on the condition that you need to run a python script to turn on the sudo python turn_on_leds.py , but it works until you reboot Debian

```
cd /home/respeaker/HermesLedControl
source ./venv/bin/activate
git clone --depth 1 https://github.com/respeaker/pixel_ring.git
cd pixel_ring
pip install setuptools-scm
pip install -U -e .
cd
cd leds_ring/
sudo python turn_on_leds.py
```

File apa102.py which is located /home/respeaker/HermesLedControl/interfaces/apa102.py, no need to edit, works with mara.Gpio(5)

[image]

Psychokiller1888, 2023-04-09:
Ok, so if you have mraa.py in your HLC venv, HLC should not make an error importing it anymore when starting?

DivanX10, 2023-04-09:
> Ok, so if you have mraa.py in your HLC venv, HLC should not make an error or tell you the device is not using mraa anymore when starting?

There are errors about mraa, but the LEDs work
[image]

Psychokiller1888, 2023-04-09:
I'll get a respeaker core v2 back and will try. The solution is simple, have mraa in your HLC venv. I don't have any devices that use mraa, can't check now

DivanX10, 2023-04-09:
Where will I get ready-made python-mraa? I use the one offered by respeaker, and installing python-mraa from respeaker, it is not in HLC venv.

Link to respeaker repositories, from where this python-mraa is put
http://respeaker.seeed.io/deb/dists/stretch/

rooferdave, 2023-07-29:
> I'll get a respeaker core v2 back and will try. The solution is simple, have mraa in your HLC venv. I don't have any devices that use mraa, can't check now

Any Luck? You were on to somthing.